espressif/arduino-esp32 #10968: Uploading LittleFS filesystem Linux bookworm

Bredahl, 2025-02-14:
### Board

ESP32 dev module, ESP32 C3, ESP32S2, ESP32 S3, ESP32c6 modules

### Device Description

After Esp32 core version 3.0.7 I can't upload to LittleFS file system anymore
I'm running Linux bookworm and arduino version 2.3.4
I get the error message below.

Uploading LittleFS filesystem
Command Line: python3 /home/jb/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py --chip esp32c3 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /home/jb/.var/app/cc.arduino.IDE2/cache/tmp-153-2KqLgxiZnKud-.littlefs.bin

python3: can't open file '/home/jb/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py': [Errno 2] No such file or directory

ERROR:  Upload failed, error code: 2

### Hardware Configuration

None

### Version

v3.1.0

### IDE Name

Arduino IDE 2.3.4

### Operating System

Linux Bookworm

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

After Esp32 core version 3.0.7 I can't upload to LittleFS file system anymore
I'm running Linux bookworm and arduino version 2.3.4
I get the error message below.

### Sketch

```cpp
none
```

### Debug Message

```plain
Uploading LittleFS filesystem
Command Line: python3 /home/jb/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py --chip esp32c3 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 2686976 /home/jb/.var/app/cc.arduino.IDE2/cache/tmp-153-2KqLgxiZnKud-.littlefs.bin

python3: can't open file '/home/jb/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3/esptool.py': [Errno 2] No such file or directory

ERROR: Upload failed, error code: 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-02-14:
The tool you are using to upload needs to be updated to use the esptool binaries that we now provide. Please contact the tool creator

closed 2025-02-14 by me-no-dev

Bredahl, 2025-02-14:
Ok thanks. perhaps tool creator has updated to use with 3.1.2