espressif/arduino-esp32 #10971: Crash in serializeJson with Empty Values after Update (Arduino-ESP32 3.1.2)

hosseinaghaie, 2025-02-15:
### Board

Board: ESP32-S3

### Device Description

Description:
After updating to Arduino-ESP32 3.1.2, we observed frequent crashes when using serializeJson with DynamicJsonDocument. The issue occurs specifically when handling empty or minimal JSON objects like {} or when the serialized String is empty (""). This behavior was not present in previous versions and might be related to recent changes in String behavior (String::move fix). The crash generates a Guru Meditation Error (LoadProhibited).

Example Code:
DynamicJsonDocument doc(2048);
doc["data"] = "{}";
String json;
serializeJson(doc, json);  // Causes crash

We also experience similar crashes when serializing responses with minimal data (doc["data"] = {} or []), which should be valid JSON structures.

By switching to std::string for handling serialization, the crash is avoided. The modified code is shown below:

#include <string>  // Use std::string

DynamicJsonDocument doc(2048);
doc["data"] = "{}";
std::string jsonString;
serializeJson(doc, jsonString);  // No crash
This might suggest an underlying issue with String conversion in the recent core update.

Environment:
Board: ESP32-S3
Core Version: Arduino-ESP32 3.1.2
ArduinoJson Version: 7.3.0
The issue seems closely related to serializeJson combined with String. Let me know if you need additional debug logs or a full backtrace.

### Hardware Configuration

no

### Version

v3.1.2

### IDE Name

Arduino 2.3.4

### Operating System

11

### Flash frequency

40m

### PSRAM enabled

no

### Upload speed

115200

### Description

no

### Sketch

```cpp
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060630  A0      : 0x820405c3  A1      : 0x3fcbf1a0  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fcbf1d0  
A10     : 0x00060a23  A11     : 0x00000000  A12     : 0x00060a20  A13     : 0x3fc9a430  
A14     : 0x3fcba5d4  A15     : 0xffffffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x400556d2:0x3fcbf1a0 0x420405c0:0x3fcbf1b0 0x42016ddf:0x3fcbf1d0 0x42016e69:0x3fcbf260 0x42017e12:0x3fcbf290 0x42019435:0x3fcbf380 0x420198bb:0x3fcbf3e0 0x4201cd7e:0x3fcbf4f0 0x42049fb0:0x3fcbf540 0x4037ee32:0x3fcbf560
```

### Debug Message

```plain
Decoding stack results
0x420405c0:  is in String::operator=(char const*) (C:\Users\Admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.2\cores\esp32\WString.cpp:293).
0x42016ddf:  is in ArduinoJson::V730PB22::detail::serialize<ArduinoJson::V730PB22::detail::JsonSerializer, String>(ArduinoJson::V730PB22::JsonVariantConst, String&) (c:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writers/ArduinoStringWriter.hpp:19).
0x42016e69:  is in ArduinoJson::V730PB22::serializeJson<String, 0>(ArduinoJson::V730PB22::JsonVariantConst, String&) (c:\Users\Admin\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:137).
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

InnuendoPi, 2025-02-15:
I also noticed this error. same with pioarduino 53.03.12. switched back to 3.1.1 (53.03.11)

maybe same issue with tzapu/WiFiManager lib: empty WLAN config (start AP mode) ends up in a exception loop, when calling 
bool res = wifiManager.autoConnect();
in setup(). same source no probs with 3.1.1 

Exception:
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008c1ad  PS      : 0x00060230  A0      : 0x8012c0ec  A1      : 0x3ffb20f0
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb20c0
A10     : 0x3ffc6541  A11     : 0x00000009  A12     : 0x00000009  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0x3ffc6538  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008c1ad  LEND    : 0x4008c1bd  LCOUNT  : 0xffffffff

Backtrace: 0x4008c1aa:0x3ffb20f0 0x4012c0e9:0x3ffb2100 0x40125072:0x3ffb2120 0x4012564d:0x3ffb2190 0x40109c80:0x3ffb2200 0x4012efdb:0x3ffb2270 0x400900c2:0x3ffb2290

sorry, cannot use exception decoder (debug flash size grows 139%).

me-no-dev, 2025-02-15:
@TD-er PTAL. This is a consequence of your changes :)

TD-er, 2025-02-15:
Yep, now that I see the crash (+ line nr), it is obvious why it crashes and is indeed due to my changes in this PR:
https://github.com/espressif/arduino-esp32/commit/5488d5d23fb0da47f463fce89a879c4593a6231f

It should not call `strlen()` on a nullptr.

TD-er, 2025-02-15:
Could you test if this will fix it? 
https://github.com/espressif/arduino-esp32/pull/10972
I just edited it in my browser, so not tested yet myself.

hosseinaghaie, 2025-02-15:
> Could you test if this will fix it? [#10972](https://github.com/espressif/arduino-esp32/pull/10972) I just edited it in my browser, so not tested yet myself.

i rechecked it,the proble solved!

enjoyneering, 2025-02-15:
Just tried the fix. My code stopped crashing with the latest ArduinoJson library.

InnuendoPi, 2025-02-16:
same test result with tzapu/WiFiManager lib: no exceptions
thanks for fast response!

closed 2025-02-17 by me-no-dev

AlainF-ing, 2025-02-19:
Arduino-ESP32 3.1.3 solved a similar issue with ESP32-C3 crashing at run time on  

serializeJson(doc, json); statement

Thanks for that quick fix

PR #10972 (2025-02-17): Fix crash in String when using nullptr (#10971)
## Description of Change
See: https://github.com/espressif/arduino-esp32/issues/10971#issuecomment-2660859418

## Related links

Fixes: #10971
commit 758cbd09 (2025-02-17): Fix crash in String when using nullptr (#10971) (#10972)

Fixes: #10971