espressif/arduino-esp32 #10975: I2S driver i2s_std_clk_config_t macro compilation error

earlephilhower, 2025-02-16:
### Board

ESP32C6 Dev Module

### Device Description

Plain generic ESP32-C6 board with no peripherals attached

### Hardware Configuration

N/A

### Version

v3.1.2

### IDE Name

Arduino IDE

### Operating System

Linux Ubuntu 22.04

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

Setting default clocking on I2S using the IDF macros fails with a compilation error (initializer out of order in the IDF header itself).  3.1.1 release builds OK, only 3.1.2 with a newer IDF shows the issue.

I'm not sure if this needs to be tracked here or the IDF repo as the issue relates to an IDF header/macro with compile errors.  I have opened https://github.com/espressif/esp-idf/issues/15405 with similar details since this is an IDF header and not something the Arduino-ESP32 guys made.

### Sketch

```cpp
`
#include <driver/i2s_std.h>
void setup() {
int freq = 44100;
i2s_std_clk_config_t clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG((uint32_t)freq);
}
void loop() {
}
`
```

### Debug Message

```plain
`
/home/earle/.arduino15/packages/esp32/tools/esp-rv32/2405/bin/riscv32-esp-elf-g++ -MMD -c @/home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/flags/cpp_flags -Wall -Wextra -Og -g3 -Werror=return-type -DF_CPU=160000000L -DARDUINO=10607 -DARDUINO_ESP32C6_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32C6_DEV\"" "-DARDUINO_VARIANT=\"esp32c6\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/flags/defines -I/tmp/.arduinoIDE-unsaved2025116-3435406-1cqshei.eked/sketch_feb16b -iprefix /home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/include/ @/home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/flags/includes -I/home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/qio_qspi/include -I/home/earle/.arduino15/packages/esp32/hardware/esp32/3.1.2/cores/esp32 -I/home/earle/.arduino15/packages/esp32/hardware/esp32/3.1.2/variants/esp32c6 @/tmp/arduino/sketches/A089F771BEEC5FF8CFA7EE808BB83FDC/build_opt.h @/tmp/arduino/sketches/A089F771BEEC5FF8CFA7EE808BB83FDC/file_opts /tmp/arduino/sketches/A089F771BEEC5FF8CFA7EE808BB83FDC/sketch/sketch_feb16b.ino.cpp -o /tmp/arduino/sketches/A089F771BEEC5FF8CFA7EE808BB83FDC/sketch/sketch_feb16b.ino.cpp.o
In file included from /tmp/.arduinoIDE-unsaved2025116-3435406-1cqshei.eked/sketch_feb16b/sketch_feb16b.ino:1:
/tmp/.arduinoIDE-unsaved2025116-3435406-1cqshei.eked/sketch_feb16b/sketch_feb16b.ino: In function 'void setup()':
/home/earle/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32c6/include/esp_driver_i2s/include/driver/i2s_std.h:220:1: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'
  220 | }
      | ^
/tmp/.arduinoIDE-unsaved2025116-3435406-1cqshei.eked/sketch_feb16b/sketch_feb16b.ino:4:32: note: in expansion of macro 'I2S_STD_CLK_DEFAULT_CONFIG'
    4 | i2s_std_clk_config_t clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG((uint32_t)freq);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved2025116-3435406-1cqshei.eked/sketch_feb16b/sketch_feb16b.ino:4:22: warning: unused variable 'clk_cfg' [-Wunused-variable]
    4 | i2s_std_clk_config_t clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG((uint32_t)freq);
      |                      ^~~~~~~

exit status 1
`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-02-17:
For Arduino it should be fixed with this https://github.com/espressif/arduino-esp32/pull/10963/files
Since this change i don't see an compile fail when building the Arduino libs with the Arduino Lib Builder.

tomcombriat, 2025-02-23:
I have stolen the same fix for [Mozzi's port for ESP32.](https://github.com/sensorium/Mozzi/pull/301)

Jason2866, 2025-03-12:
Fixed with https://github.com/espressif/arduino-esp32/pull/10963

closed 2025-03-12 by Jason2866