espressif/arduino-esp32 #10976: Fiber support (SFP Port) with ESP32

greenmagics, 2025-02-16:
### Related area

Ethernet

### Hardware specification

ESP32

### Is your feature request related to a problem?

Hello!

I plan to make devices equipped with an SFP port for long-distance communication.

At the moment I use ESP32 + LAN8720A + external media converter, but it generates a large amount of needed space.

From the list of supported ethernet IC, both KSZ8041FTL and IP101GRI support fiber optic connection, but for several days I have not been able to find any information whether someone managed to do it on ESP32.

Does the use of the SFP port depend only on the electronic schematic or are software changes also required?

Here is an example photo from the internet showing the KSZ8041FTL connection with the SFP port (I don't know if the schematic is correct)
[image]

### Describe the solution you'd like

Add fiber connection mode.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2025-02-16:
Since this question really does not have anything to with arduino, I'd recommend you cross-post at https://esp32.com/viewforum.php?f=12

lbernstone, 2025-02-18:
[Looks to me](https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/DataSheets/00002436B.pdf) like it is just a matter of the physical connections (FXSD high when connected), and that schematic is good.  Make sure you have an adequate power supply!

[image]

[image]