espressif/arduino-esp32 #10977: Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?

hzhh110, 2025-02-17:
### Board

esp32-s3-wroom

### Device Description

Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?

### Hardware Configuration

Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?

### Version

v3.1.2

### IDE Name

VSCode platformio

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?

### Sketch

```cpp
Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?
```

### Debug Message

```plain
Can the ESP32-S3 connect to LAN8720? What IO port should be configured and connected?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-02-17:
The ESP32-S3 does not include a RMII Ethernet PHY.  You can use an SPI based adapter like the W5500.

closed 2025-02-17 by me-no-dev

hzhh110, 2025-02-17:
If it is W5500, how to connect ESP32-S3?And the ArduinoHow to call the relevant interface?After reading the ETH file of ESP32, is there any relevant information?

me-no-dev, 2025-02-17:
@hzhh110 here is an example: https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_W5500_Arduino_SPI/ETH_W5500_Arduino_SPI.ino