espressif/arduino-esp32 #10978: example code for USB Host

marchingband, 2025-02-18:
### Related area

USB

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

I would like to use USB Host on S3.

### Describe the solution you'd like

I hear that USB Host support is ready from tinyusb, which is very exciting.
https://github.com/hathach/tinyusb/issues/2634#issuecomment-2646978422

Would it be possible to confirm this?
Will we see some example code at some point?
I am particularly interested in MIDI USB Host.
thank you!

### Describe alternatives you've considered

Currently I use esp-idf for USB MIDI Host, but I would like to use Arduino, because I produce development boards, and my users prefer Arduino.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2025-02-18:
we have not added host support to Arduino yet, but we will look into it in near future

me-no-dev, 2025-02-18:
Sure we can. I updated the title

JorenSix, 2025-02-25:
I just wanted to add that I am also looking for MIDI capabilities in Arduino environments with an S3. Would be great to see this working in a similar way as with the Teensy devices with MIDI and other USB device type support. Thanks in advance.

polygonfuture, 2025-05-07:
I am looking for this feature as well.  Spent three days tring to implement and it's hell.

Would be great for this to be in the arduino esp core.    Using an m5stack Core S3 (ESP32-S3)

marchingband, 2025-05-07:
@me-no-dev any updates? Is this still on the road map? Thanks!

me-no-dev, 2025-05-07:
It is on the roadmap. Can not give time estimate yet

davidandreoletti, 2025-05-13:
> It is on the roadmap. Can not give time estimate yet

@me-no-dev I understand the "no date" commitment. If this is something you can communicate, would you able to indicate whether that's a low or middle priority task at the moment ?

me-no-dev, 2025-05-13:
It's low to middle priority @davidandreoletti we have issues to fix with currently supported peripherals (including USB device)

marchingband, 2025-08-21:
Havnt tested this but it looks promising, not sure how they pulled it off.

https://github.com/enudenki/esp32-usb-host-midi-library

me-no-dev, 2025-11-05:
it's being worked on. Current core 3.3.3 already supports TinyUSB host, but still needs Arduino abstraction for it.

me-no-dev, 2025-11-11:
@lyusupov we have yet to look into that host library. feel free to make a PR with the needed changes

lyusupov, 2025-11-11:
@me-no-dev 

> feel free to make a PR with the needed changes

Filed as PR https://github.com/espressif/esp32-arduino-lib-builder/pull/330

closed 2026-09-16 by me-no-dev