espressif/arduino-esp32 #10989: ESP32 LittleFS - Corrupted dir pair at {0x0, 0x1}

dirkvranckaert, 2025-02-19:
### Board

ESP32-C6

### Device Description

This is a bare ESP32-C6

### Hardware Configuration

ESP32-C6. Version of arduino-esp32 does not allow me to select 3.1.3 (latest, so I set it to 3.1.2, though 3.1.3 is the latest one I tested it on).

### Version

v3.1.2

### IDE Name

Arduino IDE

### Operating System

MacOS

### Flash frequency

8Mhz

### PSRAM enabled

yes

### Upload speed

460800

### Description

While running the application we are writing to 2 files from time to time. However after a redeploy / restart / power-cycle it happens that we _loose_ the file system.

What we do when we prepare the board with our _installation_ sketch is to execute `LittleFS.begin(true)` , we even do a proper `LittleFS.format();` at some point. So far so good.

At startup of the _production_ sketch we execute `LittleFS.begin(false)`, false here to avoid any format of the storage since all we'll be doing is to write local updates (measurements) to it. If we would allow a format at this stage we would loose data after a power cycle as well as WiFi connection details (credentials) that are stored in the local storage.

Though when power cycling we tend to get this error all of the sudden:

```
09:25:23.629 -> E (1101) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1384:error: Corrupted dir pair at {0x0, 0x1}
```

And the only way to recover from it is to allow a format at startup by executing `LittleFS.begin(true)`.

However what can we do to avoid `corrupted dir pair at {0x0, 0x1}`?

### Sketch

```cpp
n/a
```

### Debug Message

```plain
09:25:23.629 -> E (1101) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1384:error: Corrupted dir pair at {0x0, 0x1}
09:25:23.661 -> 
09:25:23.661 -> E (1102) esp_littlefs: mount failed,  (-84)
09:25:23.661 -> E (1106) esp_littlefs: Failed to initialize LittleFS
09:25:23.661 -> [   504][E][LittleFS.cpp:79] begin(): Mounting LittleFS failed! Error: -1
09:25:23.661 -> FAILED SETTING UP LittleFS!!!!
09:25:23.661 -> [   522][E][vfs_api.cpp:106] exists(): File system is not mounted
09:25:23.661 -> [   531][E][vfs_api.cpp:106] exists(): File system is not mounted
09:25:23.693 -> [   537][E][vfs_api.cpp:194] mkdir(): File system is not mounted
09:25:23.693 -> [   543][E][vfs_api.cpp:194] mkdir(): File system is not mounted
09:25:23.693 -> [   548][E][vfs_api.cpp:23] open(): File system is not mounted
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2025-02-19:
Do you also close the file when done writing?

dirkvranckaert, 2025-02-20:
@TD-er yeah everything is properly closed.

```

void _writeFile2(fs::FS &fs, const char *path, const char *message) {
  if (!fs.exists(path)) {
    if (strchr(path, '/')) {
      Serial.printf("Create missing folders of: %s\r\n", path);
      char *pathStr = strdup(path);
      if (pathStr) {
        char *ptr = strchr(pathStr, '/');
        while (ptr) {
          *ptr = 0;
          fs.mkdir(pathStr);
          *ptr = '/';
          ptr = strchr(ptr + 1, '/');
        }
      }
      free(pathStr);
    }
  }
  Serial.printf("Writing file to: %s\r\n", path);
  File file = fs.open(path, FILE_WRITE);
  if (!file) {
    Serial.println("- failed to open file for writing");
    return;
  }
  if (file.print(message)) {
      Serial.println("- file written");
  } else {
      Serial.println("- write failed");
  }
  file.close();
}
```

@Jason2866 Was trying that QIO and DIO flash modes actually yesterday as well but did not make any difference...

TD-er, 2025-02-20:
You have stated flash frequency is 8 MHz in your start post.
Is that a typo?
Can you try setting it to 40 MHz?

How is the ESP connected to your PC for flashing?
Are you using (native) USB, or is it wired via some USB to serial chip?
If so, which one is it? The rectangular CH340 or other CHxxx or is it a square tiny one?
Those CHxxx ones may not be that reliable at higher baud rates, just to make sure there wasn't a bitflip during flashing.

Jason2866, 2025-02-20:
Using heavily LittleFS in our project (Tasmota). LittleFS does work without any issue with the C6.
Try different hardware for flashing and the target C6 device. If the C6 is a custom made board replace with a known good one.

Jason2866, 2025-03-12:
Closing since not reproducable. Feel free to reopen when the issue exists with an commercial available board.

closed 2025-03-12 by Jason2866

eleicsjaivik, 2025-03-26:
It solves the problem by unlocking the blocks at first, because some of memory datasheets has locked blocks on boot, that you might have to unlock it every time you reset the flash

    if (unlock_blocks() != ESP_OK) ESP_LOGE("FLASH", "Unlocking blocks fail!");
    else ESP_LOGI("FLASH", "Unlocking blocks success!");