espressif/arduino-esp32 #10990: ESP32S3 UART Communication Issue with Latest Board Manager Version3.1.3

miku-gotoubun, 2025-02-19:
### Board

ESP32S3

### Device Description

When using ESP32S3 to receive GPS data via UART, no data was displayed on the serial monitor. After troubleshooting and verifying hardware connections, pin assignments, and code logic, the issue was resolved by downgrading the ESP32 Board Manager version. This suggests that the problem might be related to the latest version of the Board Manager.

### Hardware Configuration

- Board: ESP32S3
- GPS Module: GYSFDMAXB
- UART Connection: RX → GPIO43, TX → GPIO44

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

- ESP32 Variant: ESP32-S3-WROOM-1 (Surface Mount)
- PCB Design: Custom PCB (Self-designed)
- GPS Module: GYSFDMAXB
- UART Connection: RX → GPIO43, TX → GPIO44, GND connected
- Power Supply: 3.3V regulated supply

### Sketch

```cpp
void setup() {
  Serial1.begin(9600, SERIAL_8N1, 43, 44);
  Serial.begin(115200);
}

void loop() {
  if (Serial1.available() > 0) {
    String line = Serial1.readStringUntil('\n');
    Serial.println("Received Data: " + line);
  }
}
```

### Debug Message

```plain
Cortex-Debug: VSCode debugger extension version 1.5.1 git(be7d3c8+dirty). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:\Users\Asdei\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2405/bin/xtensa-esp32s3-elf-objdump --syms -C -h -w C:/Users\Asdei\AppData\Local\arduino\sketches\01C892C0C33DE54ED3EB9D2ACF18FF7C\GPS_ESP32s3_ver1_20241225.ino.elf
Reading symbols from c:/users\asdei\appdata\local\arduino15\packages\esp32\tools\esp-x32\2405/bin/xtensa-esp32s3-elf-nm --defined-only -S -l -C -p C:/Users\Asdei\AppData\Local\arduino\sketches\01C892C0C33DE54ED3EB9D2ACF18FF7C\GPS_ESP32s3_ver1_20241225.ino.elf
Launching GDB: "C:\\Users\\Asdei\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp-elf-gdb\\14.2_20240403\\bin\\xtensa-esp32s3-elf-gdb.exe" -q --interpreter=mi2 "C:/Users\\Asdei\\AppData\\Local\\arduino\\sketches\\01C892C0C33DE54ED3EB9D2ACF18FF7C\\GPS_ESP32s3_ver1_20241225.ino.elf"
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: "C:\\Users\\Asdei\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.12.0-esp32-20241016/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\Asdei\\Arduino\\GPS_ESP32s3_ver1_20241225" -f "C:/Program Files/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32s3-builtin.cfg
    Please check TERMINAL tab (gdb-server) for output from C:\Users\Asdei\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20241016/bin/openocd
Finished reading symbols from objdump: Time: 307 ms
Finished reading symbols from nm: Time: 577 ms
warning: 
could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
Reading symbols from C:/Users\Asdei\AppData\Local\arduino\sketches\01C892C0C33DE54ED3EB9D2ACF18FF7C\GPS_ESP32s3_ver1_20241225.ino.elf...
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
```

### Other Steps to Reproduce

**Serial Monitor Output:**  
No output displayed. The serial monitor remains blank even after waiting for several minutes.

**UART Debugging:**  
- Added `Serial.println("Debug: Entering loop");` → Output displayed as expected.
- Added `Serial.println(Serial1.available());` → Always returns 0, indicating no data is received from UART1.

**Verbose Upload Log:**  
- Compilation and upload successful with no errors or warnings.

**Baud Rate:**  
- Code: `Serial.begin(115200);` and `Serial1.begin(9600, SERIAL_8N1, 43, 44);`
- Serial monitor: 115200 baud, NL & CR enabled.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

miku-gotoubun, 2025-02-19:
This issue was resolved by downgrading the version from 3.1.3 to 3.1.1.

TD-er, 2025-02-19:
Do you have any received data to show?
Or didn't you get any sensible data at all?

SuGlider, 2025-02-19:
@miku-gotoubun - There are no changes in the Core related to UART from 3.1.1 to 3.1.3.
How did you update/downgrade from one version to another? 

Using just a `git checkout` won't work. It is necessary to also run `tools/get.py`.

SuGlider, 2025-02-19:
I had a similar issue when I was working on the LP UART for the C6/P4.
Everything was working using 3.1.2 and then I just `git checkout master` to upgrade it to the latest Core 3.1.3 and it stoped working. 

I had to run `tools/get.py` to also update the Libraries and its binaries. After that it started to work again.

SuGlider, 2025-02-19:
> [@SuGlider](https://github.com/SuGlider) I think he uses the board manager and just did not care to set the proper version in the template.
> 
> [@miku-gotoubun](https://github.com/miku-gotoubun) please set the core debug level to verbose and post the UART output (not gdb)

By board manager, it means using the Arduino IDE. In such case, there should be no change, as the UART code (`Hardwareserial.cpp` | `esp32-hal-uart.c`)  has no changes at all.

Anyway, I'll test ESP32-S3 `Serial1` communication using an USB-UART CH340, in a separated port, using the Arduino IDE to change the Core versions.

miku-gotoubun, 2025-02-19:
> Do you have any received data to show? Or didn't you get any sensible data at all?
I didn't get any sensible data at all . I confirmed that nothing was displayed on the serial monitor in the arduino IDE.

> [@SuGlider](https://github.com/SuGlider) I think he uses the board manager and just did not care to set the proper version in the template.
I use the Arduino IDE as my development environment. When I updated the board manager in the Arduino IDE to 3.1.3, the above problem occurred. After trying all sorts of methods, I solved it by downgrading the board manager from 3.1.3 to 3.1.1.

> [@miku-gotoubun](https://github.com/miku-gotoubun) please set the core debug level to verbose and post the UART output (not gdb)
I'm home now and don't have the Esp32s3 and GPS with me, I'll do it tomorrow.

SuGlider, 2025-02-19:
@miku-gotoubun - I just noticed that `Serial1` uses UART0 default pins (44 for RX and 43 for TX), but `Serial` is initialized also using the same default pins, therefore `Serial1` will have no pins connected as GPIO43 and GPIO44 will be used by `Serial`.

### There is no way to make Serial and Serial1 work unsing this settings:
``` cpp
  Serial1.begin(9600, SERIAL_8N1, 43, 44); // ESP32-S3 will connect pins 43 as RX1 and 44 as TX1 into UART1, leaving UART0 with no pins connected

  // This is the same as Serial.begin(115200, SERIAL_8N1, 44, 43) which will use the same Serial1 pins
  Serial.begin(115200);   // ESP32-S3 will disconnect pins 43 and 44 from UART1 makeing impossible to see Serial1 data
```

[image]

SuGlider, 2025-02-19:
### Arduino Core 3.1.1 Debug output:
```
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32S3_DEV
  Arduino Variant   : esp32s3
  Arduino FQBN      : esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
[   504][V][esp32-hal-uart.c:421] uartBegin(): UART1 baud(9600) Mode(800001c) rxPin(43) txPin(44)
[   513][V][esp32-hal-uart.c:510] uartBegin(): UART1 not installed. Starting installation
[   655][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc93000
[   666][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   397332 B ( 388.0 KB)
  Free Bytes        :   356036 B ( 347.7 KB)
  Allocated Bytes   :    35752 B (  34.9 KB)
  Minimum Free Bytes:   355896 B ( 347.6 KB)
  Largest Free Block:   286708 B ( 280.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

SuGlider, 2025-02-19:
### Arduino Core 3.1.3 Debug output:

```
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Feb 19 2025 13:02:42
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-584-g489d7a2b3a-dirty
  Arduino Version   : 3.1.3
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32S3_DEV
  Arduino Variant   : esp32s3
  Arduino FQBN      : esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
[   504][V][esp32-hal-uart.c:421] uartBegin(): UART1 baud(9600) Mode(800001c) rxPin(43) txPin(44)
[   513][V][esp32-hal-uart.c:510] uartBegin(): UART1 not installed. Starting installation
[   655][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc93100
[   666][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   397076 B ( 387.8 KB)
  Free Bytes        :   355780 B ( 347.4 KB)
  Allocated Bytes   :    35752 B (  34.9 KB)
  Minimum Free Bytes:   355640 B ( 347.3 KB)
  Largest Free Block:   286708 B ( 280.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

SuGlider, 2025-02-19:
@miku-gotoubun - final conclusion is that when the firmware starts the sketch, UART0 (console) is initialized in boot using default configuration [115200, 8N1, RX0 (44) and TX0 (43)].

After executing `Serial1.begin(9600, SERIAL_8N1, 43, 44);`, Arduino will detach UART0 pins and invert default UART0 pins (RX/TX) to be used for UART1 [9600, 8N1, RX1(43) and TX1(44)]. At this point UART0 won't work anymore and it stops sending any debug information to is previous TX0 pin (43).

When `Serial.begin(115200);` is executed, it will restablish UART0 default pins back, as used in console. Log output informing it won't be visible until the pins are set (in special default TX0 pin 43). After that, UART1 driver still works, but no pins are associated to it anymore, therefore it won't work anyway.

#### There is no issue with Arduino Core as it is executing are desired by the provided sketch

SuGlider, 2025-02-22:
@miku-gotoubun - please let me know if this issue can be closed.

miku-gotoubun, 2025-02-23:
@SuGlider 
Thank you for your patience. I appreciate your insights and contributions.

At the moment, I am engaged in another project, which makes it difficult for me to address this issue. Additionally, it is unclear whether this problem is specific to my environment, which makes identifying the root cause and resolving the issue challenging at this time.

If anyone in the community is willing to take on this issue, I will be happy to provide support where possible. Otherwise, I plan to revisit this issue once my current project is completed.

Since I am unable to address this issue for the time being, I will close it for now. If you have any additional information or solutions, please feel free to share them. I will reopen the issue if necessary.

Thank you for your understanding.

closed 2025-02-23 by miku-gotoubun