espressif/arduino-esp32 #10995: Inconsistent ADC Readings Across ESP32-WROOM-32E Modules

anuraagMIPL, 2025-02-20:
### Board

ESP32-WROOM-32E

### Device Description

Plain Module

### Hardware Configuration

GPIO 36 (Sensor VP) Pin

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10 Pro

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello,

I am using three different ESP32-WROOM-32E modules and measuring ADC readings on GPIO36 (sensor VP pin) with input voltages ranging from 2.00V to 2.81V. I have verified the input voltage using a DSO, and the hardware input to the pin is identical and accurate across all modules as shown in excel.

However, I have observed significant variations in ADC readings between the three modules, which do not align with the expected formula:

ADC_Value = (4095 * V) / 3.3

I have attached a spreadsheet (ADC_Reading.xlsx) containing the raw ADC values from all three modules. According to the ESP32-WROOM-32E datasheet, ADC readings can vary by approximately ±6%, but my measured values exceed this range. This suggests that ADC behavior differs between modules, even under identical conditions.

My Questions:

Is there an inherent variation in ADC calibration across different ESP32-WROOM-32E modules?
Is there a recommended calibration procedure to normalize ADC readings across multiple ESP32 modules?
If anyone has encountered similar issues or has insights into standardizing ADC readings, I would greatly appreciate your advice! Thanks in advance.

[ADC Reading.xlsx](https://github.com/user-attachments/files/18881409/ADC.Reading.xlsx)

### Sketch

```cpp
#include <Arduino.h>

#define ADC_PIN 36
#define BUFFER_SIZE 2000

uint16_t adcBuffer[BUFFER_SIZE] = {0};
uint32_t adcSum = 0;
uint16_t bufIndex = 0;

void setup()
{
  Serial.begin(115200); // Initialize Serial Monitor
  pinMode(ADC_PIN, INPUT);
}

void loop()
{
  // Read ADC value
  uint16_t adcValue = analogRead(ADC_PIN);

  // Update moving average
  adcSum -= adcBuffer[bufIndex];  // Remove oldest value
  adcBuffer[bufIndex] = adcValue; // Store new value
  adcSum += adcValue;             // Add new value
  bufIndex = (bufIndex + 1) % BUFFER_SIZE;

  // Compute averaged ADC value
  uint16_t avgADC = adcSum / BUFFER_SIZE;

  // Print ADC value
  Serial.println(avgADC);
}
```

### Debug Message

```plain
11:23:38:342 -> 2887
11:23:38:342 -> 2887
11:23:38:342 -> 2887
11:23:38:343 -> 2887
11:23:38:343 -> 2887
11:23:38:344 -> 2887
11:23:38:346 -> 2887
11:23:38:348 -> 2887
11:23:38:349 -> 2887
11:23:38:350 -> 2887
11:23:38:351 -> 2887
11:23:38:352 -> 2887
11:23:38:353 -> 2887
11:23:38:354 -> 2887
11:23:38:355 -> 2887
11:23:38:356 -> 2887
11:23:38:357 -> 2887
11:23:38:358 -> 2887
11:23:38:359 -> 2887
11:23:38:360 -> 2887
11:23:38:361 -> 2887
11:23:38:362 -> 2887
11:23:38:363 -> 2887
11:23:38:364 -> 2887
11:23:38:365 -> 2887
11:23:38:366 -> 2887
11:23:38:367 -> 2887
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-02-20:
If you use analogReadMilliVolts, you will get the number of millivolts using a [factory calibrated reference voltage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc_calibration.html#analog-to-digital-converter-adc-calibration-driver). Note that the ADC precision will improve if you use a voltage divider and lower attenuation level than the default (11db).

TD-er, 2025-02-20:
Yep totally agree with using calibration curves.
And to make it more clear what @lbernstone means to tell you...
Here the calibration curves of different attenuation factors plotted in the same chart (standard feature in my software, ESPEasy)

[image]

As you can see, the 12 dB attnuation isn't completely flat.
Also note that on ESP32-classic the input near 0 may not be reachable.
On other ESP32-variants this may differ.

For example the calibration curves on ESP32-C3 do behave nicely around 0 and are more straight, also on the highest attenuation:

[image]

Jason2866, 2025-03-12:
Closing. Behaviour is as expected and explained

closed 2025-03-12 by Jason2866