espressif/arduino-esp32 #11002: When will the arduino-esp32 V3 support Platform-espressif32?

hzhh110, 2025-02-22:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

When will the arduino-esp32 V3 support Platform-espressif32? Pure Arduino real compilers are too hard to use

vscode+PlatformIO+Arduino mode is much easier to use than Arduino development, but now only supports V2 version, and ESP32-S3+W5500 must use V3 version, otherwise it does not support... The Arduino compiler is really hard to get used to,MAC computers

### Describe the solution you'd like

arduino-esp32 V3 support Platform-espressif32

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2025-02-22:
Uh, platformio is downstream of here.
Anyhoo, you can use v3.x in platformio simply by pointing your platformio.ini to the right platform
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
...

Jason2866, 2025-02-22:
@hzhh110 Platformio is in no way related to espressif. It is Platformio decision not to support actual versions. I have done a fork `pioarduino` as Ibernstone already mentioned which adds support for Arduino core 3.x.x and can be used with Platformio

closed 2025-02-22 by Jason2866