espressif/arduino-esp32 #11004: I2S failed to set up tx callback

copych, 2025-02-23:
### Board

ESP32S3 (LOLIN S3 PRO)

### Device Description

LOLIN S3 PRO has just one native USB (OTG), no uart chip. it has 8MB PSRAM and 16MB flash.
Just the board and i2s DAC.

### Hardware Configuration

GPIOs 5(BIT), 6(DAT) and 7(WS)  are used to connect I2S DAC PCM5102.

### Version

v3.1.2

### IDE Name

Arduino IDE 2.3.4 , actually I can't select core v.3.1.3 

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I2S doesn't work

### Sketch

```cpp
#define I2S_BCLK_PIN    5       // I2S BIT CLOCK pin (BCL BCK CLK)
#define I2S_DOUT_PIN    6       // to I2S DATA IN pin (DIN D DAT) 
#define I2S_WCLK_PIN    7       // I2S WORD CLOCK pin (WCK WCL LCK)
const i2s_port_t i2s_num = I2S_NUM_0; // i2s port number

I2SClass I2S;

void i2sInit() {
  pinMode(I2S_BCLK_PIN, OUTPUT);
  pinMode(I2S_DOUT_PIN, OUTPUT);
  pinMode(I2S_WCLK_PIN, OUTPUT);

  I2S.setPins(I2S_BCLK_PIN, I2S_WCLK_PIN, I2S_DOUT_PIN); //SCK, WS, SDOUT, SDIN, MCLK
  I2S.begin(I2S_MODE_STD, SAMPLE_RATE, I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_STEREO);
}
```

### Debug Message

```plain
E (2005) gdma: gdma_register_tx_event_callbacks(464): user context not in internal RAM
E (2005) i2s_common: i2s_init_dma_intr(734): Register tx callback failed
E (2008) i2s_std: i2s_channel_init_std_mode(236): initialize dma interrupt failed
E (8983) i2s_common: i2s_channel_write(1211): The channel is not enabled
E (9983) i2s_common: i2s_channel_write(1211): The channel is not enabled
E (10983) i2s_common: i2s_channel_write(1211): The channel is not enabled
```

### Other Steps to Reproduce

it was working with core v.3.0.7

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

cniedzi, 2025-02-23:
See also: https://github.com/schreibfaul1/ESP32-audioI2S/issues/975

me-no-dev, 2025-03-07:
Fix incoming: https://github.com/espressif/esp32-arduino-lib-builder/pull/284
Will also report to ESP-IDF

closed 2025-03-07 by me-no-dev

cniedzi, 2025-03-07:
@me-no-dev Do you know in which ESP32-Arduino core this fix will be implemented?

me-no-dev, 2025-03-07:
3.2.0-RC2

cniedzi, 2025-03-07:
> 3.2.0-RC2

Thx. Great !!!

JetForMe, 2025-03-11:
> Will also report to ESP-IDF

Seems I2S should check to see if `CONFIG_GDMA_ISR_IRAM_SAFE` is set, right? It's not clear to me how it can work if `CONFIG_GDMA_ISR_IRAM_SAFE` is set but `CONFIG_I2S_ISR_IRAM_SAFE` is not.

me-no-dev, 2025-03-11:
Neither should need to be set for the driver to work. The problem is that default config with PSRAM is failing

JetForMe, 2025-03-11:
> Neither should need to be set for the driver to work. The problem is that default config with PSRAM is failing

It's likely I misunderstood, but looking at the code, the GDMA callback registration function checks to see if the supplied context parameter is in internal RAM when `CONFIG_GDMA_ISR_IRAM_SAFE` is set. But I2S doesn't allocate that context in internal RAM unless `CONFIG_I2S_ISR_IRAM_SAFE` is set. So you have two different build settings that must be set together (or rather, if `CONFIG_GDMA_ISR_IRAM_SAFE` is set, then `CONFIG_I2S_ISR_IRAM_SAFE` must also be set).

If I2S looked at `CONFIG_GDMA_ISR_IRAM_SAFE` strictly for the purposes of allocating the channel structure (what it passes as context to GDMA), I think it would be more robust to various configurations. It can still base its own callback checks on `CONFIG_I2S_ISR_IRAM_SAFE`.