espressif/arduino-esp32 #11006: Zigbee On/Off Switch Always Prints Endpoint 10 and Ignores Additional Endpoints on the Light Device

sebastian-ionita, 2025-02-23:
### Board

ESP32c6

### Device Description

2 x Esp32c6

### Hardware Configuration

Default configuration

### Version

v3.1.2

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When using a Zigbee On/Off Switch, it always prints "Endpoint 10", even if the On/Off Light device is configured with a different endpoint. Additionaly, if the On/Off Light device registers multiple endpoints (e.g., 10 and 11), the switch only sees one endpoint, instead of recognizing both.

### Sketch

```cpp
Light example: 

#define ZIGBEE_LIGHT_ENDPOINT 11
#define ZIGBEE_LIGHT_ENDPOINT2 12

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);
ZigbeeLight zbLight2 = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT2);

void setup() {
  Serial.begin(115200);

  // Init LED and turn it OFF (if LED_PIN == RGB_BUILTIN, the rgbLedWrite() will be used under the hood)
  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  // Init button for factory reset
  pinMode(button, INPUT_PULLUP);

  //Optional: set Zigbee device name and model
  zbLight.setManufacturerAndModel("Espressif", "ZBLightBulb");
  zbLight2.setManufacturerAndModel("Espressif", "ZBLightBulb2");

  // Set callback function for light change
  zbLight.onLightChange(setLED);
  zbLight2.onLightChange(setLED2);

  //Add endpoint to Zigbee Core
  Serial.println("Adding ZigbeeLight endpoint to Zigbee Core");
  Zigbee.addEndpoint(&zbLight);
  Zigbee.addEndpoint(&zbLight2);

  // When all EPs are registered, start Zigbee. By default acts as ZIGBEE_END_DEVICE
  if (!Zigbee.begin()) {
    Serial.println("Zigbee failed to start!");
    Serial.println("Rebooting...");
    ESP.restart();
  }
  Serial.println("Connecting to network");
  while (!Zigbee.connected()) {
    Serial.print(".");
    delay(100);
  }
  Serial.println("Connected");
  zbLight.printBoundDevices(Serial);
  zbLight2.printBoundDevices(Serial);
}

Switch Example is untouched.
```

### Debug Message

```plain
Switch example prints:
Device on endpoint 10, short address: 0x0, ieee address: 40:4c:ca:ff:fe:5e:d3:c0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-02-24:
Hi @sebastian-ionita, have you enabled in switch example that it can bound to multiple lights?
If yes (correct), its maybe possible that as it have the same ieee address its only bounded to the 1st endpoint. 
But you can use the commands for controlling the light in the switch, where you can specify the endpoint/address.

sebastian-ionita, 2025-02-24:
The light exmple has two endpoints, 11 and 12. Switch example binds to endpoint 10 which shouldn't exist, this is what it prints in the console: Device on endpoint 10, short address: 0x0, ieee address: 40:4c:ca:ff:fe:5e:d3:c0. Doesn't matter what i configure in light example, switch will always bind to endpoint 10

P-R-O-C-H-Y, 2025-02-25:
Can you turn on erase flash when uploading on both devices, to make sure there is nothing left in a flash from previous uploads?
You can enable it in the Arduino IDE->Tools menu->Erase Flash before upload–>Enabled.

Can you post please whole log from both Switch and Light?

sebastian-ionita, 2025-02-25:
Switch bound to endpoint 11 after I enabled Erase flash before upload, but it only sees one endpoint. 
Here are the logs from Switch:

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1228
load:0x4086c110,len:0xd9c
load:0x4086e610,len:0x2f74
entry 0x4086c110
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x420066e6
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x420066aa
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4200666e
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006632
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x420066e6
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x420066aa
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4200666e
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006632
[    91][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 5
[   103][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_RX (2) with bus 0x4080dfa0
[   113][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_TX (3) with bus 0x4080dfa0
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 0
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   447628 B ( 437.1 KB)
  Free Bytes        :   407752 B ( 398.2 KB)
  Allocated Bytes   :    32748 B (  32.0 KB)
  Minimum Free Bytes:   407752 B ( 398.2 KB)
  Largest Free Block:   376820 B ( 368.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 40 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
[... 185 lines cut ...]
[   773][I][ZigbeeCore.cpp:218] esp_zb_app_signal_handler(): Zigbee stack initialized
[   782][I][ZigbeeCore.cpp:224] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   791][I][ZigbeeCore.cpp:231] esp_zb_app_signal_handler(): Start network steering
Connecting to network
.......................[  3024][I][ZigbeeCore.cpp:289] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
.......................................[  6912][I][ZigbeeCore.cpp:282] esp_zb_app_signal_handler(): Joined network successfully (Extended PAN ID: 40:4c:ca:ff:fe:5e:d3:c0, PAN ID: 0x4db6, Channel:24, Short Address: 0xfa6e)
Connected
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   446140 B ( 435.7 KB)
  Free Bytes        :   379340 B ( 370.4 KB)
  Allocated Bytes   :    57976 B (  56.6 KB)
  Minimum Free Bytes:   379340 B ( 370.4 KB)
  Largest Free Block:   352244 B ( 344.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     8 : RMT_TX
     9 : GPIO
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============

sebastian-ionita, 2025-02-25:
Here is the complete code for the switch:
`// Copyright 2024 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates simple Zigbee light switch.
 *
 * The example demonstrates how to use Zigbee library to control a light bulb.
 * The light bulb is a Zigbee end device, which is controlled by a Zigbee coordinator (Switch).
 * Button switch and Zigbee runs in separate tasks.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 */

#ifndef ZIGBEE_MODE_ZCZR
#error "Zigbee coordinator mode is not selected in Tools->Zigbee mode"
#endif

#include "Zigbee.h"

/* Zigbee switch configuration */
#define SWITCH_ENDPOINT_NUMBER 5

#define GPIO_INPUT_IO_TOGGLE_SWITCH BOOT_PIN
#define PAIR_SIZE(TYPE_STR_PAIR)    (sizeof(TYPE_STR_PAIR) / sizeof(TYPE_STR_PAIR[0]))

typedef enum {
  SWITCH_ON_CONTROL,
  SWITCH_OFF_CONTROL,
  SWITCH_ONOFF_TOGGLE_CONTROL,
  SWITCH_LEVEL_UP_CONTROL,
  SWITCH_LEVEL_DOWN_CONTROL,
  SWITCH_LEVEL_CYCLE_CONTROL,
  SWITCH_COLOR_CONTROL,
} SwitchFunction;

typedef struct {
  uint8_t pin;
  SwitchFunction func;
} SwitchData;

typedef enum {
  SWITCH_IDLE,
  SWITCH_PRESS_ARMED,
  SWITCH_PRESS_DETECTED,
  SWITCH_PRESSED,
  SWITCH_RELEASE_DETECTED,
} SwitchState;

static SwitchData buttonFunctionPair[] = {{GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}};

ZigbeeSwitch zbSwitch = ZigbeeSwitch(SWITCH_ENDPOINT_NUMBER);

/********************* Zigbee functions **************************/
[... 237 lines cut ...]
  }
  Serial.println("Connected");
}

void loop() {
  // Checking button for factory reset
  if (digitalRead(button) == LOW) {  // Push button pressed
    // Key debounce handling
    delay(100);
    int startTime = millis();
    while (digitalRead(button) == LOW) {
      delay(50);
      if ((millis() - startTime) > 3000) {
        // If key pressed for more than 3secs, factory reset Zigbee and reboot
        Serial.println("Resetting Zigbee to factory and rebooting in 1s.");
        delay(1000);
        Zigbee.factoryReset();
      }
    }
    // Toggle light by pressing the button
    zbLight.setLight(!zbLight.getLightState());
  }
  delay(100);
}
`

P-R-O-C-H-Y, 2025-02-25:
Can you please format the last message so its clear to read it?

sebastian-ionita, 2025-02-25:
plese check now. Also, please note that I Updated onZbButton function on light example to send a comand to all endpoint. 
smh like this:
```
for(device in bound-devices) {
    zbSwitch.lightToggle(device->endpoint, device->ieee_addr);
}
```

P-R-O-C-H-Y, 2025-02-25:
I will setup the same devices and will give it a test and find out what is wrong. Please give me some time for that. Will keep you updated.

P-R-O-C-H-Y, 2025-05-06:
Hi @sebastian-ionita this issue should be fixed by PR #11316. Can you checkout that PR and test if your issue is solved? Thanks

closed 2025-05-27 by me-no-dev