espressif/arduino-esp32 #11013: ESP32 is crashing when initializing HSPI with SPIRAM enabled.

memese64, 2025-02-25:
### Board

ESP-WROOM-32 DevKit v1

### Device Description

Replaced flash memory and added SPIRAM.

### Hardware Configuration

PSRAM /CE (pin 1) > ESP32 GPIO 16
PSRAM SO (pin 2) > flash DO
PSRAM SIO[2] (pin 3) > flash WP
PSRAM SI (pin 5) > flash DI
PSRAM SCLK (pin 6) > ESP32 GPIO 17
PSRAM SIO[3] (pin 7) > flash HOLD
PSRAM Vcc (pin 8) > ESP32 VCC_SDIO

### Version

v3.1.1

### IDE Name

VSCode

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

After initializing the HSPI and calling ps_malloc() or malloc(), the ESP crashes.
VSPI is working.

### Sketch

```cpp
#include "Arduino.h"
#include "SPI.h"

//SPIClass spi(VSPI); // works
SPIClass spi(HSPI);

void setup(){
   delay(1000);
   Serial.begin(115200);
   spi.begin(14, 12, 13, 15);
}

void loop(){
   delay(1000);
   int* test = (int*) ps_malloc(100 * sizeof(int));	
   Serial.println(ESP.getPsramSize());
   Serial.println(ESP.getFreePsram());
   free(test);
}
```

### Debug Message

```plain
[    22][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    31][I][esp32-hal-psram.c:102] psramAddToHeap(): PSRAM added to the heap.
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : D0WD-Q5
  Revision          : 3.01
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000032
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   375592 B ( 366.8 KB)
  Free Bytes        :   335172 B ( 327.3 KB)
  Allocated Bytes   :    33124 B (  32.3 KB)
  Minimum Free Bytes:   329572 B ( 321.8 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  4194304 B (4096.0 KB)
  Free Bytes        :  4192124 B (4093.9 KB)
  Allocated Bytes   :        0 B (   0.0 KB)
  Minimum Free Bytes:  4192124 B (4093.9 KB)
  Largest Free Block:  4128756 B (4032.0 KB)
  Bus Mode          : QSPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  3072.0 KB, type:  APP, subtype: OTA_0
             spiffs : addr: 0x00310000, size:   896.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Feb 25 2025 13:49:48
  Compile Host OS   : windows
  ESP-IDF Version   : v5.3.2-282-gcfea4f7c98-dirty
  Arduino Version   : 3.1.1
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:JTAGAdapter=default,PSRAM=enabled,PartitionScheme=huge_app,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=16M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=debug,EraseFlash=none,ZigbeeMode=default
============ Before Setup End ============
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   375592 B ( 366.8 KB)
  Free Bytes        :   332540 B ( 324.7 KB)
  Allocated Bytes   :    35420 B (  34.6 KB)
  Minimum Free Bytes:   326940 B ( 319.3 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x400f6364  PS      : 0x00060535  A0      : 0x800f6280  A1      : 0x3ffb2170  
A2      : 0x3fbffff8  A3      : 0x400f62b8  A4      : 0x3ffb21d4  A5      : 0x3ffb21f0  
A6      : 0x3fc00000  A7      : 0x3fbffffc  A8      : 0xffffffff  A9      : 0x3ffb2150  
A10     : 0x00000001  A11     : 0xfffffffc  A12     : 0x00000000  A13     : 0x3ffb21d4  
A14     : 0x00ff0000  A15     : 0x0000abab  SAR     : 0x00000002  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40086e58  LEND    : 0x40086e63  LCOUNT  : 0x00000000  

Backtrace: 0x400f6361:0x3ffb2170 0x400f627d:0x3ffb2190 0x400f6301:0x3ffb21b0 0x400d9c0a:0x3ffb21d0 0x400d4726:0x3ffb2210 0x400d4cbd:0x3ffb2250 0x400d46c2:0x3ffb2270 0x40089e4d:0x3ffb2290

Core  0 register dump:
PC      : 0x40087822  PS      : 0x00060135  A0      : 0x800d937d  A1      : 0x3ffbd850  
A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x3f40d180  
A6      : 0x00000001  A7      : 0x00000160  A8      : 0x800d87da  A9      : 0x3ffbd810  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffbe464  A13     : 0x00000000  
A14     : 0x00060020  A15     : 0x3ffbe508  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008781f:0x3ffbd850 0x400d937a:0x3ffbd870 0x4008ae43:0x3ffbd890 0x40089e4d:0x3ffbd8b0

CPU is: ESP32
EC: QUOS, QUOU, REMS, or REMU divisor operand is zero

0x400f6517 tlsf_walk_pool /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c 777 
0x400f6514 tlsf_walk_pool /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/tlsf/tlsf.c 771 
0x400f63e5 multi_heap_get_info_impl /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c 422 (discriminator 1) 
0x400f6479 multi_heap_get_info /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c 390 
0x400d9d72 heap_caps_get_info /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c 375 
0x400d4866 _printMemCapsInfo(unsigned long, char const*) C:\Users\MeM\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/chip-debug-report.cpp 28 
0x400d4dfd printAfterSetupInfo() C:\Users\MeM\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/chip-debug-report.cpp 332 
0x400d4802 loopTask(void*) C:\Users\MeM\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\cores\esp32/main.cpp 61 
0x40089e4d vPortTaskWrapper /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c 139 

CPU is: ESP32
EC: QUOS, QUOU, REMS, or REMU divisor operand is zero

0x40087822 esp_cpu_wait_for_intr /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c 64 
0x4008781f xt_utils_wait_for_intr /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/xtensa/include/xt_utils.h 82 
inlined by  esp_cpu_wait_for_intr /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c 55 
0x400d94ba esp_vApplicationIdleHook /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/freertos_hooks.c 58 
0x4008ae43 prvIdleTask /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c 4353 (discriminator 1) 
0x40089e4d vPortTaskWrapper /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c 139
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2025-02-25:
Can you try to reset the GPIO pins before calling `spi.begin` ?
```c++
gpio_reset_pin((gpio_num_t)12);
gpio_reset_pin((gpio_num_t)13);
gpio_reset_pin((gpio_num_t)14);
gpio_reset_pin((gpio_num_t)15);
```

memese64, 2025-02-25:
@TD-er No, it didn't work.
Research shows a problem in esp32-hal-spi.c
```#elif CONFIG_IDF_TARGET_ESP32
  if (spi_num == HSPI) {

    //ps_malloc() works

    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI2_CLK_EN);

    //ps_malloc() crash

    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI2_RST);
  } else if (spi_num == VSPI) {
    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI3_CLK_EN);
    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI3_RST);
  } else {
    DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_SPI01_CLK_EN);
    DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_SPI01_RST);
  }
```

lbernstone, 2025-02-25:
Please explain your hardware changes in greater detail, since this works on a stock module.

memese64, 2025-02-25:
@lbernstone 
Replaced the standard flash memory (4mb) with w25q128 (16mb).
I soldered ESP-PSRAM64H on top of the flash memory.
These are all the changes.

memese64, 2025-02-26:
Maybe the problem is with the SPIRAM chip.
I moved the chip to another board.
The memory test does not pass.
I am waiting for another chip to make sure.

Jason2866, 2025-03-12:
@memese64 Any news? Without feedback the issue will be closed.