espressif/arduino-esp32 #11020: SD Card isn't working (IDFGH-14750)

DRIFTYY777, 2025-02-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Im tring to to use 2 seprate SPI bus for tft display and sdcard the problem is when I use SD card separately its works but dident when both and same goes to Display, even though they are on seprate lines why I'm facing this issue?

Hardware - custom hardware 
ESP32-S3-WROOM-1-N16R8
Sd card module
tft_display st7789v  (TFT_eSPI with lvgl)

pins config for display
// HSPI pin configuration
#define HSPI_MISO 13 // SCL or MISO
#define HSPI_MOSI 11 // SDA or MOSI
#define HSPI_CLK 12  // CLK
// Display pin configuration
#define DSP_RST 46                       // RESET
#define DSP_DC 14                        // DC
#define DSP_CS 10                        // CS
#define DSP_BACKLIGTH 3                  // Backlight
#define DSP_CLK_SPEED (60 * 1000 * 1000) // 60Mhz

pins config for sdCard
// VPSI pin configuration
#define VSPI_MOSI 41 // SDA or MOSI
#define VSPI_MISO 38 // SCL or MISO
#define VSPI_CLK 40  // CLK
// SD Card pin configuration
#define SD_CS 39
#define SD_SPEED 4000000 // 4Mhz

bool SD_CARD::sd_init()
{
    pinMode(SD_CS, OUTPUT);
    digitalWrite(SD_CS, 1);
    SD_MMC.setPins(VSPI_CLK, VSPI_MOSI, VSPI_MISO);
    if (!SD_MMC.begin("/sdcard", true))
    {
        Serial.println("Card Mount Failed");
        return false;
    }
    uint8_t cardType = SD_MMC.cardType();

    if (cardType == CARD_NONE)
    {
        Serial.println("No SD card attached");
        return false;
    }
}
E (2760) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (2760) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107
E (2761) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[  2978][E][SD_MMC.cpp:138] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
Card Mount Failed

when i digital print the all sd pins its show HIGH or 1 even though I init sd or not any fixes? 

some more details 
(Arduino framework or esp-idf 4.4.4 when I log the detail via serial)
[env:esp32-s3-devkitc-1]
platform = espressif32@6.3.0
board = esp32-s3-devkitc-1
framework = arduino
board_upload.maximum_size = 1572864
board_build.f_cpu = 240000000L
board_build.flash_freq = 120m
board_build.arduino.memory_type = dio_opi
board_build.flash_size = 16MB
upload_protocol = esptool
monitor_speed = 115200

igrr, 2025-02-27:
I've moved this to arduino-esp32 project since the code snippet uses Arduino APIs. If something needs to be fixed in IDF, please feel free to create a more specific issue there.

lbernstone, 2025-02-27:
SD_MMC uses SDIO, not SPI.  You say it works when you use them separately, but then provide code that shows only the SD_MMC and has the error?
Please test with v3.x
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
```

DRIFTYY777, 2025-02-28:
Again, sd card is working separately but there's new problem now esp32s3 crashing when init initalizing bodmer/TFT_eSPI@^2.5.43
with or without sd card, Im things its due to lvgl or psdram allocation but the suspect  is lcd driver
What i'm doing wrong have any idea?

#define TFT_MISO  13 
#define TFT_MOSI  11 
#define TFT_SCLK  12 

#define TFT_CS    10 
#define TFT_DC    14 
#define TFT_RST   46 
void setup()
{
    pinMode(3, OUTPUT);
    digitalWrite(3, HIGH);

    Serial.begin(115200);
    tft.begin();
    tft.setRotation(1);
    tft.fillScreen(TFT_BLACK);

    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.setTextSize(2);
    tft.setCursor(10, 10);
    tft.println("Hello World!");
}

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379946
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420030d7  PS      : 0x00060730  A0      : 0x820031b4  A1      : 0x3fca36a0  
A2      : 0x3fc98740  A3      : 0x00000000  A4      : 0x0000000d  A5      : 0x0000000b  
A6      : 0x000000ff  A7      : 0x60004000  A8      : 0x00000010  A9      : 0x08000000  
A10     : 0x3fc9a214  A11     : 0x019bfcc0  A12     : 0x00000001  A13     : 0x00004000  
A14     : 0x00000000  A15     : 0x019bfcc0  SAR     : 0x0000000e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420030d4:0x3fca36a0 0x420031b1:0x3fca36d0 0x420031bc:0x3fca36f0 0x42002019:0x3fca3710 0x42006bee:0x3fca3740 0x4037d53a:0x3fca3760

me-no-dev, 2025-02-28:
TFT driver probably does not have a fix for the SPI register offset and is trying to access the wrong address. Your SD card issue, as @lbernstone mentioned, is that you are not using the SD SPI driver, but instead you are using SD_MMC which is a very different thing. You need SD and not SD_MMC

DRIFTYY777, 2025-02-28:
Yes you'r correct but this is because I canged the  this platform = espressif32 @ 6.5.0
to this platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip

as mention by https://github.com/espressif/arduino-esp32/issues/11020#issuecomment-2688557314 
Soo I can use tft and sd at same time but this diden't fix my problem 

Then I switched back to my early conifgs... 
[env:esp32-s3-devkitc-1]
platform = espressif32 @ 6.5.0
board = esp32-s3-devkitc-1
framework = arduino
board_upload.maximum_size = 1572864
board_build.f_cpu = 240000000L
board_build.flash_freq = 120m
board_build.arduino.memory_type = dio_opi
board_build.flash_size = 16MB
upload_protocol = esptool
monitor_speed = 115200

And you know this is working by degrading the TFT_eSPI  version=2.5.43 to version=2.5.0

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   217][E][main.cpp:110] setup(): [Main] Memory Status:
 -SPI_RAM: 8386019 Bytes
 -INTERNAL_RAM: 270100 Bytes
 -DMA_RAM: 270100 Bytes

SD Card Type: SDSC
SD Card Size: 1910MB

By the way Im just tring to use 2 spi 
spi 1 -- display ST7789 and [RFID Reader/Writer RC522 SPI S50]
spi 2 -- sd card and [2.4GHz NRF24L01+PA+LNA SMA Wireless Transceiver]

if you have any recommendations feel free to share

DRIFTYY777, 2025-02-28:
Problem solved by just changing TFT_eSPI version to lower one like 2.5.43 to 2.5.0

closed 2025-02-28 by lucasssvaz