espressif/arduino-esp32 #11026: Partition scheme

Prathameshney11, 2025-03-01:
### Board

ESP32 Dev Module 

### Device Description

It is attached with load cell using HX711,128x64 LCD,SD Card module 

### Hardware Configuration

Using SPI pins for LCD and SD card module

### Version

latest master (checkout manually)

### IDE Name

VS Code Platform IO

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Still getting the watchdog storage issue error after using the huge app partition scheme for my 4mb ESP tye flash size used by my firmware is around 38% 

### Sketch

```cpp
Code is attached
```

### Debug Message

```plain
No msg
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-03-01:
Post code and error, please. [Decode the backtrace](https://github.com/me-no-dev/EspExceptionDecoder) if necessary.

Jason2866, 2025-03-12:
closing, since no feedback. Feel free to ask for reopen when this issue is still there and the needed Infos are provided.

closed 2025-03-12 by Jason2866