espressif/arduino-esp32 #11027: Linker error: multiple definition of `opendir' closedir mkdir readdir...

pschatzmann, 2025-03-01:
### Board

ESP32 Dev Module

### Device Description

Any ESP32 board

### Hardware Configuration

not relevant

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

not relevant

### Description

Starting from 3.1.0 the [libflac](https://github.com/pschatzmann/arduino-libflac) library produces the following linker errors

/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.cache/arduino/sketches/AE2C456CDBC3582B16D2EDCE905EB999/libraries/arduino-libflac/libFLAC/objs.a(metadata_iterators.c.o):(.literal.set_file_stats_+0x1c): warning: chmod is not implemented and will always fail
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x10): warning: readdir is not implemented and will always fail
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x14): warning: closedir is not implemented and will always fail
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libmbedx509.a(x509_crt.c.obj):(.literal.mbedtls_x509_crt_parse_path+0x8): warning: opendir is not implemented and will always fail
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `opendir':
/builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/libgloss/libnosys/dir.c:16: multiple definition of `opendir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:733: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `closedir':
/builds/idf/crosstool-NG/.build/src/newlib-git-7cdcb0b0/newlib/libc/include/dirent.h:76: multiple definition of `closedir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:805: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `mkdir':
/builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/libgloss/libnosys/dir.c:30: multiple definition of `mkdir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:818: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `readdir':
/builds/idf/crosstool-NG/.build/src/newlib-git-7cdcb0b0/newlib/libc/include/dirent.h:56: multiple definition of `readdir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:750: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `telldir':
/builds/idf/crosstool-NG/.build/src/newlib-git-7cdcb0b0/newlib/libc/include/dirent.h:73: multiple definition of `telldir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:776: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `seekdir':
/builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/libgloss/libnosys/dir.c:73: multiple definition of `seekdir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:789: first defined here
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2405/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32/no-rtti/libnosys.a(dir.o): in function `rewinddir':
/builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/libgloss/libnosys/dir.c:101: multiple definition of `rewinddir'; /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-489d7a2b-v1/esp32/lib/libvfs.a(vfs.c.obj):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:800: first defined here
collect2: error: ld returned 1 exit status

### Sketch

```cpp
#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecFLAC.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"

AudioInfo info(44100, 2, 16);
SineWaveGenerator<int16_t> sineWave( 32000);  // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound( sineWave); // Stream generated from sine wave
HexDumpOutput out(Serial);
EncodedAudioStream encoder(&out, new FLACEncoder()); // encode and write
StreamCopy copier(encoder, sound);     

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Warning);

  // Setup sine wave
  sineWave.begin(info, N_B4);

  // start encoder
  encoder.begin(info);

  Serial.println("Test started...");
}

void loop() { 
  copier.copy();
}
```

### Debug Message

```plain
does not apply
```

### Other Steps to Reproduce

Install the following libraries
- https://github.com/pschatzmann/arduino-audio-tools
- https://github.com/pschatzmann/arduino-libflac

Select the partition scheme hugh app!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pschatzmann, 2025-03-01:
This is related to [this idf issue](https://github.com/espressif/esp-idf/issues/14174) which has been closed!

Jason2866, 2025-03-02:
The IDF issue has been closed since a fix is committed to IDF branch master.

me-no-dev, 2025-03-02:
This is not an Arduino issue. You can test with 3.2.0-RC1 and see if maybe the IDF team has backported this to IDF 5.4. There is really nothing we can do on our side to resolve your problem

closed 2025-03-02 by me-no-dev

Jason2866, 2025-03-02:
The fix is backported to IDF 5.4 but **not** to IDF 5.3. Arduino core 3.1.x is based on IDF 5.3.2+

pschatzmann, 2025-03-02:
I was raising this issue because it was not working in 3.2.0-RC1 either!

Jason2866, 2025-03-02:
Your failing example is showing 3.1.x

pschatzmann, 2025-03-02:
It took me quite some effort to find the  release when this started to be broken !

Jason2866, 2025-03-02:
Your example does compile successfully with 3.2.0-rc1

pschatzmann, 2025-03-02:
Strange: I was sure that's what I tested first yesterday
I reinstalled  3.2.0-rc1 and I confirm that it is is working after correcting the new compiler errors

I appologize for my mistake!