espressif/arduino-esp32 #11030: unable to open a secure HTTPClient session when MDNS.begin() has been executed

multinet33, 2025-03-02:
### Board

LilyGo TTGO Display

### Device Description

Lilygo T DIsplay ESP 32

### Hardware Configuration

no hardware

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

macOS 15.0

### Flash frequency

240MHZ

### PSRAM enabled

yes

### Upload speed

460800

### Description

I'm uanble to execute a GET() (httpclient) on a server HTTPS using a Bearer token when and ONLY when i've executed prior to the GET() a MDNS.begin("espdevice");

If i comment the MDNS.begin, my GET (httpclient) with my token is working great

### Sketch

```cpp
1. mdns
MDNS.begin("espdevice");

2. GET that is in fail (search int httpCode = httpenphase.GET() below)
bool Enphase_get_7_JWT(void)
{
  HTTPClient httpenphase;
  WiFiClientSecure secureClient;
  bool retour = false;
  String url = "/404.html";
  url = String(EnvoyJ);
  String adr = String(configmodule.hostname);
  String port = String(configmodule.port);
  String debut;

  if (ENPHASE_DEBUG) 
    Serial.println("**** Enphase_get_7_JWT ****");
  // Ne redemmande pas le tocken a chaque fois
  if (TockenValide)
    return true;

  if (String(configmodule.port) == "443")
    debut = "https://";
  else
    debut = "http://";

  String fullurl = debut + adr + ":" + port + url;
  //String fullurl = debut+adr+url;
  if (ENPHASE_DEBUG) 
    Serial.println("Enphase contrôle token : " + fullurl + "<----");
    
  if (httpenphase.begin(fullurl))
  {
    if (ENPHASE_DEBUG) 
      Serial.println("Connexion réussie, préparation des headers (token etc)");
    httpenphase.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    httpenphase.setAuthorizationType("Bearer");
    httpenphase.setAuthorization(configmodule.token);
    httpenphase.addHeader("Accept-Encoding", "gzip, deflate, br");
    httpenphase.addHeader("User-Agent", "PvRouter/1.1.1");
    const char *headerkeys[] = {"Set-Cookie"};
    httpenphase.collectHeaders(headerkeys, sizeof(headerkeys) / sizeof(char *));
    httpenphase.setReuse(true);
    int httpCode = httpenphase.GET();
    if (ENPHASE_DEBUG) {
      Serial.println("Retour HTTPCode sur GET() : " + httpenphase.errorToString(httpCode));   
      Serial.println("Réponse Headers :");
      for (int i = 0; i < httpenphase.headers(); i++)
      {
        Serial.println(httpenphase.headerName(i) + ": " + httpenphase.header(i));
      }
    }
    // httpCode will be negative on error
    if (httpCode > 0)
    {
      if (ENPHASE_DEBUG) 
        Serial.println("JWT : code retour : " + String(httpCode));

      if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY)
      {
        retour = true;
        // Token valide
        Serial.println("JWT Enphase contrôle token : TOKEN VALIDE ");
        TockenValide = true;
        SessionId.clear();
        SessionId = httpenphase.header("Set-Cookie");
        if (SessionId.indexOf("sessionId") < 0)
        {
          retour = false;
          SessionId.clear();
          Serial.println("JWT Enphase contrôle token : PAS DE SESSION ID !!!");
          Serial.println("JWT SessionId : " + SessionId);
        }
        else
        {
          SessionId.remove(0, SessionId.indexOf("sessionId"));
          if (SessionId.indexOf(";"))
          {
            SessionId.remove(SessionId.indexOf(";"));
          }
          Serial.println("JWT Enphase contrôle token sessionID: " + SessionId);
        }
      }
      else
      {
        Serial.println("Enphase contrôle token : TOKEN INVALIDE !!!");
        nbErreurCtrlTocken++;
      }
    }
    else
    {
      Serial.println("JWT Connexion refusée - Error code : " + String(httpCode));
      nbErreurCtrlTocken++;
    }
  }
  else
    nbErreurCtrlTocken++;

  if (nbErreurCtrlTocken == 10)
  {
    Serial.println("JWT Trop d'erreur  : redemmarrage");
    ESP.restart();
  }

  httpenphase.end();
  return retour;
}
```

### Debug Message

```plain
connection refused (code -1 on httpenphase.GET();
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-03:
Please provide a minimal example (full sketch) that we can compile and run

lbernstone, 2025-03-03:
In particular, we need to know what these two lines resolve to:
```
  String adr = String(configmodule.hostname);
  String port = String(configmodule.port);
```
I would strongly recommend you do the name resolution asynchronously from the http request. I know my power monitor had issues with this because the wifi is very unstable.

multinet33, 2025-03-03:
Not sure to understand : my GET connection ti this server is OK when i comment this line : MDNS.begin("espdevice"); (so when i disabled all my MNDS functions)

I can try you recommandation : "CN must match the name presented in the url for TLS to work." but i don't know how to do it

my ESP32 has the 192.168.1.63 local IP it call https://192.168.1.44:443 via httpclient GET()
it work (with Bearer token) if the line MDNS.begin("espdevice"); is commented
it got a "connexion refused" if MDNS.begin("espdevice"); is not commented

that why I suspect a conflit with mDNS but i can be wrong

lbernstone, 2025-03-03:
I don't have an answer for you. I can't see what is going on with your network.  Turn on verbose core debugging and watch carefully.  There is a redirect or something that causes the name resolution or tls verification to fail. If that doesn't show anything, I would set up a wireshark to see the whole transaction and what the difference is b/w the two requests.

multinet33, 2025-03-03:
Ok

I've found the root cause 

I have a xtaskcreate that execute MDNS query et priority 5 every 10sec

I drop priority to the lowest and pinned tadk to core 0

Then my get is working well

This issue can be closed

Thanks to all of you

closed 2025-03-04 by me-no-dev