espressif/arduino-esp32 #11033: Disabling Core0WDT in SPIFFS.format() might be the wrong WDT

Bascy, 2025-03-03:
### Board

ESP32 Dev Module

### Device Description

Propriety board, based on ESP32 chip

### Hardware Configuration

not relevant

### Version

latest master (checkout manually)

### IDE Name

VSCode with PlatformIO and pioArduino platform

### Operating System

Windows 11

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

In our factory-firmware version of the firware we call SPIFFS.format() in a pretty early stage of the setup(). This causes a very large number of `E (28769) task_wdt: esp_task_wdt_reset(705): task not found` errors to occur, before the setup continues as normal. 

This is caused by the `disableCore0WDT()` call in SPIFFS.format().

Question 1: Why is this error occuring?
Question 2: What if I have a task runnign on Core 0 that calls SPIFFS.format(), shouldnt that task disable the core1 Idle watchdog then?

### Sketch

```cpp
void setup() {
  Serial.begin(576000);
  Serial.println("Starting...");
  SPIFFS.begin(true);

  #ifdef COMPILE_FACTORY_FIRMWARE
  SPIFFS.format();
  #endif
...
```

### Debug Message

```plain
E (28766) task_wdt: esp_task_wdt_reset(705): task not found
E (28767) task_wdt: esp_task_wdt_reset(705): task not found
E (28768) task_wdt: esp_task_wdt_reset(705): task not found
E (28769) task_wdt: esp_task_wdt_reset(705): task not found
E (28770) task_wdt: esp_task_wdt_reset(705): task not found
E (28771) task_wdt: esp_task_wdt_reset(705): task not found
E (28772) task_wdt: esp_task_wdt_reset(705): task not found
E (28773) task_wdt: esp_task_wdt_reset(705): task not found
E (28774) task_wdt: esp_task_wdt_reset(705): task not found
E (28775) task_wdt: esp_task_wdt_reset(705): task not found
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-03-04:
It is disabling the idle timer. While core1 is busy, it isn't feeding the idle timer on core 0, which causes the watchdog to fire.  
The error is there because older versions of the IDF did not have any method to check the wdt status.  That should be fixed with the [latest PR](https://github.com/espressif/arduino-esp32/commit/2fecc482b721b30085f2a9983aaf45f9d38cb064).

me-no-dev, 2025-03-04:
The logic is: Core 1 WDT is OFF in Arduino, Core 0 WDT is ON. Disable Core 0 WDT while formatting to prevent WDT restarts. On single-core chips, WDT is disabled in Arduino, so the code should not do anything. We will look into the `task not found` error and try to fix it.

Bascy, 2025-03-05:
@me-no-dev Thanks for the explanation. I didn't know that Core 1 WDT was switched off in Arduino.
Hope you will find the cause of the error

lbernstone, 2025-03-05:
@me-no-dev I think it is the loopWDT that generates these errors, but I can't really see how to disable it during the format.

joseitor, 2025-04-15:
Nothing coming? I'm having the same issue

me-no-dev, 2025-10-20:
The logs come from IDF and are caused from the fact that WDT is disabled, but the IDLE task is trying to feed it. Fixed in master by silencing the log while formatting

closed 2025-10-20 by me-no-dev