espressif/arduino-esp32 #11034: Multiple issues (Wrong boot mode detected (0x13)! The chip needs to be in download mode.) Cannot upload any sketch

Juli337-afk, 2025-03-03:
### Board

ESP32-WROOM-32E

### Device Description

Custom ESP32 Board

[image]

The first image is my custom board. I've based it off the adafruit feather huzzah which is shown in the picture below (official schematic). My plan was to controll a group of PWM fans with the board, hence the 12V rail (for the fans) Then a 12V to 5V converter to use the already existing 5V to 3.3V for the ESP. There will never be both the USB and the 12V PSU connected at the same time. Usb is only for programming. I've noticed that I messed up TX and RX connections but I've solved them on the board by cutting traces and soldering new ones.

[image]

### Hardware Configuration

Various ESP pins would be connected to different Fan cables, but at the moment everything is disconnected.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When I try to upload a blank skecth, I'll get the error "Wrong boot mode detected (0x13)! The chip needs to be in download mode."
The serial monitor shows:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffeba4,len:4
load:0x4009f000,len:3248
entry 0x4009f574
⸮OHAI⸮ets Jul 29 2019 12:21:46

However if I short GPIO0 to ground, the serial monitor will show:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Now if I try to upload a sketch, I'll get the error "No serial data recieved"

If while shorting GPIO0 I press the reset button, the monitor will show:

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

But again "No serial data recieved" will appear after trying to upload a sketch

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fffeba4,len:4
load:0x4009f000,len:3248
entry 0x4009f574
⸮OHAI⸮ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-03-04:
https://support.microsoft.com/en-us/windows/find-your-files-in-windows-5c7c8cfe-c289-fae4-f5f8-6b3fdba418d2
Run this from the command prompt

Juli337-afk, 2025-03-04:
> https://support.microsoft.com/en-us/windows/find-your-files-in-windows-5c7c8cfe-c289-fae4-f5f8-6b3fdba418d2 Run esptool from the command prompt. It is probably in %userprofile%/Arduino15/packages/esp32/tools/esptool-x.x.x

I get exactly the same errors I got on the serial monitor from Arduino IDE

Juli337-afk, 2025-03-04:
> Do you get anything if you run `esptool.exe chip_id`? It sounds like your RX is not connected.

I FIXED IT. It really was an issue with the RX being not properly connected. I resoldered the CP2104 AGAIN and that fixed the issue. It was a hassle because this chip has no legs, only bottom pads. Thank you.