espressif/arduino-esp32 #11036: Support for TVOC, eCO2, AQI readings

timiman, 2025-03-04:
### Related area

Extend carbondioxide device end point of zigbee

### Hardware specification

ESP32-C6 with Gravity ENS160 Air Quality board

### Is your feature request related to a problem?

Carbondioxide end device and its example supports reporting of CO2 value only.
In case of using a Gravity ENS160 attached via I2C to a ESP32-C6 the values of TVOC and AQI are also available from the ENS160.

### Describe the solution you'd like

Extend the carbondioxide end device (+example) to also report TVOC and AQI values through zigbee.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

michapr, 2025-03-05:
I would add to "wish list" "the air fine dust value" (PM100,PM10,PM5,PM2.5) with unit ppm

P-R-O-C-H-Y, 2025-03-05:
Hi, please take a look there in esp-zigbee-sdk file:
https://github.com/espressif/esp-zigbee-sdk/blob/6c4f704c1b296f3bb104f32537cf028f7a39986d/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h#L136-L193

Those are currently all available clusters implemented in esp-zigbee-sdk.

I see there the PM2.5, but no TVOC or AQU clusters.
Can you pleas point out the exact clusters? Thanks.

P-R-O-C-H-Y, 2025-03-05:
If there is not the clusters you are looking for, I may implement that in the esp-zigbee-sdk and after its merged there and new version gets released do implementation here in Zigbee library.

P-R-O-C-H-Y, 2025-03-05:
I see in the Zigbee Cluster Specification, there are a lot of **Concentration measurement** clusters not yet supported to the esp-zigbee-sdk.

[image]

timiman, 2025-03-05:
Indeed, AQI and TVOC identifiers do not exist on this list. I thought that they would be there as I've seen zigbee devices report them, like Aqara TVOC does. I hope they will be added down the road, if this is common way when updating cluster identifiers on the sdk.
Do Aqara and ESP32 follow any official-centralized zigbee cluster identifiers list or each one creates each own ?

P-R-O-C-H-Y, 2025-03-10:
> Do Aqara and ESP32 follow any official-centralized zigbee cluster identifiers list or each one creates each own ?

As Espressif use Zigbee 3.0 specification, so if Aqara uses the same (for 99% percent they does) then yes, they follow the official.

If you are willing to add any new clusters that are not supported yet in esp-zigbee-sdk, feel free to also open a feature request there :)

closed 2025-06-30 by P-R-O-C-H-Y