espressif/arduino-esp32 #11038: Zigbee: Function to get the response from gateway after report

michapr, 2025-03-04:
### Board

ESP32C6, ESP32H2

### Device Description

ESP32C6 dev kit

### Hardware Configuration

nothing more connected

### Version

v3.1.2

### IDE Name

Arduino IDE

### Operating System

Win 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When sending a report user must be sure that report was received at gateway.
For this the gateway is sending a response to the device.

I cannot find a method to check the response from Gateway - it can be interesting for switches - but also for some important sensor values.
It should therefore be possible to query the status and repeat the transmission if necessary.
(If it is not necessary as sample for some sensors, it is possible to save energy and end the waiting process prematurely - even if this is certainly not in line with the Zigbee standard... ;) ...)

Maybe this function is anywhere - but could not find it ;)

### Sketch

```cpp
Sample temperature sketch from 
https://github.com/espressif/arduino-esp32/issues/10746
(enabled all debugging, set delay before deep sleep to 900 for debugging)
```

### Debug Message

```plain
20:23:26.407 -> [  2030][V][ZigbeeTempSensor.cpp:58] setTemperature(): Updating temperature sensor value...
20:23:26.407 -> [  2031][D][ZigbeeTempSensor.cpp:60] setTemperature(): Setting temperature to 1000
20:23:26.407 -> [  2033][V][ZigbeeTempSensor.cpp:99] setHumidity(): Updating humidity sensor value...
20:23:26.407 -> [  2035][D][ZigbeeTempSensor.cpp:101] setHumidity(): Setting humidity to 1000
20:23:26.407 -> [  2036][V][ZigbeeTempSensor.cpp:80] reportTemperature(): Temperature report sent
20:23:26.407 -> [  2041][V][ZigbeeTempSensor.cpp:122] reportHumidity(): Humidity report sent
20:23:26.407 -> Reported temperature: 10.00°C, Humidity: 10.00%
20:23:26.407 -> Going to sleep now
20:23:26.669 -> [  2299][V][ZigbeeHandlers.cpp:393] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x402) with status 0x0
20:23:26.669 -> [  2310][V][ZigbeeHandlers.cpp:393] zb_cmd_default_resp_handler(): Received default response: from address(0x0), src_endpoint(1) to dst_endpoint(10), cluster(0x405) with status 0x0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-05:
Hi @michapr, I guess just from a quick look it's about implementing a callback for `zb_cmd_default_resp_handler()` in `ZigbeeEP ` class so it can be user in the user app code. I am correct or is there any other response handlers that can be used if you know :)

michapr, 2025-03-05:
Really?
I have no idea how to do this right now... of course, this may also have something to do with my experience...

I see that the `static esp_err_t zb_cmd_default_resp_handler(const esp_zb_zcl_cmd_default_resp_message_t *message)` return only `ESP_OK`, but not as sample the cluster number....

Maybe you could show here a code fragment how to use it? We should know the received status of each transmitted value.

P-R-O-C-H-Y, 2025-03-05:
Will post a PR with the callback implemented and If you would please you can give it a try and help with testing I will appreciate it :)

def1149, 2025-03-08:
I agree. We need some way to confirm data was delivered not just using a delay

michapr, 2025-03-14:
@P-R-O-C-H-Y is here also any progress? ;)

P-R-O-C-H-Y, 2025-03-14:
@michapr its in my backlog. Will start on this soon.

mister-pjr, 2025-04-05:
@P-R-O-C-H-Y  is there any progress on this ? Also interested.

def1149, 2025-07-11:
@P-R-O-C-H-Y Any progress on this? We need some type of data delivery confirmation for battery-powered devices.  Using too short a delay after the reporting values sometimes results in data not being delivered. I've had to increase the delay to ensure delivery of all data. This wastes a lot of battery.

P-R-O-C-H-Y, 2025-07-11:
Hi @def1149, sorry for no response in few months. But you for sure saw that there were other improvements and features added to the Zigbee library. I will start on this next week.

def1149, 2025-07-11:
@P-R-O-C-H-Y Thanks. I appreciate your work

michapr, 2025-07-26:
Can confirm that the call back function is working in most cases.
With Zigbee_Temp_Hum_Sensor_Sleepy.ino sample sketch and ESP32C6 I have got 634ms.

In some cases I have seen a "confirmation" - but wrong values were reported. As sample is the registration of endpoints was not finished/successful.
If the device is working without problems after registration the call back function is working fine.

michapr, 2025-07-26:
@P-R-O-C-H-Y : What do you mean with 
`delay(50);  // 50ms delay to avoid busy-waiting`
in 
`static void meausureAndSleep(void *arg)`  ?

I have tried with much smaller values to save energy (time) - and no problems. (I have sometimes reached values of < 510ms)
What can happen here, what problems can arise?

THANKS!