espressif/arduino-esp32 #11040: WIFI does not connect after ESP NOW communication

dloyt, 2025-03-05:
### Board

Olimex ESP32-DevKit-Lipo

### Device Description

Olimex DevKit LIPO on a bread bord. Only IO35 has a switch to GND.

### Hardware Configuration

IO35 has a switch to GND.

### Version

v3.0.7

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The board has 2 operating modes. 
In mode A (with IO35 pulled HIGH) the board boots up, sends 3 packets with ESP NOW communication and then goes to sleep for 3s. (to save battery)

In Mode B (with IO35 pulled LOW) the board boots up and connects to an Access Point (for OTA firmware updates, it also starts a web server for debugging purposes)

PROBLEM:
If the board has bootet in mode A once, it will no longer connect to the access point when it boots up in mode B.

In order to make it connect to the access point again, a new firmware must be downloaded and the Flash must be erased during download. Then the board will connect to the access point again. If the board boots up in mode A again, it will no longer connect to the access point in mode B.

[esp32_ANCHORXMTDEBUG.zip](https://github.com/user-attachments/files/19091580/esp32_ANCHORXMTDEBUG.zip)

### Sketch

```cpp
I have added the zip file of the sketch.
```

### Debug Message

```plain
14:04:53.370 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:53.874 -> Connecting to WiFi..
14:04:53.874 -> E (29137) wifi:sta is connecting, return error
14:04:54.427 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:54.831 -> Connecting to WiFi..
14:04:54.831 -> E (30138) wifi:sta is connecting, return error
14:04:55.458 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:55.857 -> Connecting to WiFi..
14:04:55.857 -> E (31139) wifi:sta is connecting, return error
14:04:56.459 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:56.874 -> Connecting to WiFi..
14:04:56.874 -> E (32140) wifi:sta is connecting, return error
14:04:57.462 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:57.872 -> Connecting to WiFi..
14:04:57.872 -> E (33141) wifi:sta is connecting, return error
14:04:58.517 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
14:04:58.835 -> Connecting to WiFi..
```

### Other Steps to Reproduce

I have tried on different HW but the behavior is the same. I cannot use 3.1.3 board package because I get TCP asserts from one or more of the libraries I am using (not included in this debug sketch). Therefore I have not tried on 3.1.3

3:23:15.102 -> assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-07:
This upcoming change fixes it https://github.com/espressif/arduino-esp32/pull/11052/files#diff-0aa87997f6f48b5397f969144b34ce0ad4da8e173aee854c1cf2e1bf69ecf3aaR599-R625

You should also be able to force it with 
```cpp
WiFi.mode(WIFI_STA);
esp_wifi_set_protocol( WIFI_IF_STA , WIFI_PROTOCOL_11B|WIFI_PROTOCOL_11G|WIFI_PROTOCOL_11N);
esp_wifi_start(); // update the protocol
```

dloyt, 2025-03-07:
I have added this line. Sometimes it boots up and connects to WIFI immediately. 

Sometimes it boots up, then reboots itself and then connects to WIFI. 
18:04:22.846 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

Sometimes it goes into the same loop as above and I have to manually reboot.

16:35:51.367 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
16:35:51.552 -> [WiFi-event] event: 13 -> Connected to access point
16:35:52.328 -> Connecting to WiFi..
16:35:52.365 -> [WiFi-event] event: 14 -> Disconnected from WiFi access point
16:35:52.533 -> [WiFi-event] event: 13 -> Connected to access point
16:35:53.363 -> Connecting to WiFi..

me-no-dev, 2025-03-07:
you should enable core debug level to Verbose (in the board menu) in order to see whah and why it's happening. With the changes I mentioned, your code works fine on my end.

dloyt, 2025-03-11:
I have been trying to reproduce the problem but suddenly it always connects to WIFI.
Thank you for your quick and excellent support.