espressif/arduino-esp32 #11041: ADC analogReadMilliVolts incorrect result

ACE1046, 2025-03-05:
### Board

ESP32 C3 supermini

### Device Description

ESP32 C3 supermini

### Hardware Configuration

0.314V at pin 0 (voltage divider 16to1 from 5V)

### Version

v3.1.1

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 10

### Flash frequency

160MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

analogReadMilliVolts returns incorrect value if pin attenuation is set.

### Sketch

```cpp
void setup() {
	Serial.begin(115200);
}

void loop() {
	Serial.print("IDF version: ");
	Serial.println(esp_get_idf_version());

	analogSetAttenuation(ADC_0db);
	Serial.print("Analog, mV (global 0  dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetAttenuation(ADC_2_5db);
	Serial.print("Analog, mV (global 2.5dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetAttenuation(ADC_6db);
	Serial.print("Analog, mV (global 6  dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetAttenuation(ADC_11db);
	Serial.print("Analog, mV (global 11 dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));

	analogSetPinAttenuation(A0, ADC_0db);
	Serial.print("Analog, mV (pin 0  dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetPinAttenuation(A0, ADC_2_5db);
	Serial.print("Analog, mV (pin 2.5dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetPinAttenuation(A0, ADC_6db);
	Serial.print("Analog, mV (pin 6  dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));
	analogSetPinAttenuation(A0, ADC_11db);
	Serial.print("Analog, mV (pin 11 dB): ");
	Serial.print(analogReadMilliVolts(A0));
	Serial.print(" Raw: ");
	Serial.println(analogRead(A0));

	delay(10000);
}
```

### Debug Message

```plain
IDF version: v5.3.2-282-gcfea4f7c98-dirty
Analog, mV (global 0  dB): 318 Raw: 1502
Analog, mV (global 2.5dB): 318 Raw: 1117
Analog, mV (global 6  dB): 320 Raw: 809
Analog, mV (global 11 dB): 316 Raw: 425
Analog, mV (pin 0  dB): 1109 Raw: 1513
Analog, mV (pin 2.5dB): 822 Raw: 1116
Analog, mV (pin 6  dB): 597 Raw: 811
Analog, mV (pin 11 dB): 317 Raw: 427
```

### Other Steps to Reproduce

**analogReadMilliVolts** function ignores attenuation set for single pin (**analogSetPinAttenuation**). Global attenuation (set by **analogSetAttenuation**) is used for calculation of calibrated value.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-06:
I have looked in the code, and this behavior seems to be correct due to the following:
- ADC calibration handle (used for readMillivolts) is configured for the whole ADC Unit, not for each separate pin. So if the analogSetPinAttenuation() will also update the Attenuation in the calibration handle, it will be done for whole unit and then if you read some other pin, it wont match the RAW value with Millivolts.

I will check the documentation and add this note if it's missing. But there is nothing to be "fixed"

ACE1046, 2025-03-06:
So, if individual pin attenuation is set where is no way to get correct calibrated ADC reading? That makes individual pin attenuation useless.
I think correct way is to remember attenuation for each pin. Or, at least, add an optional attenuation parameter to readMillivolts function to use in conversion.

P-R-O-C-H-Y, 2025-03-06:
The individual pin attenuation works for reading RAW value. But for readMillivolts you need to set for whole ADC unit by calling analogSetAttenuation which sets it all used pins and used calibration handles.

ACE1046, 2025-03-06:
So now you have only 2 options:
1. Use global attenuation and get calibrated values (or change attenuation each time before reading next channel).
2. Use per pin attenuaton and read raw data. In this case, you're can't calibrate it. Uncalibrated values are very unprecise on ESP32. If you don't need precision - you don't need to change attenuation. So, it looks to me, pin attenuation is almost useless.

As I can see, it should be relatively easy to add attenuation parameter to readMillivolts.

ACE1046, 2025-03-06:
Possible solution is to add another function with additional parameter.
```
uint32_t __analogReadMilliVoltsWithAttenuation(uint8_t pin, adc_attenuation_t attenuation) {
  int value = 0;
  adc_channel_t channel;
  adc_unit_t adc_unit;
  esp_err_t err = ESP_OK;

  adc_oneshot_io_to_channel(pin, &adc_unit, &channel);
  if (err != ESP_OK) {
    log_e("Pin %u is not ADC pin!", pin);
    return value;
  }

  if (attenuation != __analogAttenuation)
  {
    // saving old values
    adc_cali_handle_t old_adc_cali_handle = adc_handle[adc_unit].adc_cali_handle;
    adc_attenuation_t old_analogAttenuation = __analogAttenuation;

    adc_handle[adc_unit].adc_cali_handle = NULL;
    __analogAttenuation = attenuation;
    value = __analogReadMilliVolts(pin); // analogReadMilliVolts will create temporary scheme

    // deleting temporary cali scheme
#if ADC_CALI_SCHEME_CURVE_FITTING_SUPPORTED
    if (adc_handle[adc_unit].adc_cali_handle) adc_cali_delete_scheme_curve_fitting(adc_handle[adc_unit].adc_cali_handle);
#elif (!defined(CONFIG_IDF_TARGET_ESP32H2) && !defined(CONFIG_IDF_TARGET_ESP32P4))  //ADC_CALI_SCHEME_LINE_FITTING_SUPPORTED
    if (adc_handle[adc_unit].adc_cali_handle) adc_cali_delete_scheme_line_fitting(adc_handle[adc_unit].adc_cali_handle);
#endif

    // restoring old values
    adc_handle[adc_unit].adc_cali_handle = old_adc_cali_handle;
    __analogAttenuation = old_analogAttenuation;
  }
  else {
    value = __analogReadMilliVolts(pin);
  }
  return value;
}
```

P-R-O-C-H-Y, 2025-03-10:
> Possible solution is to add another function with additional parameter.
>
> ```
> uint32_t __analogReadMilliVoltsWithAttenuation(uint8_t pin, adc_attenuation_t attenuation) {
>   int value = 0; [...]

Have you tested this solution? :)

ACE1046, 2025-03-10:
Yes, it works on C3 just fine. Didn't check on other boards. Just copy pasted defines from __analogChannelConfig function. And I'm not sure about thread safety. Not sure if it safe to temporary change global variables __analogAttenuation.