espressif/arduino-esp32 #11044: ESP32C6 Zigbee Analog example - Crash

oli-the-kitty, 2025-03-06:
### Board

ESP32-C6 dev module

### Device Description

ESP32-C6-WROOM-1 module on a generic ebay development board with 2 USB-C ports.

### Hardware Configuration

There is no other hardware attached to the board.

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE v1.8.19

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I wanted to try out the Zigbee analog input output example. The code compiled okay, was flashed to the board, paired up with zigbee network (Home Assistant via Sonoff dongle) and crashed. Since then it keeps crashing as soon as it connects to the network. The example was completely unmodified.

### Sketch

```cpp
// Copyright 2025 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

/**
 * @brief This example demonstrates Zigbee analog input / output device.
 *
 * The example demonstrates how to use Zigbee library to create a end device analog device.
 *
 * Proper Zigbee mode must be selected in Tools->Zigbee mode
 * and also the correct partition scheme must be selected in Tools->Partition Scheme.
 *
 * Please check the README.md for instructions and more detailed description.
 *
 * Created by Jan Procházka (https://github.com/P-R-O-C-H-Y/)
 * Modified by Pat Clay
 */
[... 251 lines cut ...]
4081dbc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081dbe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: dd417934c
```

### Other Steps to Reproduce

Steps that led to this are:
- Connect ESP32C6 to the laptop
- Open the Analog Input Output example from the examples menu
- Compile and flash
- Add the device in Home Assistant

Here are my IDE settings

[image]

Tried this with an ESP32H2 and the same thing happened.
I'm using ZHA in HA and running Silicon Labs Multiprotocol firmware on the dongle.
I could not get the exception decoded using the decoder - I pasted it in and no output appeared.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MikaFromTheRoof, 2025-03-09:
I got exactly the same problem. Same Zigbee error "Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled." and same example I am using.

I found out, that everything works fine as soon as I comment out the line for the reporting configuration:

```C
zbAnalogDevice.setAnalogInputReporting(0, 30, 10);
```

but I have no clue why this function call doesn't work. I would really like to keep it and use the reporting configuration.

Someone any ideas?

oli-the-kitty, 2025-03-10:
I can confirm that commenting this line out does prevent the crash. Decided to dig a little deeper. 

In the Zigbee library, in the file `src/ep/ZigbeeAnalog.cpp`, in the function `setAnalogInputReporting` crash seems to occur on this line:
```c
esp_zb_zcl_update_reporting_info(&reporting_info);
```
Just a few lines above, where the `reporting_info` is being prepared, I found what looks like a copy-paste error. Changing the line
```c
reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
```
to
```c
reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ANALOG_INPUT;
```
has fixed the crashing and reporting seems to work okay for me

P-R-O-C-H-Y, 2025-03-10:
@oli-the-kitty Thank you for investigations and also for posting a fix. I am sorry for that copy-paste error.

closed 2025-03-10 by me-no-dev

MikaFromTheRoof, 2025-03-10:
> I can confirm that commenting this line out does prevent the crash. Decided to dig a little deeper.
> 
> In the Zigbee library, in the file `src/ep/ZigbeeAnalog.cpp`, in the function `setAnalogInputReporting` crash seems to occur on this line:
> 
> esp_zb_zcl_update_reporting_info(&reporting_info);
> 
> Just a few lines above, where the `reporting_info` is being prepared, I found what looks like a copy-paste error. Changing the line
> 
> reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_TEMP_MEASUREMENT;
> 
> to
> 
> reporting_info.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ANALOG_INPUT;
> 
> has fixed the crashing and reporting seems to work okay for me

May I ask you how you managed to test it with the changed line right away? I am guite new to github and would like to get it to work correctly in my Arduino IDE. Is that possible?

MikaFromTheRoof, 2025-03-11:
@oli-the-kitty nice, then welcome to you too :-)

I am desperately looking for a community of other people tinkering with ESP32, zigbee and the arduino-esp32 library. Did you find any yet? Or is there just Github? I would really love to go deeper into those topics, but the code examples are few and as soon as you deviate too far from them stuff stops working quite often. And whenever I need help, I don't know where I could turn to...
In addition to that I couldn't find any good documentation for the zigbee part of the arduino-esp32 library.

How are you testing your selfmade zigbee devices? I integrate them into my Home Assistant via Zigbee2MQTT and always wonder, if my code / the zigbee on the ESP32 or Zigbee2MQTT is the problem, whenever something isn't working properly. 
Which ESP32 / zigbee projects are you currently working on? How do you get along?

At the moment I am trying to combine two zigbee endpoints into one device: a dimmable light endpoint and a analog value endpoint to send the sensor values of a simple photoresistor (light measure). But I can't even get the analog value example code working... Zigbee2MQTT finds the endpoint, but does not receive or "understand" any values...

Apart from that, a quick question: do you know if there is a function for sending personal messages on Github?

oli-the-kitty, 2025-03-11:
Hi. As far as I know, github got rid of private messages. You can message me on Telegram if you wanna talk, being honest, I don't know myself if continuing a conversation in a closed issue is the right thing to do here XD

My name on Telegram is oli_the_cat

P-R-O-C-H-Y, 2025-03-11:
Folks you can also join our Discord, where you can also chat directly :) 
https://discord.com/invite/8xY6e9crwv

MikaFromTheRoof, 2025-03-11:
you're absolutely right, Prochy!
I am already over there.
@oli-the-kitty Add me in Discord, if you want to: mikafromtheroof

oli-the-kitty, 2025-03-12:
I'll look into joining once I get home from work, I need to be on PC for discord cuz the mobile apps are crap