espressif/arduino-esp32 #11046: ESP32 compilation error triggered by recommended by third part PSRAM type (IDFGH-14763)

baruchli, 2025-03-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Arduino IDE 2.3.4

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I use SparkFun Thing Plus - ESP32-S3 board with next SW environment:
Arduino IDE: 2.3.4
Installed boards support:
Arduino ESP32 Boards v2.0.18 - arduino.5 by Arduino
esp32 v3.2.0-RC1 by Espressif
Board chousen in Arduino IDE "Tools":  esp32 --> ESP32S3 Dev Module Octal (WROOM2)

If I choose PSRAM: QSPI PSRAM option according to [Hookup guide](https://docs.sparkfun.com/SparkFun_Thing_Plus_ESP32-S3/arduino_example/#board-settings).

I had tried to compile one of examples provided by Sparkfun [RGB_LED.ino](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/blob/main/Firmware/RGB_LED/RGB_LED.ino) 

Expected behaviour: Successful compilation 

### What is the actual behavior?

Actual behavior:
Next compilation error:

> In file included from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/sys/reent.h:34,
> from …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/reent.h:13,
> from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/stdio.h:60,
> from…\Arduino15\packages\esp32\hardware\esp32\3.2.0-RC1\cores\esp32/Arduino.h:27,
> from …\AppData\Local\arduino\sketches\F2D0A6956F0C0D1ADC2714C276F620B6\sketch\RGB_LED.ino.cpp:1:
> …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
> 9 | #include “sdkconfig.h”
> | ^~~~~~~~~~~~~
> compilation terminated 

### Steps to reproduce.

Install from scretche next environment:

1. Arduino IDE: 2.3.4
2. Installed boards support:
3. Arduino ESP32 Boards v2.0.18 - arduino.5 by Arduino
4. esp32 v3.2.0-RC1 by Espressif
5. Choose next board via Arduino IDE tools:  esp32 --> ESP32S3 Dev Module Octal (WROOM2)

- Import into Arduino IDE [FastLED library](https://github.com/FastLED/FastLED)
- Clone Spakrfun [Thing Plus ESP32-S3 repository](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3)
- Open [RGB_LED.ino](https://github.com/sparkfun/SparkFun_Thing_Plus_ESP32-S3/blob/main/Firmware/RGB_LED/RGB_LED.ino) file and try to compile it. You will get compilation error.

Update: The problem can be reproduced starting from version _3.1.0-RC1_ of Espressif esp32 board support library for Arduino IDE.and onward. It does not occur in the previous version, _3.0.7_.

### Build or installation Logs.

```plain
In file included from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/sys/reent.h:34,
from …\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/reent.h:13,
from …/Arduino15/packages/esp32/tools/esp-x32/2411/xtensa-esp-elf/include/stdio.h:60,
from…\Arduino15\packages\esp32\hardware\esp32\3.2.0-RC1\cores\esp32/Arduino.h:27,
from …\AppData\Local\arduino\sketches\F2D0A6956F0C0D1ADC2714C276F620B6\sketch\RGB_LED.ino.cpp:1:
…\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-bcb3c32d-v1\esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
9 | #include “sdkconfig.h”
| ^~~~~~~~~~~~~
compilation terminated
```

me-no-dev, 2025-03-07:
Something must have went wrong when you were installing the core. The file is there and compilation should be OK. Please uninstall both core versions and install 3.2.0-RC1 again

baruchli, 2025-03-10:
> Something must have went wrong when you were installing the core. The file is there and compilation should be OK. Please uninstall both core versions and install 3.2.0-RC1 again

Installed and uninstalled and cannot reproduce now. 
I will update you if problem will persist

closed 2025-03-12 by Jason2866

baruchli, 2025-03-23:
@Jason2866
The issue has returned. I have identified an additional parameter that triggers the problem.

In the Arduino IDE, the PSRAM setting must be specified as "QSPI PSRAM", which is [listed as mandatory by my ESP32-S3 board manufacturer](https://docs.sparkfun.com/SparkFun_Thing_Plus_ESP32-S3/arduino_example/#thing-plus-esp32-s3-github-repository).

Steps to reproduce the issue:
Arduino IDE version: 2.3.4

Installed board support: esp32 v3.2.0-RC1 or any version > 3.0.7 by Espressif

PSRAM setting in Arduino IDE: QSPI PSRAM

Selected board in Arduino IDE ("Tools" menu): esp32 → ESP32S3 Dev Module Octal (WROOM2)

I can provide a compilation log if needed. Would you prefer that I open a new issue for this problem?

Jason2866, 2025-03-23:
As you mentioned for S3 matching correct settings for Type of Flash and PSRAM has to be set.
Your description looks like you use QSPI PSRAM and use settings for OPI (`Selected board in Arduino IDE ("Tools" menu): esp32 → ESP32S3 Dev Module Octal (WROOM2)`)?

Honestly i can't follow you. Before providing logs. Please describe exactly your hardware setup and the choosen settings in ArduinoIDE menu.

baruchli, 2025-03-24:
@Jason2866,

Apologies if my previous description was unclear. I'll provide a detailed explanation from scratch.

Hardware:
- SparkFun Thing Plus - ESP32-S3

Software:

- Windows 11
- Arduino IDE 2.3.4
- Board support: "esp32 by Espressif Systems" v3.2.0-RC2
- Selected board in Arduino IDE (Tools → Board):
- "ESP32S3 Dev Module Octal (WROOM2)" (from installed board support)

Custom Board Settings (Configured via Arduino IDE → Tools Menu):

- USB Mode: Hardware CDC and JTAG
- USB CDC on Boot: Enabled
- Upload Mode: UART0 / Hardware CDC
- PSRAM: QSPI PSRAM

Issue:

- When I compile any sketch (Arduino program) in this environment, I get the following error:

_...packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-489d7a2b-v1\esp32s3/include/newlib/platform_include/sys/lock.h:9:10: fatal error: sdkconfig.h: No such file or directory
    9 | #include "sdkconfig.h"
      |          ^~~~~~~~~~~~~_

Findings:
From my testing, this error seems to be triggered by the PSRAM: QSPI PSRAM setting.

Any insights or suggestions on how to resolve this would be greatly appreciated!