espressif/arduino-esp32 #11054: RequestHandlersImpl.h contains code implementation which makes it hard to derive a class

btsimonh, 2025-03-07:
### Board

esp32

### Device Description

esp32

### Hardware Configuration

any

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

all

### Flash frequency

any

### PSRAM enabled

yes

### Upload speed

any

### Description

RequestHandlersImpl.h can only be included once in a project, because it's inclusion results in binary results.  If you include it again, then the linker complains.  RequestHandlersImpl.h is included by the library, and so can't be used in user code to modify the library's features.

Circumstances:
I wanted to add authentication to StaticRequestHandler by deriving a class from it, creating a new StaticRequestHandlerAuth.
This is not possible, and I have to derive my class from RequestHandler alone instead, and reproduce the full functionality of StaticRequestHandler in my own code as a workaround.

Resolution:
Suggest move all the function definitions (e.g. RequestHandler &RequestHandler::addMiddleware(Middleware *middleware) ) into a .cpp file, so enabling RequestHandlersImpl.h to be included more than once.

Reference:
https://github.com/arendst/Tasmota/discussions/22892#discussioncomment-12427499

### Sketch

```cpp
#include "details/RequestHandlersImpl.h"

main(){
}
```

### Debug Message

```plain
linker error complains of duplicated functions.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-07:
@mathieucarbou want to have a go at this?

mathieucarbou, 2025-03-07:
@btsimonh @me-no-dev for what is worth, when implementing the middleware layer, I found that the WebServer code is quite a mess, but I think Arduino Core team knows that and I think there should be a plan to rewrite it correctly at one point.

That being said, `RequestHandlersImpl.h` or `RequestHandler.h` are headers that exist since 2018 and they are both used for different things.

**`RequestHandlersImpl.h` is an implementation class so it should not be used or extended. Only `RequestHandler.h` can be used ** 

Arduino team just decided to have some implementation in a header. Done correctly, it works, look around at ArduinoJson and several other libs which are also doing that. But I agree that this header references external things, but it is ok in this case because this is an implementation, not to be used, and it works because it is included once by WebServer.cpp after WebServer.cpp.h. I don't like that either, but well, that's how it was done.

To solve your issue you have 2 options:

1. use middleware, that's the way to go now 
2. implement your own request handler

If you go with 2), **only include RequestHandler.h**! 
And If you decide to go this route, think about using **delegation and composition instead of inheritance**. That's a far better design. You can create inside your handler an instance of the static handler you need to wrap. **No need to inherit it.** This is even worse because the day Arduino Team will change its implementation your code can break.

btsimonh, 2025-03-07:
thankyou, I went with 2) because it was effectively very close to the original written before the Impl was updated (in the version used by TAS).  If it works, then it's only really a temporary solution until someone has time to look into the middleware style implementation; because as you highlight, it's brittle to changes in the arduino libraries.  I'll add a note back in the TAS issue that it needs more attention.

closed 2025-03-12 by Jason2866