espressif/arduino-esp32 #11072: Poor Range of Zigbee End Device

Yusuf-Tuerkyilmaz, 2025-03-11:
### Board

ESP32 C6 Dev Module

### Device Description

ESP32-C6-DevKitC-1

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am using the "Zigbee_Color_Dimmable_Light" Example, didn't modify it.

Pairing with the Sonoff USB 3.0 Zigbee dongle as coordinator only works in really close proximity.
The Esp32 is after about 1m not reachable anymore and HA can't send commands.

I found this answer on espressif/esp-zigbee-sdk issues: 
https://github.com/espressif/esp-zigbee-sdk/issues/375#issuecomment-2227782471

it says to modify the **esp_zb_secur_network_min_join_lqi_set()** value but I can't find it anywhere in the arduino files or headers.

Other commercially available zigbee devices work just fine across my whole flat - so I assume cautiously that it isn't my Zigbee Mesh.

Is this a known limitation or is maybe my hardware not suitable?

### Sketch

```cpp
provided example sketch
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-11:
Hi @Yusuf-Tuerkyilmaz, if you have Espressifs ESP32-C6-DevKitC-1 I can confirm I have no issues with connecting to the Sonoff USB 3.0 Zigbee dongle as coordinator through 2 thick walls. Can you specify the devout you use?

Yusuf-Tuerkyilmaz, 2025-03-11:
Hey @P-R-O-C-H-Y 
Thanks for the fast reply.

I am using an ESP32-C6-DevKitC-1 from (https://amzn.eu/d/a3QqluI)
I think it is made by "muselab" (?)

And a Sonoff USB 3.0 Zigbee dongle with the 
CC2652P+CP2102N Chips in it 
(https://amzn.eu/d/0z44W0N)

The dongle is connected to an old Intel NUC, which runs HA and I am using ZHA to pair devices.

P-R-O-C-H-Y, 2025-03-11:
I am using the SonOFF Zigbee dongle V2 (E model): https://sonoff.tech/product/gateway-and-sensors/sonoff-zigbee-3-0-usb-dongle-plus-p/ which is never and better. So can't tell how it acts with the old one personally, but I know that many other users have no issues.

And as you are not using Espressif official devkit, it may be problem with the antenna on the devout you bought. You should reach the manufacturer of the board.

closed 2025-03-11 by P-R-O-C-H-Y