espressif/arduino-esp32 #11076: Serial1 not working with 3.1.2 and 3.1.3

ricardourio, 2025-03-12:
### Board

ESP32-WROOM-32E

### Device Description

It´s a development board using ESP32-WROOM-32E with 01 relay

### Hardware Configuration

GPIO 32 and 33 are connected to a distance sensor (TOF) XKC-KL200 with serial output.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

1152000

### Description

Compiling with 3.1.1 it works fine, but with 3.1.2 and 3.1.3 doesn´t give me any serial output at monitor. When running with WTD, I got a reset related to Core 0 WTD, that is running the serial1 reading instructions. The code doesn´t has a break from serial1 reading loop if no read, but works fine with sensor, unless version is 3.1.2 or greater. 

### Sketch

```cpp
unsigned int distancia = 0;
bool Distancia_Lida = false;

int ContaRX;
byte dadosRX[10];

void setup() {
  Serial.begin(115200);
}

byte XORChecksum8(const byte *data) {
  byte value = 0;
  for (byte i = 1; i < 9; i++) {
    value ^= data[i];
  }
  return value;
}

void LeDistancia() { //ToDo : break if no reading

  Distancia_Lida = false;
  ContaRX = 0;
  Serial1.end();
  Serial1.begin(9600, SERIAL_8N1, 33, 32);

  while (Distancia_Lida == false) {
    if (Serial1.available()) {
      byte inByte = Serial1.read();

      if (ContaRX > 0) {
        ContaRX++;
        dadosRX[ContaRX] = inByte;

        if ((ContaRX == 2) && (dadosRX[2] != 51)) {
          ContaRX = 0;
          Serial1.flush();
        }

        if (ContaRX == 9) {
          ContaRX = 0;

          if ((dadosRX[1] == 98) && (dadosRX[2] == 51)) {

            if (XORChecksum8(dadosRX) == dadosRX[9]) {
              distancia = dadosRX[6] << 8;
              distancia |= dadosRX[7];
              distancia = distancia / 10;
              ContaRX = 0;
              Distancia_Lida = true;
              Serial1.end();
              break;
            }
          }
        }
      }

      if (inByte == 98) {  
        dadosRX[1] = inByte;
        ContaRX = 1;
        distancia = 0;
      }
    }
  }
}

void loop() {
  verificarPresenca();

  for (int i = 0; i < 10; i++) {
    delay(1);
  }
}

void verificarPresenca() {

  LeDistancia();

  Serial.print("D = ");
  Serial.println(distancia);

  Serial.println(F(" "));
}
```

### Debug Message

```plain
No output
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ricardourio, 2025-03-12:
I made some more tests with oscilloscope and could see that with 3.1.2 and 3.1.3 the RX signal from sensor is dead, like a pull-down, even I set it to input in the Setup section. With version 3.1.1 the RX signal is what the picture above shows.

SuGlider, 2025-03-13:
This is interesting. We have noticed some sort of issue moving from 3.1.1 to 3.1.2.
It seems to be related to some IDF change. It needs more investigation.

Jason2866, 2025-03-13:
Is the needed 1k pull up resistor installed? 

[image]

Jason2866, 2025-03-13:
Closing since not reproducable. Hardware setup issue.

closed 2025-03-13 by Jason2866

ricardourio, 2025-03-13:
Adding pinMode(33, INPUT_PULLUP) in the Setup section solved the issue, what probably 3.1.1 did before.

Jason2866, 2025-03-14:
Honestly imho it is better as it is now. The internal pull up is very high (weak). By enabling it hides wrong hardware designs. Faster speeds longer wires (traces) will result in poor (ugly) signals. This is way more harder to find as when it just does not work completely.
The fact of not working made me think of the hardware part. And this was right. The pull up was missing. From hardware design aspect for this setup, only enabling the intern pull up is not really a solution. It is way out of the recommended set up and is just luck it works. For sure it is not reliable.

ricardourio, 2025-03-14:
Jason2866, I'm working 28 years with electronics and I very rarely saw a pratical device working with active high RX, so for me automatic pull up makes more sense, and disabling it as an option. Maybe the resistor could have a important place when using long distance or high interference fields, but the picture only indicates that the sensor has no pull up circuit internally, and it need to be provided at processor side, what an ESP32 can do internally. The 1k resistor is a guarantee for interferences, but it depends on your ambient and impedances of your circuit.  For now, I have more then 1000 devices sold, that I didn't need to install a resistor, working only with pull up, and as you can see in the picture, the impedance of signal is high enough to pull up handle it, as it can reach 3.3v and has a square wave appearance.

Jason2866, 2025-03-14:
My lesson learned in hardware and software development, never assume something.
This is based on electronic hardware experience of over 40 years.