espressif/arduino-esp32 #11079: AP restarts instead of simply stop and network events are repeated

enriquedbelec, 2025-03-12:
### Board

ESP32 Dev module

### Device Description

ESP32-WROOM-32UE

### Hardware Configuration

No

### Version

v3.1.2

### IDE Name

VSCode

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi, I've discovered an issue that is not happening in old Arduino Core 2.X. When you start WiFi STA and AP at same time (tried also in the same channel), the AP reports events `AP_START`, `AP_STOP `and `AP_START` again. Also, if you manage to stop only the AP, it reports `AP_STOP`, `AP_START`, and `AP_STOP`. These duplicate events were not reported in Arduino Core 2.X by doing the same.

In `stopAP()` function, if I try to call to `WiFi.softAPdisconnect(false);` (with false), it reports `AP_STOP`, `AP_START`, so the AP is restarted and is never stopped.

If I don't start the STA by calling WiFi.begin(...), when it manages to start at setup, it reports `AP_START`, `AP_STOP `and `AP_START`. When it manages to stop, the AP also reports the events `AP_STOP`, `AP_START`, so is not stopped, or the events are a mess.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

const char *ssid = "your-ssid";
const char *password = "your-pasword";

const char *ssidAP = "ESP32";
const char *passwordAP = "12345678";

void networkEvent(WiFiEvent_t event) {
    switch (event) {
      case ARDUINO_EVENT_WIFI_READY:               log_i("WiFi interface ready"); break;
      case ARDUINO_EVENT_WIFI_STA_START:           log_i("WiFi STA started"); break;
      case ARDUINO_EVENT_WIFI_STA_STOP:            log_i("WiFi STA stopped"); break;
      case ARDUINO_EVENT_WIFI_STA_CONNECTED:       log_i("WiFi STA connected"); break;
      case ARDUINO_EVENT_WIFI_STA_DISCONNECTED:    log_i("WiFi STA disconnected"); break;
      case ARDUINO_EVENT_WIFI_STA_AUTHMODE_CHANGE: log_i("WiFi STA authentication mode of access point has changed"); break;
      case ARDUINO_EVENT_WIFI_STA_GOT_IP:
        log_i("WIFI STA obtained IP address: %s", WiFi.localIP().toString().c_str());
        break;
      case ARDUINO_EVENT_WIFI_STA_LOST_IP:        log_i("WIFI STA lost IP address and IP address is reset to 0"); break;
     
      case ARDUINO_EVENT_WIFI_AP_START:           log_i("WiFi AP started"); break;
      case ARDUINO_EVENT_WIFI_AP_STOP:            log_i("WiFi AP stopped"); break;
      case ARDUINO_EVENT_WIFI_AP_STACONNECTED:    log_i("Client connected to AP"); break;
      case ARDUINO_EVENT_WIFI_AP_STADISCONNECTED: log_i("Client disconnected from AP"); break;
      case ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED:   log_i("AP assigned IP address to client"); break;
      case ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED:  log_i("AP received probe request"); break;
      default:                                    break;
    }
}

void startAP(){
    int apChannel = WiFi.channel();
    int maxAPClients = 1;
    log_d("Let's start AP %s in channel %d...", ssidAP, apChannel);
    WiFi.softAP(ssidAP, passwordAP, apChannel, 0, maxAPClients);
}

void stopAP(){
    log_d("Let's stop AP...");
    WiFi.softAPdisconnect(true);
}

void setup() {
    delay(3000);
    Serial.begin(115200);
    
    // Listen for events
    Network.onEvent(networkEvent);

    // Do not save nothing automatically: we handle saving the credentials
    WiFi.persistent(false);

    // Begin STA
    WiFi.begin(ssid, password);

    // Better we manage the autoreconnect policy
    WiFi.setAutoReconnect(false);

    log_d("Starting AP...");

    // Start AP
    startAP();
}

void loop() {

    // We manage to stop and start the AP at regular intervals, to observe the events produced
    delay(15000);

    if(WiFi.AP.started()){
        stopAP();
    }else{
        startAP();
    }
}
```

### Debug Message

```plain
[  3537][I][main.cpp:12] networkEvent(): WiFi interface ready
[  3614][I][main.cpp:13] networkEvent(): WiFi STA started
[  3623][D][main.cpp:61] setup(): Starting AP...
[  3633][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[  3653][I][main.cpp:23] networkEvent(): WiFi AP started
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------  
  Total Size        :   336772 B ( 328.9 KB)
  Free Bytes        :   239476 B ( 233.9 KB)
  Allocated Bytes   :    87008 B (  85.0 KB)
[  3682][I][main.cpp:15] networkEvent(): WiFi STA connected
  Minimum Free Bytes:   236224 B ( 230.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
[  3927][I][main.cpp:19] networkEvent(): WIFI STA obtained IP address: 192.168.1.143
[ 18740][D][main.cpp:41] stopAP(): Let's stop AP...
[ 18750][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 18756][I][main.cpp:23] networkEvent(): WiFi AP started
[ 18762][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 33770][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[ 33783][I][main.cpp:23] networkEvent(): WiFi AP started
[ 33981][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 33987][I][main.cpp:23] networkEvent(): WiFi AP started
[ 48978][D][main.cpp:41] stopAP(): Let's stop AP...
[ 48988][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 48994][I][main.cpp:23] networkEvent(): WiFi AP started
[ 49000][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 64007][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[ 64020][I][main.cpp:23] networkEvent(): WiFi AP started
[ 64219][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 64225][I][main.cpp:23] networkEvent(): WiFi AP started
[ 79215][D][main.cpp:41] stopAP(): Let's stop AP...
[ 79225][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 79231][I][main.cpp:23] networkEvent(): WiFi AP started
[ 79237][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 94244][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[ 94257][I][main.cpp:23] networkEvent(): WiFi AP started
[ 94455][I][main.cpp:24] networkEvent(): WiFi AP stopped
[ 94461][I][main.cpp:23] networkEvent(): WiFi AP started
[109452][D][main.cpp:41] stopAP(): Let's stop AP...
[109462][I][main.cpp:24] networkEvent(): WiFi AP stopped
[109468][I][main.cpp:23] networkEvent(): WiFi AP started
[109474][I][main.cpp:24] networkEvent(): WiFi AP stopped
[124481][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[124494][I][main.cpp:23] networkEvent(): WiFi AP started
[124692][I][main.cpp:24] networkEvent(): WiFi AP stopped
[124698][I][main.cpp:23] networkEvent(): WiFi AP started
[139689][D][main.cpp:41] stopAP(): Let's stop AP...
[139699][I][main.cpp:24] networkEvent(): WiFi AP stopped
[139705][I][main.cpp:23] networkEvent(): WiFi AP started
[139711][I][main.cpp:24] networkEvent(): WiFi AP stopped
[154718][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[154731][I][main.cpp:23] networkEvent(): WiFi AP started
[154929][I][main.cpp:24] networkEvent(): WiFi AP stopped
[154935][I][main.cpp:23] networkEvent(): WiFi AP started
[169926][D][main.cpp:41] stopAP(): Let's stop AP...
[169936][I][main.cpp:24] networkEvent(): WiFi AP stopped
[169942][I][main.cpp:23] networkEvent(): WiFi AP started
[169948][I][main.cpp:24] networkEvent(): WiFi AP stopped
[184955][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[184970][I][main.cpp:23] networkEvent(): WiFi AP started
[185167][I][main.cpp:24] networkEvent(): WiFi AP stopped
[185173][I][main.cpp:23] networkEvent(): WiFi AP started
[200164][D][main.cpp:41] stopAP(): Let's stop AP...
[200174][I][main.cpp:24] networkEvent(): WiFi AP stopped
[200180][I][main.cpp:23] networkEvent(): WiFi AP started
[200186][I][main.cpp:24] networkEvent(): WiFi AP stopped
[215193][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[215206][I][main.cpp:23] networkEvent(): WiFi AP started
[215404][I][main.cpp:24] networkEvent(): WiFi AP stopped
[215410][I][main.cpp:23] networkEvent(): WiFi AP started
[230401][D][main.cpp:41] stopAP(): Let's stop AP...
[230411][I][main.cpp:24] networkEvent(): WiFi AP stopped
[230417][I][main.cpp:23] networkEvent(): WiFi AP started
[230423][I][main.cpp:24] networkEvent(): WiFi AP stopped
[245430][D][main.cpp:36] startAP(): Let's start AP ESP32 in channel 6...
[245443][I][main.cpp:23] networkEvent(): WiFi AP started
[245641][I][main.cpp:24] networkEvent(): WiFi AP stopped
[245647][I][main.cpp:23] networkEvent(): WiFi AP started
[260638][D][main.cpp:41] stopAP(): Let's stop AP...
[260648][I][main.cpp:24] networkEvent(): WiFi AP stopped
[260654][I][main.cpp:23] networkEvent(): WiFi AP started
[260660][I][main.cpp:24] networkEvent(): WiFi AP stopped
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-12:
The reason why you see the calls is:
- When you start AP, you are first starting the interface with the default configuration, then you are resetting that configuration to your needs. That requires AP to be stopped and restarted with the new config.
- When you disconnect AP you first delete the config first, which causes the interface to restart with the default config and then the interface is turned off.

```cpp
void startAP(){
    int apChannel = WiFi.channel();
    int maxAPClients = 1;
    log_d("Let's start AP %s in channel %d...", ssidAP, apChannel);
    WiFi.AP.begin();
    WiFi.AP.create(ssidAP, passwordAP, apChannel, 0, maxAPClients);
}

void stopAP(){
    log_d("Let's stop AP...");
    WiFi.AP.end();
}
```

enriquedbelec, 2025-03-12:
Thank you for your response @me-no-dev. Is there a way to avoid this double stop/start or viceversa and avoid triple events, when the effect is only one event? In Arduino Core 2.X this does not happen.