espressif/arduino-esp32 #11121: Don't use begin() method to just change baudrate

GitHubLionel, 2025-03-13:
> Ok, in that case, we need a warning for the `begin()` method to say that if you just want to change the Baudrate you must use `updateBaudRate() ` method even if `uartBegin()` seems to allow this change. 

 _Originally posted by @GitHubLionel in [#10314](https://github.com/espressif/arduino-esp32/issues/10314#issuecomment-2721850568)_

GitHubLionel, 2025-03-13:
But I think that it will be better to correct the code in `esp32-hal-uart.c` since `uartBegin()` method should be do that.

SuGlider, 2025-03-13:
Up on the Arduino specification, `Serial.begin(baud_rate)` must correctly change the baud rate.
Therefore, I consider it as a bug.

I'll get a PR to fix it.

SuGlider, 2025-03-13:
@GitHubLionel - please test PR #11122 and let me know.

GitHubLionel, 2025-03-14:
Hi, 
No, not work because MUTEX is misplaced.
You should replace this code:
```
      UART_MUTEX_LOCK();
      //User may just want to change some parameters, such as baudrate, data length, parity, stop bits or pins
      if (uart->_baudrate != baudrate) {
        if (!uartSetBaudRate(uart, baudrate)) {
          log_e("UART%d changing baudrate failed.", uart_nr);
          retCode = false;
        } else {
          log_v("UART%d changed baudrate to %d", uart_nr, baudrate);
          uart->_baudrate = baudrate;
        }
      }
```
By this one:
```
      //User may just want to change some parameters, such as baudrate, data length, parity, stop bits or pins
      if (uart->_baudrate != baudrate) {
      	retCode = uartSetBaudRate(uart, baudrate);
      }
      UART_MUTEX_LOCK();
```

SuGlider, 2025-03-14:
@GitHubLionel - Thanks for testing and reviewing. All good now. This issue shall be closed when the PR is merged.

PR #11122 (2025-03-17): fix(UART): sets the correct uart clock source when using begin(baudrate)
## Description of Change
Arduino HardwareSerial was changed to use XTAL source clock whenever possible. This assures that the UART port doesn't chage its baudrate when APB frequency changes. The exception is related to ESP32 and ESP32-S2 that have no such clock source option.

In that case, ESP32|ESP32-S2 will use REF_TICK as clock source when the baudrate is lower than 250,000.
For a higher baudrate, it will use APB as clock source, which of course, may change the baudrate if its frequency is changed.

This PR fixes an i [... cut]
closed 2025-03-17 by me-no-dev