espressif/arduino-esp32 #11127: arduino-esp32 is using a compromised tj-actions/changed-files GitHub action (CVE-2025-30066)

eslerm, 2025-03-15:
### Board

a

### Device Description

arduino-esp32 uses a compromised version of tj-actions/changed-files. The compromised action appears to leak secrets the runner has in memory.

The action is included in:
  - https://github.com/espressif/arduino-esp32/blob/2a3de9c41533c0559e7c16ea34f4315e5fc2205d/.github/workflows/push.yml

Output of an affected runs:
  - https://github.com/espressif/arduino-esp32/actions/runs/13864794849/job/38801411946#step:3:91

Please review.

Learn about the compromise on [StepSecurity](https://www.stepsecurity.io/blog/harden-runner-detection-tj-actions-changed-files-action-is-compromised) of [Semgrep](https://semgrep.dev/blog/2025/popular-github-action-tj-actionschanged-files-is-compromised/).

This issue has been assigned [CVE-2025-30066](https://nvd.nist.gov/vuln/detail/CVE-2025-30066)

### Hardware Configuration

asd

### Version

latest master (checkout manually)

### IDE Name

a

### Operating System

d

### Flash frequency

a

### PSRAM enabled

yes

### Upload speed

d

### Description

sad

### Sketch

```cpp
asd
```

### Debug Message

```plain
sad
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

eslerm, 2025-03-15:
Note that https://github.com/espressif/pytest-embedded also uses this action.

me-no-dev, 2025-03-16:
@eslerm thanks! we are aware and actions are currently disabled

PR #11130 (2025-03-17): ci(actions): Replace changed-files
## Description of Change

Replace compromised `changed-files` with native solutions

## Tests scenarios

CI, local and fork.

## Related Issues

Closes #11127
closed 2025-03-17 by me-no-dev