espressif/arduino-esp32 #11134: BLE Client example causing corruption of characteristic UUID

pepassaco, 2025-03-17:
### Board

ESP32-S3-MINI dev board

### Device Description

Custom PCB design (barebones ESP32 module with 3.3V supply and USB for serial debugging. Nothing else attached

### Hardware Configuration

Nothing connected but power supply and USB for serial debugging

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

QIO 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

My BLE device (server) has a custom service with 3 custom characteristics, each with a different UUID and different handle. Characteristics are of the form 000000140x-0000-0000-0000-000000000000. If I use an Android app (nrf scan) or my PC with the NRF24 dongle, I can discover them and notify/read them. However, the ESP32 with the stock Client example (just by editing the UUID of the service and characteristic) is not able to find them. Moreover, at every iteration it finds 3 characteristics in the desired service, with the same UUID (which seems random or corrupted) and some handles and descriptors (which are indeed correct, the only glitchy field is the characteristic UUID).

### Sketch

```cpp
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001600-0000-0000-0000-000000000000");
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID("00001400-0000-0000-0000-000000000000");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic *pRemoteCharacteristic;
static BLEAdvertisedDevice *myDevice;

static void notifyCallback(BLERemoteCharacteristic *pBLERemoteCharacteristic, uint8_t *pData, size_t length, bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
[... 176 lines cut ...]
[ 12494][D][BLERemoteService.cpp:177] retrieveCharacteristics(): Found a characteristic: Handle: 13, UUID: d8000605-3042-003a-4100-000000000000
[ 12496][V][BLERemoteCharacteristic.cpp:34] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 13 0x13, uuid: d8000605-3042-003a-4100-000000000000
[ 12498][V][BLERemoteCharacteristic.cpp:268] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: d8000605-3042-003a-4100-000000000000
[ 12500][D][BLERemoteCharacteristic.cpp:295] retrieveDescriptors(): Found a descriptor: Handle: 14, UUID: 00002902-0000-1000-8000-00805f9b34fb
[ 12501][V][BLERemoteCharacteristic.cpp:305] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[ 12502][V][BLERemoteCharacteristic.cpp:44] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[ 12504][D][BLERemoteService.cpp:177] retrieveCharacteristics(): Found a characteristic: Handle: 16, UUID: d8000605-3042-003a-4100-000000000000
[ 12505][V][BLERemoteCharacteristic.cpp:34] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 16 0x16, uuid: d8000605-3042-003a-4100-000000000000
[ 12507][V][BLERemoteCharacteristic.cpp:268] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: d8000605-3042-003a-4100-000000000000
[ 12509][D][BLERemoteCharacteristic.cpp:295] retrieveDescriptors(): Found a descriptor: Handle: 17, UUID: 00002902-0000-1000-8000-00805f9b34fb
[ 12511][V][BLERemoteCharacteristic.cpp:305] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[ 12512][V][BLERemoteCharacteristic.cpp:44] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[ 12513][D][BLERemoteService.cpp:177] retrieveCharacteristics(): Found a characteristic: Handle: 19, UUID: d8000605-3042-003a-4100-000000000000
[ 12515][V][BLERemoteCharacteristic.cpp:34] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 19 0x19, uuid: d8000605-3042-003a-4100-000000000000
[ 12516][V][BLERemoteCharacteristic.cpp:268] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: d8000605-3042-003a-4100-000000000000
[ 12518][E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
[ 12519][V][BLERemoteCharacteristic.cpp:305] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[ 12520][V][BLERemoteCharacteristic.cpp:44] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[ 12521][V][BLERemoteService.cpp:190] retrieveCharacteristics(): << getCharacteristics()
Failed to find our characteristic UUID: 00001400-0000-0000-0000-000000000000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-03-18:
ESP32-S3 works with BLE 5. It will not connect to BLE 4.1 or Classic BT 3.0 servers.

pepassaco, 2025-03-18:
Sorry if I misunderstood your message, but is this screenshot from

pepassaco, 2025-03-18:
Sorry if I misunderstood your message, but is this screenshot from NRF Connect enough?

[image]

I mistakenly closed the issue before sorry

pepassaco, 2025-03-18:
Interestingly, if I make my code filter the (corrupted) found characteristics by handle, I can perfectly interact with the device by writing commands and receiving notifications. But the UUIDs are definitely not working

SuGlider, 2025-03-18:
> Interestingly, if I make my code filter the (corrupted) found characteristics by handle, I can perfectly interact with the device by writing commands and receiving notifications. But the UUIDs are definitely not working

By corrupted, you mean UUID d8000605-3042-003a-b700-000000000000, for instance?
Note that the BLE server may advertise a different UUID than the one that is actually used for connection.

nRF Connect APP has a Scan first Page that allow to see the advertised information while scanning the Server.
Could you send a screenshot of it?

pepassaco, 2025-03-18:
> By corrupted, you mean UUID d8000605-3042-003a-b700-000000000000, for instance?

Yes

> Note that the BLE server may advertise a different UUID than the one that is actually used for connection.

But is it normal that any other device sees as 3 different characteristic UUIDs (000000140x-000...) while the ESP32 sees 3 characteristics with the same UUID? (and if I try to search for the "correct" char. UUIDs no characteristic is found).

> Could you send a screenshot of it?

This?

[image]

pepassaco, 2025-03-18:
> Please tap it and send a screenshot

(Didn't know about that, sorry!) 

[image]

SuGlider, 2025-03-18:
Please tap [RAW] button to see the the advertising raw data. If possible also send that screenshot.
We will have all necessary information from the BLE Server.

SuGlider, 2025-03-18:
From the data, it seems to be a BLE 4.1 device. As I said, ESP32-S3 only works well with BLE 5.
Maybe the ESP32 (old SoC) that has support for BLE 4.1 will be able to connect correctly using the same UUIDs.

pepassaco, 2025-03-18:
> Let me dig a bit into it.

Since I am the owner of the BLE device, please let me know if you need any other information or if I can help in any way

SuGlider, 2025-03-18:
The 3rd field has type 0xFF and it has the Device Hash code.
| Length | Type | Value |
|---------|-------|-------|
| 8 | 0xFF | 0x59004461746100 |

The first 4 bytes of the hash code is the Nordic Company ID (0x0059)
The other 5 byte 0x4461746100 are a hash code that only the BLE Client that wants to connect knows how to use it.
It could be any thing, including a deviceID, a beacon information or a cypher code etc.

Usually Beacons only use a single field (0xFF) with its data - no UUID service or any other thing.
What is this BLE Server device?

pepassaco, 2025-03-18:
> What is this BLE Server device?

It's an indistrual device with some sensors (resistivity and accelerometer) that sends data depending on the commands it receives: It creates a BLE server and by writing to a characteristic you may get the data when subscribing to notifications on another characteristic. We tested this procedure from an android phone as well as from a computer with a dirty python script and a nrf52 dongle, and it worked flawlessly (both devices could discover the service UUID and the 3 characteristics with different UUIDs) so when we were trying to implement a portable solution we went for the ESP32 and found this issue

SuGlider, 2025-03-18:
I see. This is interesting. So the dirty python script + nrf52 dongle is able to connect using the listed UUID. The nrf APP also can see that UUID correctly. But the ESP32 can't.

Could it be something specific that only Nordic BLE layer knows how to deal with?
Have you tried with some other BLE dongle or even using a generic Laptop/Computer that has BLE and is able to connect/list UUIDs?

SuGlider, 2025-03-18:
I have a BLE HID device here and my ESP32-S3 can see the advertised Service UUID and connect to it using the same Characteristics UUIDs that I also see using the nRF APP (Android). But it doesn't advertise the manufacturer id, therefore I have no idea which chip/BLE stack it uses.

I can only tell you that it works as suposed.

pepassaco, 2025-03-18:
> Have you tried with some other BLE dongle or even using a generic Laptop/Computer that has BLE and is able to connect/list UUIDs?

Just tried with [the first BLE Scanner app I found](https://apps.microsoft.com/detail/9n0ztkf1qd98?hl=en-us&gl=US) and yes, the characteristics are being detected

SuGlider, 2025-03-18:
I'll try to investigate this issue with some other device.
Anyway we are changing our BLE Library to work with NimBLE intead of Bluedroid stack.

You may try using the Arduino-NimBLE library to check if it works.
Please let me know about it.

**https://github.com/h2zero/NimBLE-Arduino**

pepassaco, 2025-03-18:
> You may try using the Arduino-NimBLE library to check if it works.

With NimBLE it works flawlessly!

pepassaco, 2025-03-18:
> Anyway we are changing our BLE Library to work with NimBLE intead of Bluedroid stack.

When will this be available approx.? (Should I recode my app using NimBLE or it's worth waiting? 😅)

SuGlider, 2025-03-18:
> > Anyway we are changing our BLE Library to work with NimBLE intead of Bluedroid stack.
> 
> When will this be available approx.? (Should I recode my app using NimBLE or it's worth waiting? 😅)

@lucasssvaz is working on it right now.
Not sure when it would be ready.
We would like to release it with the ESP32 Arduino Core 3.3.

SuGlider, 2025-03-18:
Closing this issue, given that it has been solved. Feel free to open it again.

closed 2025-03-18 by SuGlider

Rob58329, 2025-07-26:
Note also that the current "ESP32 BLE Client example" also won't work for BLE-servers which don't transmit the "ServiceUUID" in the Initial Advertising packet (as for these Devices the "`advertisedDevice.haveServiceUUID()`" command always fails).

For such devices (eg. Shelly Home-Automation-Relays) the ServiceUUID is sent after the initial-Advertising-Packet, in a later "Scan Responce Packet".  However, I  cant currently find a way to access this in arduino (eg. with "BLESCANPARAM" or similar), but instead it's possible to use the Device's MAC address (which IS sent in the Initial Advertising packet) to identify the Device you want (and so you can then just connect to it and start looking for the CharacteristicUUID you want).