espressif/arduino-esp32 #11138: Cant compile

niekie813, 2025-03-18:
### Board

ESP32

### Device Description

Breadboard

### Hardware Configuration

GPIO 18 = button
GPIO 19 = button
GPIO 5 = buzzer

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

I cant compile.

### Sketch

```cpp
#include <BleKeyboard.h>

#define DOT_PIN 18
#define DASH_PIN 19
#define BUZZER_PIN 5

BleKeyboard bleKeyboard;  // Maak een instance van de BleKeyboard

String morseInput = "";
unsigned long lastInputTime = 0;
const unsigned long morseTimeout = 1000;  // Pauze na het invoeren van morsecode

// Morsecode vertalingen
const char* morseAlphabet[] = {
  ".-", "-...", "-.-.", "-..", ".", "..-.", "--.", "....", "..", 
  ".---", "-.-", ".-..", "--", "-.", "---", ".--.", "--.-", ".-.", 
  "...", "-", "..-", "...-", ".--", "-..-", "-.--", "--.."
};
const char morseLetters[] = "ABCDEFGHIJKLMNOPQRSTUVWXYZ";

void setup() {
  Serial.begin(115200);
  pinMode(DOT_PIN, INPUT_PULLUP);
  pinMode(DASH_PIN, INPUT_PULLUP);
  pinMode(BUZZER_PIN, OUTPUT);
[... 217 lines cut ...]
c:\Users\niek\OneDrive\Documenten\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp: At global scope:
c:\Users\niek\OneDrive\Documenten\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:164:6: error: variable or field 'setName' declared void
  164 | void BleKeyboard::setName(string deviceName) {
      |      ^~~~~~~~~~~
c:\Users\niek\OneDrive\Documenten\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:164:27: error: 'string' was not declared in this scope
  164 | void BleKeyboard::setName(string deviceName) {
      |                           ^~~~~~
c:\Users\niek\OneDrive\Documenten\Arduino\libraries\ESP32_BLE_Keyboard\BleKeyboard.cpp:164:27: note: suggested alternatives:
In file included from C:/Users/niek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/string:41,
                 from C:\Users\niek\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.3\libraries\BLE\src/BLECharacteristic.h:15:
C:/Users/niek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/bits/stringfwd.h:77:33: note:   'std::string'
   77 |   typedef basic_string<char>    string;
      |                                 ^~~~~~
C:/Users/niek/AppData/Local/Arduino15/packages/esp32/tools/esp-x32/2405/xtensa-esp-elf/include/c++/13.2.0/string:66:11: note:   'std::pmr::string'
   66 |     using string    = basic_string<char>;
      |           ^~~~~~

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-03-18:
@niekie813 - 

The BLE stack has been changed in Arduino Core version 3.0.0-Alpha1 to replace all `std::string` for `String` class from Arduino.
The change was done in https://github.com/espressif/arduino-esp32/pull/8724
https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha1

The library `BleKeyboard` is compatible with Arduino Core 2.0.x instead.
You can try to change the ESP32 Arduino Core version to 2.0.17 and test this sketch.

This BLE Keyboard library must be modified in order to be compatible with Arduino Core 3.0.x and latter.

SuGlider, 2025-03-20:
Closing this issue with the resolution described above. Feel free to open it again.

closed 2025-03-20 by SuGlider