espressif/arduino-esp32 #11139: Revisiting WiFi Connection Speed (<200ms with static ip + bssid+ channel, 1.5-2 secs without static IP)

kapyaar, 2025-03-18:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module

### Hardware Configuration

Base Board with no peripherals. 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I am working on a battery powered application, hence connection speed to wifi is being tested. I have gone through [1675](https://github.com/espressif/arduino-esp32/issues/1675), but wanted to check if there has been any changes (for good or bad) in this context.
 Is the almost 10x time disparity between static ip and dhcp expected?

### Sketch

```cpp
#include <WiFi.h>
const char *ssid = "wifi_ssid";          
const char *password = "wifi_password"; 

void setup() {
  Serial.begin(115200);
  Serial.print("Start Connect At ");
  Serial.println(millis());

  uint8_t bssid[6] = { 0xDE, 0xAD, 0xBE, 0xEF, 0x10, 0x3E };
  int32_t channel = 6;

  IPAddress local_IP(192, 168, 2, 61);
  IPAddress gateway(192, 168, 2, 1);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress primaryDNS(8, 8, 8, 8);
  IPAddress secondaryDNS(8, 8, 4, 4);

  WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS);  // Static IP connects in ~200mS. Without, Takes upto 2 seconds.
  WiFi.begin(ssid, password, channel, bssid, 1);

  //WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(10);
    Serial.print(".");
  }
  Serial.println("WiFi connected at ");
  Serial.println(millis());
}
void loop() {
  esp_sleep_enable_timer_wakeup(10 * 1000000);
  esp_deep_sleep_start();
}
```

### Debug Message

```plain
Just the time difference in connection, under 

  while (WiFi.status() != WL_CONNECTED) {
    delay(10);
    Serial.print(".");
  }

No Error messages.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kapyaar, 2025-03-18:
Got it, thanks! In my tests, the time difference seemed fairly consistent so I was curious whether any of the scan related parameter was at play.

lbernstone, 2025-03-19:
scan related parameters? I don't see any scanning in your code. I ran the simplest connection code possible with verbose logging
```
#include <WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.begin("ssid","pass");
  WiFi.waitForConnectResult();
}
void loop() {delay(-1);}
```
I get this in the logs:
```
15:14:05.859 -> [   681][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 112 - STA_CONNECTED
15:14:05.859 -> [   689][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 112 - STA_CONNECTED
15:14:05.892 -> [   705][V][NetworkInterface.cpp:78] _onIpEvent(): sta Got New IP: 192.168.2.240 MASK: 255.255.255.0 GW: 192.168.2.1
15:14:05.892 -> [   716][V][NetworkEvents.cpp:117] _checkForEvent(): Network Event: 115 - STA_GOT_IP
15:14:05.892 -> [   724][V][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 115 - STA_GOT_IP
15:14:05.892 -> [   731][V][STA.cpp:171] _onStaArduinoEvent(): STA IP: 192.168.2.240, MASK: 255.255.255.0, GW: 192.168.2.1
```
That gap between STA_CONNECTED and the finish of GOT_IP is the DHCP transaction. Any extra code other than that simple begin is unnecessary (well, some error checking is good). If you aren't allowing the device to save the wifi connection information (which includes the lease info), then that will slow your connection down.

lbernstone, 2025-03-19:
You would need to wireshark (or have logs on the DHCPd) to see exactly what is going on in the dhcp negotiation.  But, it seems like the slowdown is not on the esp32.

kapyaar, 2025-03-19:
I did a quick wireshark capture with WiFi.begin(ssid,pwd) and sleep for 10sec. I see something like this. 

[image]

And they look like this.
```
215	33.877358	Espressif_4f:90:e4	Broadcast	XID	52	Basic Format; Type 1 LLC (Class I LLC); Window Size 0
Frame 215: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface en0, id 0
    Section number: 1
    Interface id: 0 (en0)
        Interface name: en0
        Interface description: Wi-Fi
    Encapsulation type: Ethernet (1)
    Arrival Time: Mar 19, 2025 14:12:24.773200000 EDT
    UTC Arrival Time: Mar 19, 2025 18:12:24.773200000 UTC
    Epoch Arrival Time: 1742407944.773200000
    [Time shift for this packet: 0.000000000 seconds]
    [Time delta from previous captured frame: 0.818851000 seconds]
    [Time delta from previous displayed frame: 0.818851000 seconds]
    [Time since reference or first frame: 33.877358000 seconds]
    Frame Number: 215
    Frame Length: 52 bytes (416 bits)
    Capture Length: 52 bytes (416 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:llc]
    [Coloring Rule Name: Broadcast]
    [Coloring Rule String: eth[0] & 1]
IEEE 802.3 Ethernet 
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
        .... ..1. .... .... .... .... = LG bit: Locally administered address (this is NOT the factory default)
        .... ...1 .... .... .... .... = IG bit: Group address (multicast/broadcast)
    Source: Espressif_4f:90:e4 (90:15:06:4f:90:e4)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Length: 38
    [Stream index: 3]
Logical-Link Control
    DSAP: NULL LSAP (0x00)
        0000 000. = SAP: NULL LSAP
        .... ...0 = IG Bit: Individual
    SSAP: NULL LSAP (0x01)
        0000 000. = SAP: NULL LSAP
        .... ...1 = CR Bit: Response
    Control field: U, func=XID (0xAF)
        101. 11.. = Response: Exchange identification (0x2b)
        .... ..11 = Frame type: Unnumbered frame (0x3)
Logical-Link Control Basic Format XID
    XID Format: LLC basic format (0x81)
    LLC Types/Classes: Type 1 LLC (Class I LLC) (0x01)
    Receive Window Size: 0

===
218	33.980969	Espressif_4f:90:e4	Broadcast	ARP	52	Who has 192.168.2.61? (ARP Probe)
Frame 218: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface en0, id 0
    Section number: 1
    Interface id: 0 (en0)
        Interface name: en0
        Interface description: Wi-Fi
    Encapsulation type: Ethernet (1)
    Arrival Time: Mar 19, 2025 14:12:24.876811000 EDT
    UTC Arrival Time: Mar 19, 2025 18:12:24.876811000 UTC
    Epoch Arrival Time: 1742407944.876811000
    [Time shift for this packet: 0.000000000 seconds]
    [Time delta from previous captured frame: 0.000289000 seconds]
    [Time delta from previous displayed frame: 0.000289000 seconds]
    [Time since reference or first frame: 33.980969000 seconds]
    Frame Number: 218
    Frame Length: 52 bytes (416 bits)
    Capture Length: 52 bytes (416 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:arp]
    [Coloring Rule Name: ARP]
    [Coloring Rule String: arp]
Ethernet II, Src: Espressif_4f:90:e4 (90:15:06:4f:90:e4), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
        .... ..1. .... .... .... .... = LG bit: Locally administered address (this is NOT the factory default)
        .... ...1 .... .... .... .... = IG bit: Group address (multicast/broadcast)
    Source: Espressif_4f:90:e4 (90:15:06:4f:90:e4)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
    [Stream index: 3]
    Trailer: 00000000000000000000
Address Resolution Protocol (ARP Probe)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    [Is probe: True]
    Sender MAC address: Espressif_4f:90:e4 (90:15:06:4f:90:e4)
    Sender IP address: 0.0.0.0
    Target MAC address: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Target IP address: 192.168.2.61
```

I don't know what I am looking for, so could nt get much mileage out of it. :) 
IMO,
1. No matter what we find, if what you stated holds (that esp has nothing to do with the delay in getting the dhcp ip), then, there is not much point in troubleshooting, and I will live with what various routers out there offer in terms of their dhcp behaviour.
2. On the other hand, if there is any use that may come out of looking deeper, I will be happy to look for specifics and report back.

me-no-dev, 2025-11-06:
There are two things at play. First one is on which channel is your AP and is the time it takes to scan the channels until it gets to yours. Second is the DHCP transactions, which depends on your DHCP server and network load. We have done all we can to shorten the DHCP time (request the previous IP if available) but that is as much as we can do. Fastest connect would be as you saw, static IP and BSSID+Channel. For me DHCP sometimes also takes 1 second and there is nothing I can do about that. With static IP and without setting BSSID/channel, connection takes between 90 and 120ms

closed 2025-11-06 by me-no-dev