espressif/arduino-esp32 #11142: SD doesn't begins; There is no valid FAT volume part 2

KenFellows, 2025-03-19:
### Board

ESP32 VROOM 32D

### Device Description

Only the Micro SD module

### Hardware Configuration

Only the one module attached.

### Version

v3.1.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Connecting to the Micro SD module.  I am using the default pins.  I keep getting the following error message.
When I connect the board to an Arduino UNO it works fine.
Was this bug ever been resolved?

### Sketch

```cpp
#include <SD.h>

#define PIN_SPI_CS 5 // The ESP32 pin GPIO5

File myFile;

void setup() {
Serial.begin(9600);

if (!SD.begin(PIN_SPI_CS)) {
while (1) {
Serial.println(F("SD CARD FAILED, OR NOT PRESENT!"));
delay(1000);
}
}

Serial.println(F("SD CARD INITIALIZED."));

if (!SD.exists("/esp32.txt")) {
Serial.println(F("esp32.txt doesn't exist. Creating esp32.txt file..."));
// create a new file by opening a new file and immediately close it
myFile = SD.open("/esp32.txt", FILE_WRITE);
myFile.close();
}

// recheck if file is created or not
if (SD.exists("/esp32.txt"))
Serial.println(F("esp32.txt exists on SD Card."));
else
Serial.println(F("esp32.txt doesn't exist on SD Card."));
}

void loop() {
}
```

### Debug Message

```plain
32-hal-periman.c:160] perimanSetPinBus(): Pin 23 successfully set to type GPIO (1) with bus 0x18
[ 2473][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 23 successfully set to type SPI_MASTER_MOSI (39) with bus 0x4
[ 2602][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x400fe564
[ 2734][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 5 successfully set to type GPIO (1) with bus 0x6
[ 2849][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type SD_SS for pin 5
[ 2972][E][sd_diskio.cpp:761] sdcard_mount(): f_mount failed: (13) There is no valid FAT volume
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-03-19:
I think ExFAT is disabled by default.  How is the disk formatted?

Jason2866, 2025-03-29:
Closing since issue can not reproduced with a SD Card formatted with FAT

closed 2025-03-29 by Jason2866