espressif/arduino-esp32 #11148: Errors trying to compile with idf.py command interface

Jim-Foo, 2025-03-20:
### Board

esp32-c3

### Device Description

esp32-c3-zero. The board is not connected right now because I'm just trying to compile.

### Hardware Configuration

Nothing attached. I'm trying to compile the hello world example.

### Version

latest master (checkout manually)

### IDE Name

esp-idf command interface

### Operating System

Linux

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm getting lots of errors when trying to compile a simple example such as system headers not found. I'm using esp-idf v5.4 with arduino-esp32 git master. Is there some combination of the two that is known to work?

At the moment I'm getting the error:

Compilation failed because HttpsOTAUpdate.cpp (in "arduino-esp32" component) includes esp_http_client.h, provided by esp_http_client component(s).
However, esp_http_client component(s) is not in the requirements list of "arduino-esp32"
To fix this, add esp_http_client to PRIV_REQUIRES list of idf_component_register call in /home/jim/dev/src/esp-idf-5.4/components/arduino-esp32/CMakeLists.txt.

I've made a few modifications to the CMakeLists.txt file in the arduino-esp32 directory but more errors keep appearing. the foreach loop in the CMakeLists.txt file assumes that headers are in a directory name "src" so it misses some includes. I've had to add REQUIRES to some CMakeLists.txt so the component can find headers such as the esp_mm component that couldn't find freertos headers.

### Sketch

```cpp
#include "Arduino.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```

### Debug Message

```plain
Compilation failed because HttpsOTAUpdate.cpp (in "arduino-esp32" component) includes esp_http_client.h, provided by esp_http_client component(s).
However, esp_http_client component(s) is not in the requirements list of "arduino-esp32"
To fix this, add esp_http_client to PRIV_REQUIRES list of idf_component_register call in /home/jim/dev/src/esp-idf-5.4/components/arduino-esp32/CMakeLists.txt.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-20:
It seems that something is wrong with your build environment? Maybe start here: https://components.espressif.com/components/espressif/arduino-esp32/versions/3.2.0-rc2/examples/hello_world?language=en

Jason2866, 2025-03-20:
It is easy with the Platformio fork [pioarduino](https://github.com/pioarduino/platform-espressif32) to compile Arduino as an component of IDF. See example https://github.com/pioarduino/platform-espressif32/tree/main/examples/espidf-arduino-blink