espressif/arduino-esp32 #11156: Zigbee Color Dimmable Light Example: wrong RGB values

wf2020, 2025-03-20:
### Board

ESP32-C6

### Device Description

Development Board

### Hardware Configuration

GPIO15 connected to LED

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Tried the Zigbee Color Dimmable Light Example.
When setting colors with Zigbee2Mqtt (Home Assistant) the RGB values set in the example don't match the values sent:

Setting red to 255 sets red only to 69.
Setting green to 255 sets green only to 34.
Setting blue to 255 sets blue only to 128.

### Sketch

```cpp
ZigbeeColorDimmableLight zbColorLight = ZigbeeColorDimmableLight(ZIGBEE_RGB_LIGHT_ENDPOINT);

/********************* RGB LED functions **************************/
void setRGBLight(bool state, uint8_t red, uint8_t green, uint8_t blue, uint8_t level) {
  if (!state) {
    rgbLedWrite(led, 0, 0, 0);
    return;
  }
  float brightness = (float)level / 255;
  rgbLedWrite(led, red * brightness, green * brightness, blue * brightness);

/*** Output RGB values ***/
  Serial.print("R: ");
  Serial.print(red,DEC);
  Serial.print(" G: ");
  Serial.print(green,DEC);
  Serial.print(" B: ");
  Serial.println(blue,DEC);
}
```

### Debug Message

```plain
R: 69 G: 0 B: 0
R: 0 G: 34 B: 0
R: 0 G: 0 B: 128
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-21:
Hi @wf2020, can you post some more logs from the ESP? Please select in the Arduino IDE Tools menu -> Zigbee mode -> End Device - DEBUG and also set Core Debug Level -> Verbose and post whole log here :)

Also If anyone using Z2M can test this and confirm, I will appreciate the help.

P-R-O-C-H-Y, 2025-03-24:
Can you please provide full log? Thanks

wf2020, 2025-03-24:
Here including pairing:

[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): DeingbeeCore.cpp:289] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
..........ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
......................ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:251   No dev for join
ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 0
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
[ 11512][I][ZigbeeCore.cpp:289] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
..........ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
.......................ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:186   low lqi 0
ZBOSS: nwk/nwk_join.c:293   Will assoc to pan 0x4306 on channel 11, dev 0x5312, iface_id 0
.....ZBOSS: mac/mac_associate.c:448   received association response
ZBOSS: zdo/zdo_app_join.c:771   CONGRATULATIONS! joined status 0, addr 0xd969, iface_id 0, ch 11, rejoin 0
ZBOSS: zdo/zdo_app_join.c:773   xpanid 8c:d2:90:cd:df:f3:d4:b9
.ZBOSS: secur/zdo_secur.c:211   NWK key fb:46:61:d6:fd:7b:0a:e0:20:e7:99:d5:aa:78:b4:c6
...........ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 0
[ 16433][I][ZigbeeCore.cpp:282] esp_zb_app_signal_handler(): Joined network successfully (Extended PAN ID: 8c:d2:90:cd:df:f3:d4:b9, PAN ID: 0x4306, Channel:11, Short Address: 0xd969)

=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   445308 B ( 434.9 KB)
  Free Bytes        :   380484 B ( 371.6 KB)
  Allocated Bytes   :    56464 B (  55.1 KB)
  Minimum Free Bytes:   380484 B ( 371.6 KB)
  Largest Free Block:   352244 B ( 344.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     9 : GPIO
    12 : USB_DM
    13 : USB_DP
    15 : RMT_TX
    16 : UART_TX[0]
    17 : UART_RX[0]
============ After Setup End =============
[ 47292][V][ZigbeeHandlers.cpp:77] zb_attribute_set_handler(): Received message: endpoint(10), cluster(0x300), attribute(0x3), data size(2)
[ 47294][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 15 - TX MODE - MemSize[48] - Freq=10000000Hz
[ 47294][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 15 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 47295][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 15 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
R: 69 G: 0 B: 0
[ 51803][V][ZigbeeHandlers.cpp:77] zb_attribute_set_handler(): Received message: endpoint(10), cluster(0x300), attribute(0x4), data size(2)
[ 51803][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 15 - TX MODE - MemSize[48] - Freq=10000000Hz
[ 51804][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 15 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 51804][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 15 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
R: 0 G: 34 B: 0
[ 53667][V][ZigbeeHandlers.cpp:77] zb_attribute_set_handler(): Received message: endpoint(10), cluster(0x300), attribute(0x4), data size(2)
[ 53667][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 15 - TX MODE - MemSize[48] - Freq=10000000Hz
[ 53668][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 15 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[ 53668][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 15 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
R: 0 G: 0 B: 128

ESP32the2nd, 2025-03-26:
Same issue here. Is there some color correction within the Zigbee magic which makes the colors appear the same brightness? If so, it would be interesting to switch that off. I used the Zigbee2MQTT web interface for testing the example sketch using a ESP32C6 Super Mini board.

pavelvencl, 2025-04-23:
Have same issue. Home Assistant - Zigbee2Mqtt get same values as mentioned.

Setting red to 255 sets red only to 69.
Setting green to 255 sets green only to 34.
Setting blue to 255 sets blue only to 128.

thorrak, 2025-06-29:
Having dug into the code a bit for a different (but potentially related?) issue #11527 there's no color correction taking place, but there is a quirk in the way that the Zigbee standard treats color (hue and saturation) and how we typically think of them 

Per the Zigbee standard, colors are sent as hue & saturation, not RGB, and both these values must be in the range of 0-254 -- not the 0-255 that you would typically expect. For the three colors in question - #FF0000, #00FF00, and #0000FF, the saturation would map to 255 -- higher than the standard allows.

I don't know anything about Zigbee2Mqtt, but is it possible to see the actual message it transmits to see how it is handling this case? My guess is that it is actually sending the 255 saturation value, which is resulting in some kind of undefined behavior on the ESP.

lucasssvaz, 2025-07-31:
@P-R-O-C-H-Y Was this fixed in #11624 ? Should we close this issue ?

wf2020, 2025-08-01:
> [@P-R-O-C-H-Y](https://github.com/P-R-O-C-H-Y) Was this fixed in [#11624](https://github.com/espressif/arduino-esp32/pull/11624) ? Should we close this issue ?

Yes.
 I can confirm that all colors are now set correctly to max level 255.
Thank you.

closed 2025-08-01 by wf2020