espressif/arduino-esp32 #11162: Uninitialized variable in USB.cpp

petersenm, 2025-03-22:
### Board

ESP32-S2

### Device Description

Custom

### Hardware Configuration

Custom

### Version

v3.1.1

### IDE Name

None (command line)

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

961200

### Description

When compiling with --warnings all, the tud_mount_cb(), tud_umount_cb(), and tud_resume_cb() produce warnings concerning the uninitialized variable p.

`In function 'esp_err_t arduino_usb_event_post(esp_event_base_t, int32_t, void*, size_t, TickType_t)',
    inlined from 'void tud_mount_cb()' at /home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:103:25:
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:87:27: warning: 'p' may be used uninitialized [-Wmaybe-uninitialized]
   87 |   return esp_event_post_to(arduino_usb_event_loop_handle, event_base, event_id, event_data, event_data_size, ticks_to_wait);
      |          ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.h:22,
                 from /home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:15:
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h: In function 'void tud_mount_cb()':
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h:391:11: note: by argument 4 of type 'const void*' to 'esp_err_t esp_event_post_to(esp_event_loop_handle_t, esp_event_base_t, int32_t, const void*, size_t, TickType_t)' declared here
  391 | esp_err_t esp_event_post_to(esp_event_loop_handle_t event_loop,
      |           ^~~~~~~~~~~~~~~~~
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:102:28: note: 'p' declared here
  102 |   arduino_usb_event_data_t p;
      |                            ^
In function 'esp_err_t arduino_usb_event_post(esp_event_base_t, int32_t, void*, size_t, TickType_t)',
    inlined from 'void tud_umount_cb()' at /home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:110:25:
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:87:27: warning: 'p' may be used uninitialized [-Wmaybe-uninitialized]
   87 |   return esp_event_post_to(arduino_usb_event_loop_handle, event_base, event_id, event_data, event_data_size, ticks_to_wait);
      |          ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h: In function 'void tud_umount_cb()':
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h:391:11: note: by argument 4 of type 'const void*' to 'esp_err_t esp_event_post_to(esp_event_loop_handle_t, esp_event_base_t, int32_t, const void*, size_t, TickType_t)' declared here
  391 | esp_err_t esp_event_post_to(esp_event_loop_handle_t event_loop,
      |           ^~~~~~~~~~~~~~~~~
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:109:28: note: 'p' declared here
  109 |   arduino_usb_event_data_t p;
      |                            ^
In function 'esp_err_t arduino_usb_event_post(esp_event_base_t, int32_t, void*, size_t, TickType_t)',
    inlined from 'void tud_resume_cb()' at /home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:126:25:
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:87:27: warning: 'p' may be used uninitialized [-Wmaybe-uninitialized]
   87 |   return esp_event_post_to(arduino_usb_event_loop_handle, event_base, event_id, event_data, event_data_size, ticks_to_wait);
      |          ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h: In function 'void tud_resume_cb()':
/home/railfan/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32s2/include/esp_event/include/esp_event.h:391:11: note: by argument 4 of type 'const void*' to 'esp_err_t esp_event_post_to(esp_event_loop_handle_t, esp_event_base_t, int32_t, const void*, size_t, TickType_t)' declared here
  391 | esp_err_t esp_event_post_to(esp_event_loop_handle_t event_loop,
      |           ^~~~~~~~~~~~~~~~~
/home/railfan/.arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:125:28: note: 'p' declared here
  125 |   arduino_usb_event_data_t p;
      |                            ^
`

tud_suspend_cb() does not generate that warning because it does initialize p before calling arduino_usb_event_post():

`  arduino_usb_event_data_t p;
  p.suspend.remote_wakeup_en = remote_wakeup_en;
  arduino_usb_event_post(ARDUINO_USB_EVENTS, ARDUINO_USB_SUSPEND_EVENT, &p, sizeof(arduino_usb_event_data_t), portMAX_DELAY);
`

The functions producing the warning skip the p.suspend.remote_wakeup_en = remote_wakeup_en; line (or something similar to initialize the variable).

### Sketch

```cpp
void setup()
{
	// Open serial communications and wait for port to open:
	Serial.begin(115200);
}

void loop()
{
	Serial.println("Blah!");
}
```

### Debug Message

```plain
N/A - see description
```

### Other Steps to Reproduce

arduino-cli compile -b esp32:esp32:esp32s2:CDCOnBoot=cdc,UploadMode=cdc,PSRAM=enabled --build-path ./build --build-property upload.maximum_size=4063232 --build-property compiler.cpp.extra_flags='-DGIT_REV=0xb7018aL -DVERSION_STRING="0.0.0"' --build-property compiler.optimization_flags=-O2 test --warnings all

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2025-04-07:
Issue confirmed.

SuGlider, 2025-04-07:
This shall be fixed by #11222

PR #11222 (2025-04-09): fix(usb): uninitilized variable warning message
## Description of Change
Using ESP32-S2 with USB CDC enabled and arduino-cli with all warnings set, there is a warning message error:

```
arduino15/packages/esp32/hardware/esp32/3.1.1/cores/esp32/USB.cpp:102:28: note: 'p' declared here 102 
   arduino_usb_event_data_t p; 
   In function 'esp_err_t arduino_usb_event_post(esp_event_base_t, int32_t, void*, size_t, TickType_t)', inlined from 'void tud_umount_cb()'
```

## Tests scenarios
ESP32-S2 with Arduino CDC on Boot enbaled using Arduino CLI with all warnings activated.

`arduino-cli compile -b [... cut]
closed 2025-04-09 by me-no-dev