espressif/arduino-esp32 #11163: Crash during compilation on Windows if username contains accented characters (e.g. "é") – introduced in 3.1.0

ericpommierfdj, 2025-03-22:
### Board

Lolin S2 Mini (ESP32-S2)

### Device Description

Nothing

### Hardware Configuration

Nothing

### Version

v3.1.2

### IDE Name

Arduino IDE 2.3.4

### Operating System

Windows 11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

## Bug Summary

On Windows, if the current user's username contains an accented character (e.g. "Gérard"), version 3.1.0, 3.1.1 and 3.1.2 of the ESP32 Arduino core crash during compilation with this Rust panic:

**thread 'main' panicked at 'assertion failed: (left != right) left: 0, right: 0: Failed to get path name. Error code: 3', main.rs:65:9 exit status 101**

## Affected Versions

- ✅ 3.0.7 → Works fine
- ❌ 3.1.0 → Fails
- ❌ 3.1.1 → Fails
- ❌ 3.1.2 → Fails

## System

- Windows 11
- Arduino IDE 2.3.4
- Board: Lolin S2 Mini (ESP32-S2)
- User account name: `Gérard` → stored in path like `C:\Users\Gérard`

## Steps to Reproduce

1. Use a Windows user account with an accented character in the name (é, à, etc.)
2. Install ESP32 core 3.1.0 or 3.1.1 via Board Manager
3. Compile any sketch (even basic examples)
4. Compilation fails immediately with panic

## Notes

- This issue was not present in 3.0.7 and earlier.
- Likely related to newer tools based on Rust that do not handle UTF-8 paths properly under Windows.

### Sketch

```cpp
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

### Debug Message

```plain
thread 'main' panicked at 'assertion failed: (left != right) left: 0, right: 0: Failed to get path name. Error code: 3', main.rs:65:9 exit status 101
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-24:
Can you please provide screenshots and full output of the compilation?

lucasssvaz, 2025-03-24:
If you roll back the version to 3.0.7 it compiles fine ? AFAIK we don't have any tool that uses rust.

ericpommierfdj, 2025-03-25:
> If you roll back the version to 3.0.7 does it compile properly ? AFAIK we don't have any tool that uses rust.

Yes, confirmed — 3.0.7 compiles without any issue using the same sketch and environment.
The error only happens with 3.1.0, 3.1.1 and 3.1.2.

The crash message mentions main.rs:65:9 and a panic in thread 'main', which looks very similar to a Rust binary panic.
Maybe it's not directly from your toolchain, but something being invoked during build?

I'm compiling using Arduino IDE 2.3.4 on Windows, and my username contains an accent (é).

I’ll provide detailed logs and a screenshot shortly.

lucasssvaz, 2025-08-21:
@SuGlider It is still not applied in IDF. Only in IDF 6 probably.

me-no-dev, 2026-10-01:
Closing. Should be working now

closed 2026-10-01 by me-no-dev