espressif/arduino-esp32 #11164: LEDC resolution issue

Vladislav-Jovanov, 2025-03-22:
### Board

ESP32 Dev Board

### Device Description

Terminal adapter

### Hardware Configuration

GPIO 5,2,14 were tested out

### Version

latest master (checkout manually)

### IDE Name

Arduion IDE

### Operating System

Linux Mint

### Flash frequency

40 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am not sure if this is a software or hardware issue.
I was testing out different combinations of frequency and ledc resolution and duty cycle. I wanted to set resolution to LEDC_TIMER_1_BIT and duty to 1 (50%) and then to vary frequency from really low (few Hz to MHz).

What I noticed was that combination LEDC_TIMER_1_BIT and duty of 1 does not work for low frequencies of 1Hz but does work for 10kHz. But I don't understand why. Is that a software or hardware issue. In technical documentation there is no information about this. An answer to this issue would be sufficient.

### Sketch

```cpp
#include "driver/ledc.h"

#if defined (ESP32)
    #define LEDC_TIMER              LEDC_TIMER_0
    #define LEDC_MODE               LEDC_HIGH_SPEED_MODE
    #define LEDC_CHANNEL            LEDC_CHANNEL_0
    #define LEDC_DUTY_RES           LEDC_TIMER_1_BIT // Set duty resolution to 13 bits
    #define LEDC_DUTY               1 // Set duty to 50%. (2 ** 8) * 50% = 4096
    #define LEDC_FREQUENCY          1 // Frequency in Hertz. Set frequency at 4 kHz
#endif

#define LED_GPIO 5

void setup() {
  #if defined (ESP32)
        ledc_timer_config_t ledc_timer = {
            .speed_mode       = LEDC_MODE,
            .duty_resolution  = LEDC_DUTY_RES,
            .timer_num        = LEDC_TIMER,
            .freq_hz          = LEDC_FREQUENCY,  // Set output frequency at 5 kHz
            .clk_cfg          = LEDC_AUTO_CLK
        };

        ledc_channel_config_t ledc_channel = {
            .gpio_num       = LED_GPIO,
            .speed_mode     = LEDC_MODE,
            .channel        = LEDC_CHANNEL,
            .intr_type      = LEDC_INTR_DISABLE,
            .timer_sel      = LEDC_TIMER,
            .duty           = LEDC_DUTY, // Set duty to 0%
            .hpoint         = 0 //phase shift
        };
        ledc_timer_config(&ledc_timer);
        ledc_channel_config(&ledc_channel);
    #endif
}

void loop() {

}
```

### Debug Message

```plain
No message.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2025-03-22:
This is a [hardware issue](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions).  The peripheral is designed primarily for controlling LEDs, so it is not capable of working at very low frequencies. There are other peripherals that will work.  I don't know your use case, but I would recommend either [SDM](https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32/api-reference/peripherals/sdm.html) or the [wave generator](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/dac.html#cosine-wave-output-cosine-mode)

Jason2866, 2025-03-25:
Closing since not an issue. The question is answered

closed 2025-03-25 by Jason2866