espressif/arduino-esp32 #11165: ESP_NOW_Broadcast_Slave unable to handle more than 1 master

faultylee, 2025-03-23:
### Board

XIAO_ESP32C3

### Device Description

Plain XIAO_ESP32C3

### Hardware Configuration

WIFI
Serial
Built-in LED (GPIO 8)

### Version

v3.0.7

### IDE Name

Arduino IDE

### Operating System

ArchLinux

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

I adopted the example of `libraries/ESP_NOW/examples/ESP_NOW_Broadcast_Slave/ESP_NOW_Broadcast_Slave.ino` and proceed to setup 2 masters. It works fine with 1 master. Once the 2nd master started transmit and the slave registered it, it stop reacting to the first 1 master, as if nothing was received. I use the example as a push button remote control, and it's only transmitting one at a time when I was testing back and forth.

The master code is the same as the example except it transmit every 10ms continiously when the button is pressed.

I then found example online that uses the `esp_now` library directly instead of through `ESP_NOW` as per the example, and directly registering the callback using `esp_now_register_recv_cb`, then I was able to see messages from both the masters, while still using the same example code for master.

I couldn't figure out what went wrong in the slave example, but I suspect something within the `vector` or the peer registration that prevents the callback from working. Or that only 1 copy of ccallback is supported by `esp_now`?

### Sketch

```cpp
/*
    ESP-NOW Broadcast Slave
    Lucas Saavedra Vaz - 2024

    This sketch demonstrates how to receive broadcast messages from a master device using the ESP-NOW protocol.

    The master device will broadcast a message every 5 seconds to all devices within the network.

    The slave devices will receive the broadcasted messages. If they are not from a known master, they will be registered as a new master
    using a callback function.
*/

#include "ESP32_NOW.h"
#include "WiFi.h"

#include <esp_mac.h>  // For the MAC2STR and MACSTR macros

#include <vector>

[... 281 lines cut ...]
09:09:48.482 ->   Message: Testing
09:09:50.605 -> Unknown peer 34:b7:da:f8:2c:f8 sent a broadcast message
09:09:50.605 -> Registering the peer as a master

09:09:50.605 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:50.605 ->   Message: Testing
09:09:50.638 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:50.638 ->   Message: Testing
09:09:50.638 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:50.638 ->   Message: Testing

NOTE: sending message from first master here and nothing show up

09:09:55.815 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:55.815 ->   Message: Testing
09:09:55.848 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:55.848 ->   Message: Testing
09:09:55.848 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:55.848 ->   Message: Testing
09:09:55.848 -> Received a message from master 34:b7:da:f8:2c:f8 (broadcast)
09:09:55.848 ->   Message: Testing

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2025-06-20:
Thanks for the report! Opened a PR with the fix.

faultylee, 2025-06-20:
> Thanks for the report! Opened a PR with the fix. The problem was dangling pointers when resizing the vector. Changed vector to use pointers.

Thanks for the PR, that is interesting. TIL `nullptr`

closed 2025-06-23 by me-no-dev