espressif/arduino-esp32 #11166: assert not working any more in 3.2

pschatzmann, 2025-03-23:
### Board

Any

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino

### Operating System

Linux

### Flash frequency

n/a

### PSRAM enabled

no

### Upload speed

n/a

### Description

Up to and including R 3.1 we could using assert was giving the following type of error messages

assert failed: void setup() file_speeds.ino:100 (SD.begin(PIN_AUDIO_KIT_SD_CARD_CS))

Backtrace: 0x40082541:0x3ffb1f30 0x4008d919:0x3ffb1f50 0x40093afe:0x3ffb1f70 0x400d404b:0x3ffb20a0 0x400dbbbf:0x3ffb2270 0x4008e592:0x3ffb2290

Starting R 3.2 asserts are ignored

### Sketch

```cpp
Just add an assert(false); to the setup()
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pschatzmann, 2025-04-05:
Any progress ? 
I think this is quite serious!

me-no-dev, 2025-04-08:
It took me a bit of searching :) Seems that `-DNDEBUG` is added by esp-matter by default and that disables assert. You can probably do your own macro that uses `abort()` instead, which works

me-no-dev, 2025-04-08:
cc: @SuGlider

SuGlider, 2025-04-08:
I'll take a look, but if Matter Library disables it, it can be fixed by Lib Builder for the next release.

SuGlider, 2025-04-09:
@pschatzmann - I have found a work around that fixes this issue using Core 3.2.0 (or any 3.x).
Add a file called `build_opt.h` to your Arduino project.
In this file add a line with `-UNDEBUG`, save it, build and upload the project to the ESP32. `assert()` will work now.

This will undefine `NDEBUG` and retore `assert()` functionality.

pschatzmann, 2025-04-09:
Thanks a lot for the work around: I confirm that this is working.

Looking forward for the final correction...

me-no-dev, 2025-04-09:
@SuGlider what worries me is that matter disables `assert` throughout our ESP-IDF precompiled libs. Can you check this with the Matter team please?

SuGlider, 2025-04-09:
> [@SuGlider](https://github.com/SuGlider) what worries me is that matter disables `assert` throughout our ESP-IDF precompiled libs. Can you check this with the Matter team please?

Actually, when CHIP builds using release mode, it sets `-DNDEBUG` in order to avoid executing not necessary debugging code.
Arduino Lib Builder pick it up and includes it into the `flags/defines` file.

It doesn't affect other IDF components. 
It only affects CHIP building process (there is a CHIP scriptBUILD.gn  for creating CMakeLists.txt in CHIP before building it).

me-no-dev, 2025-04-09:
it picks it up because the project files are compiled with this flag. If they are compiled with it, chances are that others are compiled also. We extract flags from the 3 empty files in the lib-builder main (c, cpp and S)

SuGlider, 2025-04-14:
The main problem here is that `-DNDEBUG`  is used to build all Arduino Static Libraries, therefore, it implies in that all IDF and other Libraries will disable `assert()` internally as @me-no-dev already mentioned.

I have already isolated the origin of this issue within ESP Matter Library.
I'm working on a definitive fix that will remove `-DNDEBUG` in the origin and assure that it doesn't affect the static libraries.
`-DNDEBUG` will also not be picked up by Lib Builder from the very beginning, avoiding it to be copied forward to the DEFINES file in the FLAGS folder.

Jason2866, 2025-04-14:
Mhh, in this xase the suggested workaround `-UNDEBUG` makes no sense either.

SuGlider, 2025-04-14:
> Mhh, in this case the suggested workaround `-UNDEBUG` makes no sense either.

Correct. It helps only within the Sketch.
I've confirmed that when using Lib Builder, `-DNDEBUG` is evaluated at the very beginning of CMake process and as consequence all Arduino Static Libraries are built with `assert()` disabled.

SuGlider, 2025-04-14:
> it makes sense as temporary patch for Arduino code, but not as actual full solution. Maybe we can employ it while [@SuGlider](https://github.com/SuGlider) is looking into a final solution with the matter team. What do you say [@SuGlider](https://github.com/SuGlider) ?

@me-no-dev @Jason2866 - I have found the issue.
Matter Library is included as ESP Registry Component.
It has a `-DNDEBUG` explicitly declared in its own `CmakeLists.txt` which is not in the `Matter v1.3` reporitory and neither in the `master` repository. It is there only in the `v1.4` branch.

I already talked to the matter team and they will push a new ESP Registry Component v1.4.1 to fix it.

SuGlider, 2025-04-14:
I also asked matter team to fix a warning message by applying the PR https://github.com/project-chip/connectedhomeip/pull/38069/files into esp_matter v1.4.
It shall be back ported this week to v1.4.1.

SuGlider, 2025-04-14:
> I also asked matter team to fix a warning message by applying the PR https://github.com/project-chip/connectedhomeip/pull/38069/files into esp_matter v1.4. It shall be back ported this week to v1.4.1.

CHIP PR already in place for the warning message:
https://github.com/project-chip/connectedhomeip/pull/38385

As soon as it is committed, a new ESP_MATTER v1.4.1 will be released with the `-DNDEBUG` issue also solved.
We shall have it sorted out for the next Arduino Core 3.2.1 release.

SuGlider, 2025-05-13:
@pschatzmann - This issue has been solved by a new Matter version that doesn't disable `assert` by using `-DNDEBUG` in the compilation time.

That will be fully working for Arduino Core 3.2.1+ and 3.3.0+

SuGlider, 2025-05-13:
Closing this issue as solved. 
Feel free to ask more information or reopen it.

closed 2025-05-13 by SuGlider