espressif/arduino-esp32 #11168: Support Endpoint Identification from Multiple Zigbee Devices

kiansiong85, 2025-03-24:
### Related area

Zigebee

### Hardware specification

ESP32C6

### Is your feature request related to a problem?

Current Zigbee library examples focus on one-to-one communication. In real use cases, like mine, we often connect multiple end devices (e.g. 5) to a single coordinator. No endpoint/device information is displayed when receiving data (e.g. temperature).

It is difficult to identify which device sent the data.

The existing temperature (I tried to modify 2 devices connected; the second one took a long time) example does not show how to scale to multiple devices.

### Describe the solution you'd like

Suggested improvements:

Show the sender's endpoint or device address when receiving data.

Add an example of multiple devices sending data to one coordinator.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2025-06-03:
Hi @kiansiong85, please if you are able to test the changes in the linked PR. It adds the functionality to receive endpoint+address of the device that sends the temperature or you can directly ask for the temperature or configuration to specific device.

P-R-O-C-H-Y, 2025-06-12:
Hi @kiansiong85 Thank you for the feedback. Can you share your code please?

kiansionglau, 2025-06-12:
Sure, need your help. sure can .

[Code.zip](https://github.com/user-attachments/files/20703687/Code.zip)

Appreciate your strong support and guide.  I don't know how send to you, but i zip the file and add attachment. if you no received, welcome inform me anytime.

P-R-O-C-H-Y, 2025-06-12:
I see the problem :) You need to add to the `setup()` where you are setting up the zbThermostat device `zbThermostat.allowMultipleBinding(true);` to allow multiple devices to bind to the Thermostat endpoint. Its just simple as this :) The multi-device binding have been improved in some last updates so it should work without any issues :)

kiansionglau, 2025-06-12:
Hi  P-R-O-C-H-Y，   Thank you so much — it's working now! I really appreciate your help

P-R-O-C-H-Y, 2025-06-17:
There shouldn't be any issue adding binding new devices any time. Does it print anything when the device joins the network to the log? Please set for Thermostat before flashing in IDE Tools menu -> `Zigbee mode: Zigbee ZCZR (coordinator/router) - Debug` and also set `Core Debug Level: Verbose` and post logs when the 3rd device joins.

kiansionglau, 2025-06-18:
Dear P-R-O-C-H-Y，   thanks a lot your advise, i use 2 device to test first.  this is the logs from serial port.                                                                 I think I can see the problem when using debug mode, but I'm not sure how to solve it yet.
All the temperature devices are using the same code, just with different endpoints.
I've already connected the device with endpoint 12. Now, I want to add another device with endpoint 13, and the thermostat allows connections for 180 seconds.
From the message below, it looks like the Zigbee device was added successfully, with short address: 0xEA90.

Serial port:
Temperature Device, TEMP_SENSOR_ENDPOINT_NUMBER 13

[ 49622][I][ZigbeeCore.cpp:271] esp_zb_app_signal_handler(): Start network steering
.............................[ 52450][I][ZigbeeCore.cpp:325] esp_zb_app_signal_handler(): Joined network successfully (Extended PAN ID: 40:4c:ca:ff:fe:5c:b9:e4, PAN ID: 0xa75b, Channel:11, Short Address: 0xea90)

Zigbee connected
[ 52599][D][ZigbeeTempSensor.cpp:81] setTemperature(): Setting temperature to 2551
Temperature: 25.58 
Temperature: 25.55 
Temperature: 25.61 

Serial port:
thermostat show New device commissioned or rejoined (short: 0xea90)

endpoint 12 is existing one. but no display the endpoint 13 temperature.

[  1357][I][ZigbeeCore.cpp:423] esp_zb_app_signal_handler(): Network(0xa75b) is open for 180 seconds
Temperature: 27.51°C from endpoint 12, address 0x6c3c
[  7069][I][ZigbeeCore.cpp:342] esp_zb_app_signal_handler(): New device commissioned or rejoined (short: 0xea90)
[  7069][D][ZigbeeCore.cpp:360] esp_zb_app_signal_handler(): Checking endpoint 1
[  7070][D][ZigbeeCore.cpp:370] esp_zb_app_signal_handler(): Device already bound to endpoint 1
[  7147][I][ZigbeeCore.cpp:423] esp_zb_app_signal_handler(): Network(0xa75b) is open for 180 seconds

Current device status:
Device 0x6c3c (Endpoint 12): Temp 27.51°C, Status: Online
Temperature: 27.51°C from endpoint 12, address 0x6c3c

Current device status:
Device 0x6c3c (Endpoint 12): Temp 27.48°C, Status: Online
Temperature: 27.42°C from endpoint 12, address 0x6c3c

Current device status:
Device 0x6c3c (Endpoint 12): Temp 27.42°C, Status: Online
Temperature: 27.42°C from endpoint 12, address 0x6c3c

Current device status:

[ziptocheck.zip](https://github.com/user-attachments/files/20791660/ziptocheck.zip)

P-R-O-C-H-Y, 2025-06-18:
I see no issue there why it should happen. The only thing if you are reprogramming some board, idk if you use erase_flash. So what may happen that if you don't remove the device from thermostat, but you are reflashing it so its connecting as new (from the temp sensor point of view), the Thermostat already knows this device so the binding is not processed as shown here in the log:
*Device already bound to endpoint 1*
```
[  7069][I][ZigbeeCore.cpp:342] esp_zb_app_signal_handler(): New device commissioned or rejoined (short: 0xea90)
[  7069][D][ZigbeeCore.cpp:360] esp_zb_app_signal_handler(): Checking endpoint 1
[  7070][D][ZigbeeCore.cpp:370] esp_zb_app_signal_handler(): Device already bound to endpoint 1
[  7147][I][ZigbeeCore.cpp:423] esp_zb_app_signal_handler(): Network(0xa75b) is open for 180 seconds
```

P-R-O-C-H-Y, 2025-06-18:
Problem I see is that you are using the ShortAddress as the device identifier only. For example if you reboot the Thermostat, you will lost the shortAdresses and from getBoundDevices you will receive only ieee addresses.