espressif/arduino-esp32 #11170: NetworkClientSecure errors (multiple)

Admsher, 2025-03-24:
### Board

ESP32 CHIP as specified by ESP-IDF  v5.3

### Device Description

ESP32 One board : https://www.waveshare.com/wiki/ESP32_One

### Hardware Configuration

ESP32 One has multiple connections with an FPC Connector and also SD Card slot. It also has GPIO connections to rspberry pi which can be seen in the link given.

### Version

latest master (checkout manually)

### IDE Name

ESP IDF

### Operating System

Ubuntu 24.04

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The issue comes for ssl_client.cpp file , it seems it doesn't get compiled and then eventually we get this error 

when did further digging got to this error:
ssl_client.cpp:9:10: fatal error: Arduino.h: No such file or directory
    9 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.

### Sketch

```cpp
#include "SD.h"
#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h"
#include "AudioTools/Disk/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"
#include <base64.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include "esp_camera.h"
#include "FS.h"
#include "SPI.h"
#include <fstream>
#include <ArduinoJson.h>

#define CAMERA_MODEL_ESP_EYE 
#include "camera_pins.h"

const char *ssid = "";
[... 232 lines cut ...]
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:102:(.text._ZN19NetworkClientSecure4stopEv+0x6): undefined reference to `_Z15stop_ssl_socketP17sslclient_context'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZZN19NetworkClientSecureC4EvENKUlP17sslclient_contextE_clES1_':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:36:(.text._ZZN19NetworkClientSecureC4EvENKUlP17sslclient_contextE_clES1_+0x6): undefined reference to `_Z15stop_ssl_socketP17sslclient_context'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecure5writeEPKhj':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:231:(.text._ZN19NetworkClientSecure5writeEPKhj+0x16): undefined reference to `_Z13send_net_dataP17sslclient_contextPKhj'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:239:(.text._ZN19NetworkClientSecure5writeEPKhj+0x74): undefined reference to `_Z13send_ssl_dataP17sslclient_contextPKhj'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecure4readEPhj':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:253:(.text._ZN19NetworkClientSecure4readEPhj+0x10): undefined reference to `_Z15get_net_receiveP17sslclient_contextPhi'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:286:(.text._ZN19NetworkClientSecure4readEPhj+0xdc): undefined reference to `_Z15get_ssl_receiveP17sslclient_contextPhi'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecure9availableEv':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:298:(.text._ZN19NetworkClientSecure9availableEv+0xe): undefined reference to `_Z16peek_net_receiveP17sslclient_contexti'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:302:(.text._ZN19NetworkClientSecure9availableEv+0x2e): undefined reference to `_Z12data_to_readP17sslclient_context'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecure7connectE9IPAddresstPKcS2_S2_S2_':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:148:(.text._ZN19NetworkClientSecure7connectE9IPAddresstPKcS2_S2_S2_+0x2c): undefined reference to `_Z16start_ssl_clientP17sslclient_contextRK9IPAddressmPKciS5_bS5_S5_S5_S5_bPS5_'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:151:(.text._ZN19NetworkClientSecure7connectE9IPAddresstPKcS2_S2_S2_+0x3f): undefined reference to `_Z22ssl_starttls_handshakeP17sslclient_context'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecure7connectEPKctS1_S1_':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:196:(.text._ZN19NetworkClientSecure7connectEPKctS1_S1_+0x3f): undefined reference to `_Z16start_ssl_clientP17sslclient_contextRK9IPAddressmPKciS5_bS5_S5_S5_S5_bPS5_'
/home/admsher/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/arduino/libarduino.a(NetworkClientSecure.cpp.obj): in function `_ZN19NetworkClientSecureC2Ev':
/home/admsher/esp32-new/components/arduino/libraries/NetworkClientSecure/src/NetworkClientSecure.cpp:35:(.text._ZN19NetworkClientSecureC2Ev+0x33): undefined reference to `_Z8ssl_initP17sslclient_context'
collect2: error: ld returned 1 exit status
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-24:
While you were compiling, there should have been warnings that you should enable some mbedtls cyphers. See this guard: https://github.com/espressif/arduino-esp32/blob/master/libraries/NetworkClientSecure/src/ssl_client.cpp#L22

lbernstone, 2025-03-24:
NB the default sdkconfigs used to build the sdk for this repo are located at https://github.com/espressif/esp32-arduino-lib-builder/tree/master/configs

Admsher, 2025-03-25:
> While you were compiling, there should have been warnings that you should enable some mbedtls cyphers. See this guard: https://github.com/espressif/arduino-esp32/blob/master/libraries/NetworkClientSecure/src/ssl_client.cpp#L22

This worked for me , although I suggest to go through the default  sdkconfig as well for anyone facing the same issue. Also increasing flash size will help. Closing this issue as resolved

closed 2025-03-25 by Admsher