espressif/arduino-esp32 #11174: Zigbee_On_Off_Light fails to connect on ESP32C6 after reboot

Snyfir, 2025-03-25:
### Board

ESP32C6

### Device Description

Seeed Studio XIAO ESP32C6

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

VSCode

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I can switch the light on even if the coordinator (Sonoff Zigbee dongle) is on the second floor, far from the end device. However, if I reboot my ESP32-C6, it can no longer reconnect to the coordinator unless I'm within 10 cm of it. I don’t understand why, because once it is connected, I can move much farther away, and the connection remains stable.

For example, Zigbee2MQTT reports an LQI of 188, and I can control the light. But if I shut down and restart my ESP32-C6, it is unable to reconnect unless I am very close to the coordinator.

### Sketch

```cpp
#include <Arduino.h>
#include "Zigbee.h"

#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif

#define WIFI_ENABLE 3
#define WIFI_ANT_CONFIG 14
#define USER_LED 15

/* Zigbee light bulb configuration */
#define ZIGBEE_LIGHT_ENDPOINT 10
uint8_t led = USER_LED;
uint8_t button = BOOT_PIN;

ZigbeeLight zbLight = ZigbeeLight(ZIGBEE_LIGHT_ENDPOINT);

void setLED(bool value)
{
  digitalWrite(led, !value);
}

void setup()
{
[... 174 lines cut ...]
I (681) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (781) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (781) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
Adding ZigbeeLight endpoint to Zigbee Core
Connecting to network...
I (787) ESP_ZIGBEE_CORE: SDK Version: v1.6.3-9d7f2d69-0b5a5412-f9a2362; esp32c6; 2025-02-14 08:41:28 UTC
I (798) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (840) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 10
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
```

### Other Steps to Reproduce

[image]

I have a Sonoff Zigbee dongle and no router
espressif/arduino-esp32: 3.2.0-rc2

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Parsaabasi, 2025-03-25:
Hi @P-R-O-C-H-Y 

Would you please check this one? 

Thanks

P-R-O-C-H-Y, 2025-03-25:
Hi @Snyfir, I am using the Espressifs C6 Devkits and I did not run in this situation. Can you post the network map if possible?

P-R-O-C-H-Y, 2025-03-25:
Also what can help is to use the Zigbee debug libs and send a log from the ESP.
TO use the debug libs go to `Arduino IDE Tools menu -> Zigbee mode -> Zigbee ED (end device) - Debug`

Snyfir, 2025-03-25:
Hi @P-R-O-C-H-Y, thank you for your help.

I have only one coordinator and my end device:
[image]

`
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3048
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4 2nd stage bootloader
I (23) boot: compile time Mar 25 2025 14:17:13
I (24) boot: chip revision: v0.1
I (24) boot: efuse block revision: v0.3
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (30) boot.esp32c6: SPI Mode       : DIO
I (34) boot.esp32c6: SPI Flash Size : 4MB
I (38) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (64) boot:  2 factory          factory app      00 00 00010000 00200000
I (71) boot:  3 zb_storage       Unknown data     01 81 00210000 00010000
I (77) boot:  4 zb_fct           Unknown data     01 81 00220000 00001000
I (84) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=39e0ch (237068) map
I (138) esp_image: segment 1: paddr=00049e34 vaddr=40800000 size=061e4h ( 25060) load
I (144) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=a07a4h (657316) map
I (266) esp_image: segment 3: paddr=000f07cc vaddr=408061e4 size=0ae8ch ( 44684) load
I (277) esp_image: segment 4: paddr=000fb660 vaddr=40811070 size=025d0h (  9680) load
I (283) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
I (294) cpu_start: Unicore app
I (303) cpu_start: Pro cpu start user code
I (303) cpu_start: cpu freq: 160000000 Hz
I (303) app_init: Application information:
I (304) app_init: Project name:     zigbee-light-bulb
I (308) app_init: App version:      ddcbd85-dirty
I (312) app_init: Compile time:     Mar 25 2025 14:17:04
I (317) app_init: ELF file SHA256:  d841814e1...
I (322) app_init: ESP-IDF:          v5.4
I (325) efuse_init: Min chip rev:     v0.0
I (329) efuse_init: Max chip rev:     v0.99
I (333) efuse_init: Chip rev:         v0.1
I (337) heap_init: Initializing. RAM available for dynamic allocation:
I (343) heap_init: At 408185D0 len 00064040 (400 KiB): RAM
I (348) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (353) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (360) spi_flash: detected chip: generic
I (363) spi_flash: flash io: dio
[     6][V][ZigbeeEP.cpp:18] ZigbeeEP(): Endpoint: 10
[    23][V][ZigbeeLight.cpp:10] ZigbeeLight(): Light endpoint created 10
[    29][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (38) successfully set to 0x42011894
--- 0x42011894: HWCDC::deinit(void*) at D:/Documents/GitHub/ESP32/zigbee-light-bulb/managed_components/espressif__arduino-esp32/cores/esp32/HWCDC.cpp:280

[    41][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (39) successfully set to 0x42011894
--- 0x42011894: HWCDC::deinit(void*) at D:/Documents/GitHub/ESP32/zigbee-light-bulb/managed_components/espressif__arduino-esp32/cores/esp32/HWCDC.cpp:280

[    52][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4201085a
--- 0x4201085a: _uartDetachBus_RX at D:/Documents/GitHub/ESP32/zigbee-light-bulb/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-uart.c:268

[    63][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42010818
--- 0x42010818: _uartDetachBus_TX at D:/Documents/GitHub/ESP32/zigbee-light-bulb/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-uart.c:275

[    74][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x420107d6
--- 0x420107d6: _uartDetachBus_CTS at D:/Documents/GitHub/ESP32/zigbee-light-bulb/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-uart.c:282

[... 143 lines cut ...]
[ 14308][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
[ 17138][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
[ 19968][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
[ 22798][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
[ 25628][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
ZB_TRACE_LOG[0]: nwk/nwk_join.c:186    low lqi 0
ZB_TRACE_LOG[0]: nwk/nwk_join.c:629    Rejoin failure: no dev with xpanid a5:d7:53:55:21:0a:ac:f1 or can't choose best parent
[ 28458][W][ZigbeeCore.cpp:254] esp_zb_app_signal_handler(): Commissioning failed, trying again...
[ 30657][E][ZigbeeCore.cpp:75] begin(): ZigbeeCore begin failed or timeout
Zigbee failed to start!
Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1e (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM
`

P-R-O-C-H-Y, 2025-03-25:
The 10cm is really close. I think there is an issue with your boards antenna or with the coordinator antenna.
Please refer to this issue https://github.com/espressif/arduino-esp32/issues/10951

Jason2866, 2025-03-25:
The code looks like to switch and connect the antenna, but probably this is not working correctly.

Snyfir, 2025-03-25:
I tried with an external antenna and set WIFI_ANT_CONFIG to HIGH, and it's better—I can connect at about 10 meters.
But what's strange is that when the LQI is below 200, it cannot reconnect when I restart the ESP32.
I tried walking to the street. I can walk up to 30 meters without losing the signal, and if I go back, the connection comes back at 30 meters.
However, if I restart my ESP at 15 meters, it will not reconnect. 
Strange!

Jason2866, 2025-03-25:
Establish a connection is different than keep one. All connection parameters are known. Device can and make retries here.

closed 2025-03-25 by Jason2866