espressif/arduino-esp32 #11177: Add the Battery Voltage Cluster to Battery powered Zigbee End Points

def1149, 2025-03-25:
### Related area

Zigbee 

### Hardware specification

NA

### Is your feature request related to a problem?

Devices lacking a battery monitor (gas gauge) only provide battery voltage. The discharge slope and rate of lithium batteries are nonlinear and dependent on multiple conditions, so manual conversion to percentage remaining is not accurate. For those devices, tracking voltage is an option.

### Describe the solution you'd like

Add the Battery Voltage cluster to battery powered EPs

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

closed 2025-04-09 by me-no-dev

def1149, 2025-04-09:
@P-R-O-C-H-Y  If battery voltage is not reportable, then what is it good for?

P-R-O-C-H-Y, 2025-04-10:
@def1149 It was requested by you, so I added. I found out later by checking the Zigbee cluster specification that it's not reportable attribute. But you can still read it manually.

def1149, 2025-04-10:
@P-R-O-C-H-Y It shows up as a bindable attribute in HA, which is why I thought it needed to be added 

[image]