espressif/arduino-esp32 #11178: Make Reboot after Factory Reset on an EP optional

def1149, 2025-03-25:
### Related area

Zigbee 

### Hardware specification

C6, H2

### Is your feature request related to a problem?

When an EP executes a factory reset, it also executes a device reboot. When I execute a factory reset, I  may want to shut the device down, not reboot. Currently, I need to set a shutdown flag in flash, and upon reboot, if the flag is set, clear it and shut the device down. That works, but it's a workaround. 

### Describe the solution you'd like

Add a boolean to factoryReset(bool reboot) 
If true, reboot.
If false, return to the caller 

### Describe alternatives you've considered

Flash-based flag to signal shutdown after the reboot from a factoryReset() call.

### I have checked existing list of Feature requests and the Contribution Guide

- [x] I confirm I have checked existing list of Feature requests and Contribution Guide.

Citrus101, 2025-03-26:
Hello, can I take up this issue or is anyone else working on this?

P-R-O-C-H-Y, 2025-03-26:
Hi @Citrus101, I have this assigned to me as I am maintainer of Zigbee Library :)
But if you would like to take this, is. a simple change feel free to open a PR with the change and ping me there.

Citrus101, 2025-03-26:
Oh okay perfect, will do.

Citrus101, 2025-03-26:
@P-R-O-C-H-Y sorry to bother you like this but can you provide me your thought process for the solution you were going to come up with?
I'm struggling since I thought a simple modification to the factoryReset function would suffice but it doesn't seem like it.
I cannot code the described solution since esp_zb_factory_reset is a Core API function and can't avoid the reboot performed by the device.

P-R-O-C-H-Y, 2025-03-26:
@Citrus101 I have some thought but I would need to investigate the options. Maybe it's not that simple as I imagine. So please let me work on this fr. Would be easier for me to do it directly when I would need to also investigate.

Citrus101, 2025-03-26:
Yes agreed. I'll sit this one out.

PR #11153 (2025-03-27): feat(zigbee): Add check, boolean returns, fix Analog, add optional reset on factoryReset
## Description of Change

- Fix memory leak
- Updated almost all methods to return a boolean value and print error message with code and string of the error.
- `factoryReset` method now have an optional parameter `bool restart`, by default its set to TRUE, but if false is passed, the Zigbee will just erase the nvram and the reset or reboot will need to be managed in user sketch. Example of how it can be used is the Temperature Humidity Sleepy example.
- Simplified the `AnalogSensor` Endpoint -> removed Ana [... cut]
closed 2025-03-27 by me-no-dev