espressif/arduino-esp32 #11180: I2C Wire causes getMaxAllocHeap exception 'Interrupt wdt timeout' when reading more than 132 bytes

bhcuong2008, 2025-03-26:
### Board

ESP32 S3 ESP-4827S043

### Device Description

Board HMI 4.3" with ESP32 S3 revision 0.1. I2C (pin 19 SDA, 20 SCL) connected to touch GT911.

### Hardware Configuration

No.

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Ubuntu Desktop 22.04

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

1152000

### Description

The test based on WireMaster example. If reading bytes more than 132 bytes, `Wire.requestFrom(I2C_DEV_ADDR, 133)`, then exception occurs if calling `ESP.getMaxAllocHeap()` after the call `requestFrom` anywhere in the program.

Arduino ESP32: 3.2.0-RC2, IDF 5.4

### Sketch

```cpp
#include "Wire.h"

#define I2C_DEV_ADDR 0x5D

uint32_t i = 0;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Wire.begin(19, 20);
}

void loop() {
  delay(5000);

  //Write message to the slave
  Wire.beginTransmission(I2C_DEV_ADDR);
  Wire.write(0x8047 >> 8);
  Wire.write(0x8047 & 0x00ff);
  uint8_t error = Wire.endTransmission(true);
  Serial.printf("endTransmission: %u\n", error);

  //Read 16 bytes from the slave
  // 132 bytes is OK. 133 failed
  uint8_t bytesReceived = Wire.requestFrom(I2C_DEV_ADDR, 133);
  Serial.printf("requestFrom: %u\n", bytesReceived);
  if ((bool)bytesReceived) {  //If received more than zero bytes
    uint8_t temp[bytesReceived];
    Wire.readBytes(temp, bytesReceived);
    log_print_buf(temp, bytesReceived);
  }

  Serial.printf("Max alloc heap: %u\n", ESP.getMaxAllocHeap());
}
```

### Debug Message

```plain
Core  0 register dump:
PC      : 0x4202aad2  PS      : 0x00060833  A0      : 0x8202a9e7  A1      : 0x3fca4380  
A2      : 0xeb8465ec  A3      : 0x4202aa24  A4      : 0x3fca43e4  A5      : 0x3fca4400  
A6      : 0xeb8465f4  A7      : 0xeb8465f0  A8      : 0xabba1234  A9      : 0x3fca4360  
A10     : 0x00000001  A11     : 0xabba1234  A12     : 0x00000001  A13     : 0x3fca43e4  
A14     : 0x3fca4740  A15     : 0x0000cdcd  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xeb8465f0  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4202aacf:0x3fca4380 0x4202a9e4:0x3fca43a0 0x4202aa69:0x3fca43c0 0x4200bfef:0x3fca43e0 0x4200c03d:0x3fca4420 0x42002b6d:0x3fca4460 0x42001fe5:0x3fca4480 0x42007270:0x3fca44b0 0x4037d272:0x3fca44d0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2025-03-26:
Looks like uint8_t is used but the underlying code does not handle more than 7 bits correctly.
So do not read more than the verified working.

bhcuong2008, 2025-03-27:
I2C Wire is still working well to get data from touch GT911. Actually, in this library, `https://github.com/TAMCTec/gt911-arduino/blob/main/TAMC_GT911.cpp`, it reads up to 185 bytes. It causes exception only when calling getMaxAllocHeap.