espressif/arduino-esp32 #11181: esp_bt_init falied : 259

Admsher, 2025-03-26:
### Board

ESP32 CHIP

### Device Description

ESP32 One : https://www.waveshare.com/wiki/ESP32_One

### Hardware Configuration

OV2460 Camera with SD Card reader, Classic BT and TLS enabled 

### Version

latest master (checkout manually)

### IDE Name

ESP-IDF v5.3

### Operating System

Ubunt 24.04

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am currently using an A2DP library : https://github.com/pschatzmann/arduino-audio-tools
This library need arduino-eps32 to run some of it's program and also i had initially written my code on Arduino IDE and then migrated to ESP-IDF.
I went through this related issue with the same error : https://github.com/espressif/arduino-esp32/issues/3436
This has put me in doubt that if the issue above is solved in the latest version of esp-idf , if not , there is a workaround provided in the issue but I am not sure how to use it in my case.

### Sketch

```cpp
#define HELIX_LOGGING_ACTIVE false
#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/music/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext, 15);
A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Warning);
  // setup player
  // Setting up SPI if necessary with the right SD pins by calling 
  SPI.begin(14, 12, 13, 15);
  player.setVolume(0.8);
  player.begin();

  // setup output - We send the test signal via A2DP - so we conect to a Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.silence_on_nodata = true; // prevent disconnect when there is no audio data
  cfg.name = "CLIP 5";  // set the device here. Otherwise the first available device is used for output
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);
}

void loop() {
  player.copy();
}
```

### Debug Message

```plain
Please go through this issue I created in : https://github.com/pschatzmann/arduino-audio-tools/issues/1977
```

### Other Steps to Reproduce

Pleas apply this configuration: https://github.com/pschatzmann/ESP32-A2DP/wiki/Espressif-IDF-as-a-Component 
and it should lead you to the error i am stuck with.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2025-03-26:
is `btStart();` called from within that AudioTools library?

me-no-dev, 2025-03-26:
I see it being called here: https://github.com/pschatzmann/ESP32-A2DP/blob/8b29acb85bf5edeb321c2c0cd234a29661eb4f45/src/BluetoothA2DPCommon.cpp#L400

Does the sketch work in Arduino IDE?

Admsher, 2025-03-26:
Yes it works on Arduino IDE

Jason2866, 2025-03-26:
@Admsher Does PR #11476 solve the issue?

Admsher, 2025-03-26:
Nope  still the same error. If I am correct we only remove the duplicate call from esp32-hal-bt.c ?

Jason2866, 2025-03-26:
No it removes the weak trickery. It solved the issue for this example https://github.com/pioarduino/test-rainmaker/tree/wifiprov

Admsher, 2025-03-26:
I see , I was under the impression that IDF version > 5.3 should suffice, will check for the same

Jason2866, 2025-03-26:
Checked with your code (commented the SD part...) and my PR Seems BT part is doing
```
I (1247) BT_AV: app_task_handler
I (1248) BT_AV: set_scan_mode_connectable false
I (1249) BT_AV: Starting device discovery...
W (1253) BT_BTC: btc_avrc_tg_set_rn_supported_evt failed: AVRC TG not yet initialized

W (1260) BT_BTC: A2DP Enable with AVRC
W (1261) BT_AVRC: For AVRCP Target Cateory 1 and 3, SDP record shall contain additional protocol list
I (1279) BT_AV: bt_app_av_sm_hdlr state DISCOVERING, evt 0x4
I (1283) BT_AV: event: 10
I (1287) BT_AV: event: 10
I (1290) BT_AV: event: 10
I (1292) BT_AV: event: 10
I (1297) BT_AV: Discovery started.
I (11249) BT_AV: bt_app_av_sm_hdlr state DISCOVERING, evt 0xff00
I (14102) BT_AV: Device discovery failed, continue to discover...
I (14107) BT_AV: Discovery started.
I (21249) BT_AV: bt_app_av_sm_hdlr state DISCOVERING, evt 0xff00
I (26913) BT_AV: Device discovery failed, continue to discover...
I (26918) BT_AV: Discovery started.
I (31249) BT_AV: bt_app_av_sm_hdlr state DISCOVERING, evt 0xff00
I (39724) BT_AV: Device discovery failed, continue to discover...
I (39730) BT_AV: Discovery started.
I (41249) BT_AV: bt_app_av_sm_hdlr state DISCOVERING, evt 0xff00
```

Admsher, 2025-03-27:
That is quite strange, so is the SD part causing a bluetooth issue?

Jason2866, 2025-03-27:
No, everything is working. Don't have a board with a SD Card for testing.

Admsher, 2025-03-27:
Sorry for my previous comment, this commit indeed works and weak attributes seem to be the problem. Closing this issue as resolved.

Jason2866, 2025-03-29:
@me-no-dev The issue is not fixed.

me-no-dev, 2025-03-29:
Closed because "Closing this issue as resolved." was stated. I do know we still need to fix this for component and pioarduino

Jason2866, 2025-04-07:
Fix for Platformio/pioarduino is to add `lib_archive = no` to platformio.ini

Compile Arduino as an component of IDF still fails.

closed 2026-01-12 by me-no-dev