espressif/arduino-esp32 #11185: Zigbee EP that's reporting is shown as unknown in ZHA HA logbook?

def1149, 2025-03-26:
### Board

Adafruit Feather ESP32-C6

### Device Description

Unmodified Adafruit Feather ESP32-C6

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Using 3.2.0-RC2 
Developed a Zigbee Temperature Sensor that's reporting in, but the HA logbook shows it as unknown.

[image]

### Sketch

```cpp
N/A
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [x] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2025-03-28:
Hi, please post a reproducible sketch.

def1149, 2025-03-29:
It looks like this is "normal". I see it with commercial Zigbee devices as well

closed 2025-03-29 by def1149